openwrt/packages #1637: libs/intltool require XML::Parser perl module

nsapa, 2015-08-03:
libs/intltool fails to build if the system doesn't have the XML::Parser perl module.
This module should probably be packaged and be made available to the rebuild host perl.
```
(cd /home/openwrt-builder/openwrt/build_dir/host/intltool-0.40.6/; if [ -x configure ]; then cp -fpR /home/openwrt-builder/openwrt/scripts/config.{guess,sub} /home/openwrt-builder/openwrt/build_dir/host/intltool-0.40.6// &&  bash ./configure CC="gcc" CFLAGS="-O2 -I/home/openwrt-builder/openwrt/staging_dir/host/include -I/home/openwrt-builder/openwrt/staging_dir/host/usr/include" CPPFLAGS="-I/home/openwrt-builder/openwrt/staging_dir/host/include -I/home/openwrt-builder/openwrt/staging_dir/host/usr/include" LDFLAGS="-L/home/openwrt-builder/openwrt/staging_dir/host/lib -L/home/openwrt-builder/openwrt/staging_dir/host/usr/lib" SHELL="/usr/bin/env bash" --target=x86_64-linux-gnu --host=x86_64-linux-gnu --build=x86_64-linux-gnu --program-prefix="" --program-suffix="" --prefix=/home/openwrt-builder/openwrt/staging_dir/host --exec-prefix=/home/openwrt-builder/openwrt/staging_dir/host --sysconfdir=/home/openwrt-builder/openwrt/staging_dir/host/etc --localstatedir=/home/openwrt-builder/openwrt/staging_dir/host/var --sbindir=/home/openwrt-builder/openwrt/staging_dir/host/bin ; fi )
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking for perl... /home/openwrt-builder/openwrt/staging_dir/host/bin/perl
checking for perl >= 5.8.1... 5.20.2
checking for XML::Parser... configure: error: XML::Parser perl module is required for intltool
Makefile:47: recipe for target '/home/openwrt-builder/openwrt/build_dir/host/intltool-0.40.6/.configured' failed
make[3]: *** [/home/openwrt-builder/openwrt/build_dir/host/intltool-0.40.6/.configured] Error 1
make[3]: Leaving directory '/home/openwrt-builder/openwrt/feeds/packages/libs/intltool'
package/Makefile:191: recipe for target 'package/feeds/packages/intltool/host/compile' failed
make[2]: *** [package/feeds/packages/intltool/host/compile] Error 2
make[2]: Leaving directory '/home/openwrt-builder/openwrt'
package/Makefile:188: recipe for target '/home/openwrt-builder/openwrt/staging_dir/target-mips_mips32_musl-1.1.10/stamp/.package_compile' failed
make[1]: *** [/home/openwrt-builder/openwrt/staging_dir/target-mips_mips32_musl-1.1.10/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/openwrt-builder/openwrt'
/home/openwrt-builder/openwrt/include/toplevel.mk:181: recipe for target 'world' failed
make: *** [world] Error 2
```

plntyk, 2015-08-03:
This requirement is documented for a long time at : http://wiki.openwrt.org/doc/howto/buildroot.exigence

thess, 2015-08-12:
Is there any reason this cannot be closed? Do we really want to accept #1648 which forces a build of perl for the host?

Comments?

closed 2015-08-17 by thess