openwrt/packages #1642: privoxy: any legitimate reason for hooking up the server with OpenWRT's hotplug events?

braveheartleo, 2015-08-06:
I noticed that Privoxy 3.0.23 is started at first boot not via its init.d script, but via its hotplug.d/iface script. It is inefficient to say the least.

The server is repeatedly and unnecessarily restarted after every ifup hotplug event. I have setup my router for 1 LAN and 4 WLAN interfaces, all of them routed. As you can imagine, as these interfaces are brought up, the privoxy server is restarted. I log 6 to 8 times of privoxy restarts in its logfile. This is magnified even more when I have setup a Dual-Wan configuration using mwan3, that initiates iface hotplug events as it initializes every configured WAN interface.

I have dropped the hotplug script, and modified the init.d script so that its inherited boot() function is no longer overridden. This way Privoxy starts up normally at boot without error, whether I have set it up to listen on ALL ip addresses (with ACLs of course), or just my local interfaces. And it is no longer chained to hotplug events that causes the server to be restarted. Most of all, this cuts the router's startup time by a couple of seconds.

Is there any reason to have that hotplug script in the first place for privoxy, when the server starts up normally without the need to listen on hotplug events? I might add that the server is accessible in all my interfaces even when started this way.

chris5560, 2015-08-06:
I setup using hotplug events because if an interface ip changes (I'm using 6to4 for IPv6 and provider IPv4 might change) you need to restart privoxy. Other problem I've seen that privoxy comes up but interfaces where not up with their IP's so privoxy did not recognize them later on. privoxy auto-reconnect only when sending request to an existing/connected interface. privoxy has no kill -SIGHUB or similar to reload config and reconnect listen address/port.
I already asked privoxy development and get the answer "restart privoxy".

braveheartleo, 2015-08-06:
> Other problem I've seen that privoxy comes up but interfaces where not up with their IP's so privoxy did not recognize them later on

Indeed, the server cannot bind a specific IP address if it were not present when the server is started. But these issues only tell that Privoxy is being started much earlier than when interfaces have been fully setup, and is a matter of startup timing and adjusting its startup sequence appropriately.

To cover the different setups that each router have, and different timings at that, Privoxy may be setup by default to listen on all IP addresses, and then have `permit-access` rules specify the IP ranges that the server will be allowed to talk to. This is how Dnsmasq is setup, and may be limited for DNS service to local interfaces via its `local-service` config. Just as well, its DHCP service may be limited for service via its `no-dhcp-interface` config.

Of course the above are not necessary for simple setups like bridged LAN and WLAN interfaces, which is the default in OpenWRT. I'm just saying that it is unnecessary, and quite inefficient, to be having Privoxy restarted for every (!) ifup hotplug events, because it is expected that the server be started at the appropriate moment when conditions for its proper operations are met (at least this seems to be the case for Privoxy, because some servers are more flexible, see avahi below).

>  setup using hotplug events because if an interface ip changes (I'm using 6to4 for IPv6 and provider IPv4 might change) you need to restart privoxy.

I haven't used 6to4, but I would assume that if Privoxy were designed to cope up with dynamically changing interfaces (like the Avahi daemon where it can join or leave mDNS multicast groups and withdraw workstation service for interfaces as they come and go or change addresses), then it would have some mechanisms for doing so.

> privoxy has no kill -SIGHUB or similar to reload config and reconnect listen address/port

And like you said, Privoxy is pretty much expecting a static environment when run, and any changes in this environment would require its manual restart, just like how its dev tell us. (However, Privoxy can automatically re-read its action and filter files for changes, for example).

Given all these, it would seem that having Privoxy start normally at boot (within the appropriate start sequence of course) and not have any hotplug script for it, is sufficient for the default OpenWRT setup. Any further requirement other than this (OpenWRT's default configuration SHOULD always be our baseline) is already non-default, and is up to each individual to implement.

Of course if we could request to the Privoxy devs for features like dynamically coping up with changing interfaces would be great, if the devs would entertain it. I don't mind if you would have a hotplug script for it, just not in its current shape now.

chris5560, 2015-08-07:
I think we start a philosophical discussion.

Since netifd nobody knows when all interfaces are up.
In "good old times" when "S20network" has finished network was up and afterwards various services i.e. firewall were started.

I will not bring old times back, but we need to provide a system that guarantee "home users", that an installed service works as expected.
If the service was restarted 1 or 100 times at startup not really matter.
Using procd does also not fix the problem. Other services using procd are also restarted on every "ifup" i.e. firewall.
Why not start firewall if all interfaces are up ? Because you cannot detect if all interfaces are up !!!

I looked into my LinuxMint 17.2 box and found that inside /etc/init are a lot of *.conf files with thousands of definitions what condition to wait for before start/stop or anything else. This is no solution for OpenWrt system.

I think you go the right way to disable hotplug scripts and enable boot start for privoxy inside your environment.
Yes it is ineffective to restart multiple times.
You are the "non casual user" how can work around and know what to do if a service not working as expected.

I use dnsmasq together with a long blocklist and it takes 2 seconds until my Netgear WNDR3800 has read it and dnsmasq is up and running.
and than these 8 "ifup" events: loop lan wan wan6 wlan2 wlan5 wlan2guest wlan5guest
and every time firewall and others restarting, but at the end of the day the system is up and running fine.

Reading the forum entry's and the kind of questions, we must provide a system for "casual users": 
Install package -> configure via LuCI -> works as expected
without system knowledge. (Usable for "GUI" users).

Christian

braveheartleo, 2015-08-08:
> I think we start a philosophical discussion.

There's nothing philosophical about the foregoing discussion. Technical is much more appropriate. Though, if you like it philosophical, then you're committing the Fallacy of the Golden Past by saying that "in the 'good old times', before netifd, nobody knows when all interfaces are up," and so you thought it was "good" to have YOUR hotplug script that trigger-happily restarted the server back then and now.

Did I say that a default Privoxy install started successfully at boot, WITHOUT the need for the hotplug script, in the default OpenWRT configuration? I guess you just chose to ignore that.

If the default OpenWRT configuration isn't "casual" enough, then I don't know what is.

> Using procd does also not fix the problem. Other services using procd are also restarted on every "ifup" i.e. firewall.

Dnsmasq is also hooked up for ANY "ifup" event. But Dnsmasq uses procd, and thus hotplugging Dnsmasq become "smart" enough to know the timing when to restart the service, avoiding UNNECESSARY restarts, than for EVERY "ifup" event, literally. Do you see the difference?

> If the service was restarted 1 or 100 times at startup not really matter.

The issue has been moot right from the start if that has been your coding etiquette/programming practice. You are the maintainer after all, and if you think that way then so be it. I have nothing further to add.

diizzyy, 2017-06-07:
@chris5560 @hauke @jow- 
Is there a more elegant way to work around this issue?

jow-, 2017-06-08:
I think the elegant solution would be switching to procd, that would auto-coalesce the ifup events and also not issue an actual instance restart if the calculated desired state did not change.

feckert, 2020-11-04:
This issue has been sitting around here for a long time. So I would say this can be closed. There has been no progress.

closed 2020-11-04 by feckert