openwrt/packages #1643: luasocket: requiring io in mime.lua fails under certain conditions

Diftraku, 2015-08-06:
We had this small edge case where we use Lua internally in a daemon-like program which includes a bunch on .lua files and then runs them from a predefined entrypoint. Platform aside, we use luasocket from Barrier Breaker tag and `mime.lua` breaks when it tries to `require('io')` (located in `/usr/lib/lua/socket`, while `mime.lua` is a directory up the path.)

Removing the `require` from `/usr/lib/lua/mime.lua` @ line 13 fixes the issue on our internal daemon and does not affect the `lua`-console (`socket.http` works regardless of the aforementioned fix.)

KanjiMonster, 2015-08-20:
@MikePetullo

diizzyy, 2017-06-07:
@MikePetullo 
Is this still valid?

MikePetullo, 2017-06-07:
I would have to defer to the original submitter. @Diftraku?

Diftraku, 2017-06-07:
We would need to do a test build without the patch we have been using for the past years and check against our current Chaos Calmer version.

Quick glance at the packages in CC and BB would suggest the version of liblua has not changed during the past two years which would suggest the bug is still present (assuming it is not some other package causing it than liblua and scoket.http).

karlp, 2017-10-26:
Isn't this a bug in luasocket then? not in openwrt's package?

neheb, 2020-01-12:
Is this still an issue?

commodo, 2023-05-27:
closing;

feel free to re-open if i'm wrong

closed 2023-05-27 by commodo