openwrt/packages #1649: dnsmasq config options are ignored

alexlist, 2015-08-11:
The documentation makes certain claims regarding configuration of dnsmasq:

http://wiki.openwrt.org/doc/uci/dhcp

Trying to specify a nameserver for a particular domain like this:
`option server '/example.com/192.0.2.1'`

is not reflected in the generated config in ```/var/etc/dnsmasq.conf```

I found that the relevant snippet is missing from ```/etc/init.d/dnsmasq```

This change will at least fix this for the ```server``` option, however I suggest to go over the books and verify that everything that's mentioned in the documentation is respected.

```
--- dnsmasq	2015-08-11 05:15:43.000000000 +0000
+++ dnsmasq.opkg	2015-08-11 05:14:57.000000000 +0000
@@ -145,7 +145,6 @@
 	append_parm "$cfg" "queryport" "--query-port"
 	append_parm "$cfg" "domain" "--domain"
 	append_parm "$cfg" "local" "--server"
-	append_parm "$cfg" "server" "--server"
 	config_list_foreach "$cfg" "server" append_server
 	config_list_foreach "$cfg" "address" append_address
 	config_list_foreach "$cfg" "ipset" append_ipset
```

Encountered on TP-Link Archer C7 v2 using the stock dnsmasq (not dnsmasq-full!)

```
# cat /etc/openwrt_release 
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='15.05-rc3'
DISTRIB_REVISION='r46163'
DISTRIB_CODENAME='chaos_calmer'
DISTRIB_TARGET='ar71xx/generic'
DISTRIB_DESCRIPTION='OpenWrt Chaos Calmer 15.05-rc3'
DISTRIB_TAINTS=''
```

jow-, 2015-08-11:
It must be a uci list in the form:

    list server '/example.com/192.0.2.1'
    list server '/example.org/192.3.2.1'

closed 2015-08-11 by jow-

alexlist, 2015-08-12:
Thanks for the clarification... and sorry, yes it is documented as being a list of strings.