openwrt/packages #1654: coreutils-stdbuf does not work on for-14.07 branch

rlsaurabh, 2015-08-12:
issue in for-14.07 branch

using stdbuf on openwrt gives me this error

stdbuf: failed to find 'libstdbuf.so'

rlsaurabh, 2015-08-12:
Not sure about the path procedure, but this patch works.
This is influenced from https://dev.openwrt.org/ticket/11076

diff --git a/utils/coreutils/Makefile b/utils/coreutils/Makefile
index 4500e88..addcc91 100644
--- a/utils/coreutils/Makefile
+++ b/utils/coreutils/Makefile
@@ -100,6 +100,11 @@ define BuildPlugin
   define Package/$(1)/install
 	$(INSTALL_DIR) $$(1)/usr/bin
 	$(INSTALL_BIN) $(PKG_INSTALL_DIR)/usr/bin/$(2) $$(1)/usr/bin/
+
+    ifeq ($(2),stdbuf)
+	$(INSTALL_DIR) $$(1)/usr/lib/coreutils
+	$(INSTALL_BIN) $(PKG_INSTALL_DIR)/usr/lib/coreutils/libstdbuf.so $$(1)/usr/lib/coreutils/
+    endif
   endef
 
   $$(eval $$(call BuildPackage,$(1)))

bittorf, 2016-06-23:
can you please send a pullrequest for 15.05 and trunk?

makefu, 2016-09-15:
i was stumbling over the same issue - can this patch be included into trunk as well as into the backports? because without it stdbuf is unusable ...

diizzyy, 2017-06-02:
@hnyman @thess 
14.07 is frozen, close this.

closed 2017-06-02 by thess