openwrt/packages #1665: bluez compile error

taeasy, 2015-08-18:
nwrt-linux-musl-" ARCH="x86_64" ;
make[4]: Entering directory `/home/main/x86/build_dir/target-x86_64_musl-1.1.10/bluez-5.33'
make --no-print-directory all-am
/bin/bash ./libtool  --tag=CC   --mode=link x86_64-openwrt-linux-musl-gcc    -I/home/main/x86/staging_dir/target-x86_64_musl-1.1.10/usr/include/dbus-1.0 -I/home/main/x86/staging_dir/target-x86_64_musl-1.1.10/usr/lib/dbus-1.0/include  -I/home/main/x86/staging_dir/target-x86_64_musl-1.1.10/usr/include/glib-2.0 -I/home/main/x86/staging_dir/target-x86_64_musl-1.1.10/usr/lib/glib-2.0/include  -pthread -I/home/main/x86/staging_dir/target-x86_64_musl-1.1.10/usr/include/glib-2.0 -I/home/main/x86/staging_dir/target-x86_64_musl-1.1.10/usr/lib/glib-2.0/include   -Os -pipe -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -I/home/main/x86/staging_dir/target-x86_64_musl-1.1.10/usr/lib/libiconv-full/include -I/home/main/x86/staging_dir/target-x86_64_musl-1.1.10/usr/lib/libintl-full/include   -L/home/main/x86/staging_dir/target-x86_64_musl-1.1.10/usr/lib -L/home/main/x86/staging_dir/target-x86_64_musl-1.1.10/lib -L/home/main/x86/staging_dir/toolchain-x86_64_gcc-4.8-linaro_musl-1.1.10/usr/lib -L/home/main/x86/staging_dir/toolchain-x86_64_gcc-4.8-linaro_musl-1.1.10/lib -znow -zrelro -L/home/main/x86/staging_dir/target-x86_64_musl-1.1.10/usr/lib/libiconv-full/lib -L/home/main/x86/staging_dir/target-x86_64_musl-1.1.10/usr/lib/libintl-full/lib  -o client/bluetoothctl client/main.o client/display.o client/agent.o client/gatt.o monitor/uuid.o gdbus/libgdbus-internal.la -L/home/main/x86/staging_dir/target-x86_64_musl-1.1.10/usr/lib -lglib-2.0 -lintl  -L/home/main/x86/staging_dir/target-x86_64_musl-1.1.10/usr/lib -lgthread-2.0 -pthread -lglib-2.0 -lintl  -L/home/main/x86/staging_dir/target-x86_64_musl-1.1.10/usr/lib -ldbus-1 -lreadline -lncurses 
libtool: link: x86_64-openwrt-linux-musl-gcc -I/home/main/x86/staging_dir/target-x86_64_musl-1.1.10/usr/include/dbus-1.0 -I/home/main/x86/staging_dir/target-x86_64_musl-1.1.10/usr/lib/dbus-1.0/include -I/home/main/x86/staging_dir/target-x86_64_musl-1.1.10/usr/include/glib-2.0 -I/home/main/x86/staging_dir/target-x86_64_musl-1.1.10/usr/lib/glib-2.0/include -pthread -I/home/main/x86/staging_dir/target-x86_64_musl-1.1.10/usr/include/glib-2.0 -I/home/main/x86/staging_dir/target-x86_64_musl-1.1.10/usr/lib/glib-2.0/include -Os -pipe -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z -Wl,now -Wl,-z -Wl,relro -I/home/main/x86/staging_dir/target-x86_64_musl-1.1.10/usr/lib/libiconv-full/include -I/home/main/x86/staging_dir/target-x86_64_musl-1.1.10/usr/lib/libintl-full/include -znow -zrelro -o client/bluetoothctl client/main.o client/display.o client/agent.o client/gatt.o monitor/uuid.o -pthread  -L/home/main/x86/staging_dir/target-x86_64_musl-1.1.10/usr/lib -L/home/main/x86/staging_dir/target-x86_64_musl-1.1.10/lib -L/home/main/x86/staging_dir/toolchain-x86_64_gcc-4.8-linaro_musl-1.1.10/usr/lib -L/home/main/x86/staging_dir/toolchain-x86_64_gcc-4.8-linaro_musl-1.1.10/lib -L/home/main/x86/staging_dir/target-x86_64_musl-1.1.10/usr/lib/libiconv-full/lib -L/home/main/x86/staging_dir/target-x86_64_musl-1.1.10/usr/lib/libintl-full/lib gdbus/.libs/libgdbus-internal.a -lgthread-2.0 -lglib-2.0 -lintl -ldbus-1 -lreadline -lncurses -pthread
/home/main/x86/staging_dir/toolchain-x86_64_gcc-4.8-linaro_musl-1.1.10/lib/gcc/x86_64-openwrt-linux-musl/4.8.3/../../../../x86_64-openwrt-linux-musl/bin/ld: warning: libiconv.so.2, needed by /home/main/x86/staging_dir/target-x86_64_musl-1.1.10/usr/lib/libgthread-2.0.so, not found (try using -rpath or -rpath-link)
/home/main/x86/staging_dir/target-x86_64_musl-1.1.10/usr/lib/libglib-2.0.so: undefined reference to `libiconv'
/home/main/x86/staging_dir/target-x86_64_musl-1.1.10/usr/lib/libglib-2.0.so: undefined reference to `libiconv_open'
/home/main/x86/staging_dir/target-x86_64_musl-1.1.10/usr/lib/libglib-2.0.so: undefined reference to `libiconv_close'
collect2: error: ld returned 1 exit status
make[5]: *** [client/bluetoothctl] Error 1
make[4]: *** [all] Error 2
make[4]: Leaving directory `/home/main/x86/build_dir/target-x86_64_musl-1.1.10/bluez-5.33'
make[3]: *** [/home/main/x86/build_dir/target-x86_64_musl-1.1.10/bluez-5.33/.built] Error 2
make[3]: Leaving directory `/home/main/x86/feeds/packages/utils/bluez'
make[2]: *** [package/feeds/packages/bluez/compile] Error 2
make[2]: Leaving directory `/home/main/x86'
make[1]: *** [/home/main/x86/staging_dir/target-x86_64_musl-1.1.10/stamp/.package_compile] Error 2
make[1]: Leaving directory `/home/main/x86'
make: *** [world] Error 2
main@bogon:~/x86$

hefish, 2015-09-10:
i find all packages depended on libiconv are broken.

hauke, 2015-09-11:
This was fixed in this commit:
https://dev.openwrt.org/changeset/46801

closed 2015-09-11 by hauke