openwrt/packages #1674: coreutils-stdbuf not working

rlsaurabh, 2015-08-19:
I'm sending icmp packets on eth0 interface on an OpenWRT device(tl-mr3020). When I do a tcpdump and pipe the output, to filter timestamps, its getting buffered.
This is the command I used
tcpdump -eni eth0 -v icmp | awk '{print $1}'

I managed to compile stdbuf and used 
stdbuf -o0 tcpdump -eni eth0 -v icmp | awk '{print $1}'
But even this did not help. 

I think there is some bug in the OpenWRT implementation on stdbuf, as this worked for me on my ubuntu terminal. Any workaround for this?

diizzyy, 2017-06-07:
@rlsaurabh 
Still valid?

nfitzkee, 2017-06-18:
I believe this is still a problem. I am experiencing a similar problem with coreutils-8.23.  I am using Entware, but the core sources are still largely OpenWRT.

yousong, 2017-06-18:
fyi, tcpdump itself supports a -l option to make its output line-buffered

nfitzkee, 2017-06-18:
I have confirmed that stdbuf is still broken with both coreutils-8.23 and coreutils-8.27.  While I am getting an error message, I believe the problem is related to @rlsaurabh 's problem above.  See output below:

```
root@DD-WRT:~# echo $PATH
/bin:/usr/bin:/sbin:/usr/sbin:/jffs/sbin:/jffs/bin:/jffs/usr/sbin:/jffs/usr/bin:/mmc/sbin:/mmc/bin:/mmc/usr/sbin:/mmc/usr/bin:/opt/sbin:/opt/bin:/opt/usr/sbin:/opt/usr/bin
root@DD-WRT:~# which stdbuf
/opt/bin/stdbuf
root@DD-WRT:~# stdbuf -o0 date
Error relocating /opt/bin/libstdbuf.so: __fprintf_chk: symbol not found
root@DD-WRT:~# export PATH="/opt/bin:$PATH"
root@DD-WRT:~# stdbuf -o0 date
Fri Jun 16 20:32:32 CDT 2017
root@DD-WRT:/opt/bin# stdbuf -oL tcpdump -eni eth0 icmp
Error relocating /opt/bin/libstdbuf.so: __fprintf_chk: symbol not found
```

As I said, I am seeing the same behavior on both 8.23 (current OpenWRT) as well as 8.27 (latest).  I haven't been able to get prior versions to compile; however, based on an old installation of Optware (OTWR2), 8.4 at least has the potential to work. That said, though, there might be something far more subtle going on here.  I have posted to the coreutils mailing list and will post updates here if I hear anything.

nfitzkee, 2017-06-18:
@yousong Thanks for the tip - unfortunately, my application doesn't use tcpdump, but the OP's example is far clearer than me trying to explain my specific application (which uses sed). Unfortunately, sed -u isn't quite good enough for what I need - which is true line-buffering.

Even if sed did support line buffering though, stdbuf would still be broken.

nfitzkee, 2017-06-19:
I contacted the Coreutils developers, and Pádraig Brady suggested a workable solution. For more details, please see http://lists.gnu.org/archive/html/coreutils/2017-06/msg00026.html

The solution is to add two CFLAGS when building libstdbuf.so.  Using coreutils 8.23, I added the following to configure.ac:

```
CFLAGS+=" -fno-stack-protector -D_FORTIFY_SOURCE=0"
```

Then, I ran autoconf (or autoreconf) to generate the new configure script and re-built the packages.  After recompiling, stdbuf is now giving the correct behavior in the tcpdump example above.

Technically, the CFLAGS statement adds these flags to all parts of the coreutils build, not simply libstdbuf.so, but I didn't see an easy way to do that in the autoconf/automake framework.  That issue is outside of my expertise, but it's probably trivial for someone who regularly works with OpenWRT packages.

yousong, 2017-06-19:
Interesting.  I thought and confirmed this was caused by missing libstdbuf.so in the packaging of coreutils-stdbuf.  I am actually testing my changes adding that .so file...

yousong, 2017-06-19:
Hi, I just opened a pull request #4493 which should fix the issue.  Please review and test

PR #4493 (2017-06-19): coreutils: stdbuf: fix missing libstdbuf.so
Maintainer: @jow- 
Compile tested: qemu malta be
Run tested: qemu malta be

Fixes #1674
commit 094cf18b (2017-06-19): coreutils: stdbuf: fix missing libstdbuf.so

Fixes #1674
commit 46014e36 (2017-06-19): coreutils: stdbuf: fix missing libstdbuf.so

Fixes #1674
closed 2017-06-19 by jow-