openwrt/packages #1679: vpnc fails: arbitrary naming tunnel interface is not supported in this Version

psx01de, 2015-08-19:
Hi, in cc with the new configuration with uci - the connection fails with:
"arbitrary naming tunnel interface is not supported in this Version"

With the old method I get: "can't initialise tunnel interface: No such file or Directory"

I ported my old vpnc-default.conf to these options:
config interface 'vpn'
    option server 'server'
    option username 'user'
    option password 'pass'
    option authgroup 'user'
    option passgroup 'secret'
    option local_port '0'

I try to set more option but it doesn't matter....

KanjiMonster, 2015-08-20:
@danielg4

danielg4, 2015-08-24:
Are you sure the server is running Cisco VPN Concentrator?

psx01de, 2015-08-24:
The server is a FritzBox router from AVM. The connection already works in an older version of openwrt and vpnc.

danielg4, 2015-08-24:
Is this with the latest revision of the package in trunk?

psx01de, 2015-08-25:
I compiled openwrt and the packages again with the latest sources from trunk (OpenWrt Chaos Calmer r46724 / LuCI Master (git-15.236.28194-084d435) ) and the problem continues:

Tue Aug 25 23:24:29 2015 daemon.notice netifd: Interface 'MYVPN' is setting up now
Tue Aug 25 23:24:29 2015 user.notice vpnc: initializing...
Tue Aug 25 23:24:30 2015 user.notice vpnc: executing 'vpnc --no-detach --pid-file /var/run/vpnc-MYVPN.pid --ifname vpn-MYVPN --non-inter --script /lib/netifd/vpnc-script /var/etc/vpnc-MYVPN.conf'
Tue Aug 25 23:24:30 2015 daemon.notice netifd: MYVPN (4718): error: arbitrary naming tunnel interface is not supported in this version
Tue Aug 25 23:24:30 2015 daemon.notice netifd: MYVPN (4718):
Tue Aug 25 23:24:30 2015 user.notice vpnc: bringing down vpnc
Tue Aug 25 23:24:30 2015 daemon.notice netifd: Interface 'MYVPN' is now down

danielg4, 2015-08-25:
OK, I have looked in the source code, and the line that prints the error you report is never executed if vpnc is compiled for Linux. What kind of system are you building on? What development headers do you have?

psx01de, 2015-08-26:
Thank you for analyzing! I am using the OpenWRT Buildroot like it's described in the wiki on an Ubunutu 14.04.3 LTS System (virtualized in VirtualBox). I have done nothing else than 1:1 follow the instructions (git clone (cc trunk), update & install Feeds, make menuconfig, make). The target device is TL-MR3220v1. For the latest builds I have compiled vpnc with openssl, but I tried it also with gnutls before.

danielg4, 2015-08-26:
You know what? I think this is probably an artifact of the migration to musl between BB and CC. I will look into this.

psx01de, 2015-08-30:
Oh you're right, I tried it with the latest 15.05 repository and there is no such error. But even I don't get any connection because the connection is shut down directly without any error. 

Sun Aug 30 10:21:34 2015 daemon.notice netifd: Interface 'vpn' is setting up now
Sun Aug 30 10:21:35 2015 user.notice vpnc: initializing...
Sun Aug 30 10:21:35 2015 user.notice vpnc: bringing down vpnc
Sun Aug 30 10:21:35 2015 daemon.notice netifd: Interface 'vpn' is now down

If I try to bring up the interface the second time, nothing happens anymore (even no initializing). It seems to stop between logger -t vpnc "initializing..." and mkdir -p /var/etc in vpnc.sh, but this is another problem.. I try to analyze is as far as I can and report it maybe in another issue report.

Thanks for your help!

danielg4, 2015-08-30:
There is a bug in the connection script in the 15.05 repository.

danielg4, 2015-08-30:
Also, you have "option interface 'vpn'" but do you already have such an interface defined? If you have only one interface that is connected to the Internet, you can leave that line out.

danielg4, 2015-08-30:
Is there a file /tmp/etc/vpnc-vpn.conf created?

danielg4, 2015-08-30:
Analyzing your issue as now reported, it's waiting for the server address to be routeable.

danielg4, 2015-08-30:
Please disregard the now-deleted comments; I just woke up. Try 15.05 for the system and trunk for packages, and report back.

danielg4, 2015-08-30:
Also, make sure you did not typo the server name.

psx01de, 2015-08-30:
Which packages I should install from trunk? If I change the opkg source for packages to trrunk I get many dependency errors.

danielg4, 2015-08-30:
vpnc

psx01de, 2015-08-31:
I try to install the vpnc package from trunk (opkg install vpnc_0.5.3.r550-3_ar71xx.ipk) but then vpnc is broken (root@OpenWrt:# vpnc results in -ash: vpnc: not found).

danielg4, 2015-08-31:
I meant, when you build your system, build vpnc from trunk and the rest from 15.05, sorry I wasn't clear.

psx01de, 2015-09-01:
Hey, I think it works with the latest 15.05 System with vpn compiled from trunk (vpnc_0.5.3.r550-3). The error from above is gone but actually the connection does not work and breaks after about a minute - but I think ist a detail problem with the configuration of vpnc or the firewall. I'll try to change the configuration and write back if it works.

danielg4, 2015-09-01:
@KanjiMonster 
Can you backport the most recent vpnc commits to the 15.05 branch? I do not have write access.

jow-, 2015-09-03:
@danielg4 I cherry picked all updates into the for-15.05 branch.

danielg4, 2015-09-03:
@jow- 
See my question at https://github.com/openwrt/packages/issues/1416

sudden6, 2015-11-09:
Hello I have exactly the same problem with trunk an CC

```
Mon Nov  2 10:26:52 2015 daemon.notice netifd: vpn (2798): 
Mon Nov  2 10:26:52 2015 user.notice vpnc: bringing down vpnc
Mon Nov  2 10:26:52 2015 daemon.notice netifd: Interface 'vpn' is now down
Mon Nov  2 10:26:52 2015 daemon.notice netifd: Interface 'vpn' is setting up now
Mon Nov  2 10:26:52 2015 user.notice vpnc: initializing...
Mon Nov  2 10:26:52 2015 user.notice vpnc: executing 'vpnc --no-detach --pid-file /var/run/vpnc-vpn.pid --ifname vpn-vpn --non-inter --script /lib/netifd/vpnc-script /var/etc/vpnc-vpn.conf'
Mon Nov  2 10:26:52 2015 daemon.notice netifd: vpn (2823): error: arbitrary naming tunnel interface is not supported in this version
```

the .config I was using
https://gist.github.com/sudden6/03684e6d3314f6a7fa2f

sudden6, 2015-11-10:
I think it is the same problem as discussed here: http://www.openwall.com/lists/musl/2012/10/11/4

I don't have a clue how to solve it though :(

danielg4, 2015-11-12:
Try it with the new update I made. I'm hoping it does the trick.

sudden6, 2015-11-12:
Sadly I bricked my router in the process of testing a few solutions, but I will report when I have fixed it.

danielg4, 2015-11-13:
I've tried it myself now, and it's working.

sudden6, 2015-12-03:
I was also able to compile it now, I think this can be closed.

diizzyy, 2017-06-07:
@thess @hnyman 
Resolved, please close this.

closed 2017-06-07 by thess