openwrt/packages #1686: vsftpd has wrong maintainer

KanjiMonster, 2015-08-20:
net/vsftpd still has the maintainer from the old packages feed, which does not match the pull request initiator.

@hnyman since you added the package, please change the maintainer to yourself.

hnyman, 2015-08-23:
I have seen nowhere an explicit requirement that the pull request initiator should be the maintainer, if the package already has a maintainer who is active.  

https://github.com/openwrt/packages/blob/master/CONTRIBUTING.md :
A (PKG_)MAINTAINER definition listing either yourself or another person in the field.

The currently defined package maintainer @obsy has also been active on the package during the past year, so I see nothing really fundamentally wrong with the current maintainer situation.

But I guess that I could assume the maintainership, if that change is really needed.

KanjiMonster, 2015-08-23:
If @obsy is fine with it, then there is no issue, I just assumed that not changing the original MAINTAINER was an oversight and you intended to take over. Usually I would expect packages where the original maintainer is still interested in maintaining it to be brought over by the maintainer him-/herself ;-)

Maybe we should require an Acked-by from the original maintainer if it differs for future pull requests.

obsy, 2015-08-23:
@hnyman: no problem, if you want maintain this package.

karlp, 2016-05-24:
Sounds like this ticket can be closed? Is there anything further here?

closed 2016-05-24 by hnyman