openwrt/packages #1692: packeges mosquitto and mosquitto-nossl broken in snapshots/trunk/ar71xx/generic/packages/packages

mrdunk, 2015-08-20:
originally posted here in error:
https://dev.openwrt.org/ticket/20362

i believe this is un-reported:
 - problem:
   - the mosquitto broker crashes with the message "Bus error" when any node publishes to the broker via the "mqtt" protocol.
   - this happens reliably. (100% of the time.)
   - the "websocket" protocol appears unaffected.

- these 2 nodes exhibit the problem:
  - GL-iNet 6416A v1
  OpenWrt Chaos Calmer r46612 / LuCI (git-15.216.69575-bb7ea3e)

  - TP-Link TL-WR710N v1
  OpenWrt Chaos Calmer r46589 / LuCI (git-15.216.69575-bb7ea3e)

- steps to reproduce:
  on OpenWRT node:
  ```
root@OpenWrt:/etc# opkg install mosquitto
root@OpenWrt:/etc# cd 
root@OpenWrt:~# opkg install mosquitto
Installing mosquitto (1.4.2-2) to root...
Downloading http://downloads.openwrt.org/snapshots/trunk/ar71xx/generic/packages/packages/mosquitto_1.4.2-2_ar71xx.ipk.
Configuring mosquitto.

root@OpenWrt:~# killall mosquitto

root@OpenWrt:~# /usr/sbin/mosquitto -v -c /etc/mosquitto/mosquitto.conf
1440012117: mosquitto version 1.4.2 (build date 2015-08-17 09:48:38+0000) starting
1440012117: Config loaded from /etc/mosquitto/mosquitto.conf.
1440012117: Opening ipv4 listen socket on port 1883.
1440012117: Opening ipv6 listen socket on port 1883.
  ```
  on another node (where 192.168.192.9 is the OpenWRT node):
  ```
duncan@lapdancer:~$ mosquitto_pub -h 192.168.192.9 -p 1883 -t homeautomation/lighting/all/all -m "#000" -d
Received CONNACK
Sending PUBLISH (d0, q0, r0, m1, 'homeautomation/lighting/all/all', ... (4 bytes))
duncan@lapdancer:~$
  ```
  back on the OpenWRT node:
  ```
root@OpenWrt:~# /usr/sbin/mosquitto -v -c /etc/mosquitto/mosquitto.conf
1440012242: mosquitto version 1.4.2 (build date 2015-08-17 09:48:38+0000) starting
1440012242: Config loaded from /etc/mosquitto/mosquitto.conf.
1440012242: Opening ipv4 listen socket on port 1883.
1440012242: Opening ipv6 listen socket on port 1883.
1440012246: New connection from 192.168.192.62 on port 1883.
1440012246: New client connected from 192.168.192.62 as mosq_pub_4904_duncan (c1, k60).
1440012246: Sending CONNACK to mosq_pub_4904_duncan (0, 0)
1440012246: Received PUBLISH from mosq_pub_4904_duncan (d0, q0, r0, m0, 'homeautomation/lighting/all/all', ... (4 bytes))
1440012246: Received DISCONNECT from mosq_pub_4904_duncan
1440012246: Client mosq_pub_4904_duncan disconnected.
Bus error
root@OpenWrt:~# ps |grep mosquitto
 7318 root      1192 S    grep mosquitto
root@OpenWrt:~#
  ```

please let me know if i can be of any assistance.
dunk.

karlp, 2015-08-23:
Impressive.  I'll handle this monday back in the office (and will also be updating to mosquitto 1.4.3 as well)  I've not done a lot of testing in CC, but I've got "many" nodes running this on BB, so I'm kinda impressed at getting a bus error :)

karlp, 2015-08-23:
Did I read that right that you're installing the mosquitto package from snapshots (ie, trunk) onto a CC branch device?  Is that even supported?

mrdunk, 2015-08-23:
hi,
i'm fairly inexperienced with OpenWRT so i hope i get the naming conventions correct here...
(apologies in advance if i don't.)
both the nodes i experienced this issue on were installed from trunk and are running trunk packages (including mosquitto).
to confirm i'm getting the naming scheme correct, the images for both these test nodes were found here:
https://downloads.openwrt.org/snapshots/trunk/ar71xx/generic/

i believe trunk is the only option for the GL-iNet device as it's not supported yet in any of the pre-built releases.

the TP-Link TL-WR710N doesn't have to run trunk; i only installed that to confirm the issues i was sing on the GL-iNet.

mrdunk, 2015-08-23:
also,
BB on the TP-Link TL-WR710N works correctly: mosquitto does not crash.

i *think* the non-trunk version of CC from here: Index of /chaos_calmer/15.05-rc3/ar71xx/generic/
worked correctly on the TP-Link too. (though i'm not 100% sure i'm remembering that correctly though... let me know if you want me to test.)

karlp, 2015-08-24:
Reproduced here, I guess it's some musl fun times. 

--- SIGBUS {si_signo=SIGBUS, si_code=SI_USER, si_errno=ENETUNREACH, si_pid=0, si_uid=0} ---

talking with upstream on whether they have any ideas on it.

karlp, 2015-08-24:
http://paste.fedoraproject.org/258526/14404227/ fixes this, but I'm holding off on committing this as a patch while I get some feedback from upstream.

karlp, 2015-08-25:
I've just pushed a fix for this to master.  Do you have an environment with the CC (for-15.05) branch is setup?  I will eventually get around to rebuilding that branch too, but I'm a little tight on disk space right now :|

mrdunk, 2015-08-25:
i just set a build environment set up and set it running.
this feed right? git://git.openwrt.org/15.05/openwrt.git
first time building from scratch so lets see how it goes.
i'm guessing it'll take a few hours so hope to get time to test it before i leave the house tomorrow.

dunk.

mrdunk, 2015-08-26:
i didn't manage to test that build yet.
should get to it this evening.

dunk.

mrdunk, 2015-08-26:
yup.
confirmed working.
Mosquitto on "OpenWrt Chaos Calmer 15.05" does not crash.
thanks for the quick turnaround!

dunk.

karlp, 2015-08-27:
Well, I didn't do anything there :)  It just means that it's probably musl libc fallout.  I'll still update the CC/for-15.05 branch to 1.43, it's just not as urgent as it was ealier :) Thanks for testing.

closed 2015-08-27 by karlp