openwrt/packages #1703: lsof 4.89: dsock.c:40:23: fatal error: sys/xattr.h: No such file or directory

stintel, 2015-08-23:
lsof 4.89 doesn't compile on octeon (uClibc):
```
make[3]: Entering directory '/home/stijn/Development/OpenWRT/openwrt-packages/utils/lsof'
make -C /home/stijn/Development/OpenWRT/git/build_dir/target-mips64_octeon_64_uClibc-0.9.33.2/lsof_4.89
make[4]: Entering directory '/home/stijn/Development/OpenWRT/git/build_dir/target-mips64_octeon_64_uClibc-0.9.33.2/lsof_4.89'
(cd lib; make DEBUG="-O" CFGF="-Os -pipe -mno-branch-likely -march=octeon -mabi=64 -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -msoft-float -Wformat -Werror=format-security -D_FORTIFY_SOURCE=2 -Wl,-z,now -Wl,-z,relro -DLINUXV=318000 -DGLIBCV=2 -DHASIPv6 -DHASIPv6 -DNEEDS_NETINET_TCPH -DHASUXSOCKEPT -D_FILE_OFFSET_BITS=64 -D_LARGEFILE64_SOURCE -DLSOF_VSTR=\"3.18.0\"")
make[5]: Entering directory '/home/stijn/Development/OpenWRT/git/build_dir/target-mips64_octeon_64_uClibc-0.9.33.2/lsof_4.89/lib'
make[5]: Nothing to be done for 'all'.
make[5]: Leaving directory '/home/stijn/Development/OpenWRT/git/build_dir/target-mips64_octeon_64_uClibc-0.9.33.2/lsof_4.89/lib'
mips64-openwrt-linux-uclibc-gcc  -Os -pipe -mno-branch-likely -march=octeon -mabi=64 -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -msoft-float -Wformat -Werror=format-security -D_FORTIFY_SOURCE=2 -Wl,-z,now -Wl,-z,relro -DLINUXV=318000 -DGLIBCV=2 -DHASIPv6 -DHASIPv6 -DNEEDS_NETINET_TCPH -DHASUXSOCKEPT -D_FILE_OFFSET_BITS=64 -D_LARGEFILE64_SOURCE -DLSOF_VSTR=\"3.18.0\"    -O   -c -o dsock.o dsock.c
dsock.c:40:23: fatal error: sys/xattr.h: No such file or directory
compilation terminated.
<builtin>: recipe for target 'dsock.o' failed
make[4]: *** [dsock.o] Error 1
make[4]: Leaving directory '/home/stijn/Development/OpenWRT/git/build_dir/target-mips64_octeon_64_uClibc-0.9.33.2/lsof_4.89'
Makefile:68: recipe for target '/home/stijn/Development/OpenWRT/git/build_dir/target-mips64_octeon_64_uClibc-0.9.33.2/lsof_4.89/.built' failed
make[3]: *** [/home/stijn/Development/OpenWRT/git/build_dir/target-mips64_octeon_64_uClibc-0.9.33.2/lsof_4.89/.built] Error 2
make[3]: Leaving directory '/home/stijn/Development/OpenWRT/openwrt-packages/utils/lsof'
package/Makefile:191: recipe for target 'package/feeds/packages/lsof/compile' failed
make[2]: *** [package/feeds/packages/lsof/compile] Error 2
make[2]: Leaving directory '/home/stijn/Development/OpenWRT/git'
package/Makefile:188: recipe for target '/home/stijn/Development/OpenWRT/git/staging_dir/target-mips64_octeon_64_uClibc-0.9.33.2/stamp/.package_compile' failed
make[1]: *** [/home/stijn/Development/OpenWRT/git/staging_dir/target-mips64_octeon_64_uClibc-0.9.33.2/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/stijn/Development/OpenWRT/git'
/home/stijn/Development/OpenWRT/git/include/toplevel.mk:181: recipe for target 'world' failed
make: *** [world] Error 2
```

ryzhovau, 2015-09-21:
I surprised too. 
> sof: update to 4.89
- build from original tarball
- fix ipv6 support: enable if not completely disabled
- drop useless LSOF_INCLUDE override

Who said `LSOF_INCLUDE` is useless? Now `/usr/uncude/linux` is used from host system while detecting kernel capabilities.

jolouis, 2015-09-23:
Not sure how this worked before or if xattr is new in the latest version of lsof... but I was able to get mine to compile by doing the following:
1) Change "<sys/xattr.h>" to "<attr/xattr.h>" in dsock.c (I created a quick patch to do this)
2) Add dependency on libattr. (Update package Makefile to include +libattr in DEPENDS line )
3) I manually updated the Makefile in the build_dir to include "-lattr" in the CFGL line, but I don't think that's necessary if you do the other two steps first.

The issue basically stems from glibc confusion... apparently sys/xattr.h is included as a glibc header to remove the need to always relay on libattr, but if you're using uClibc then that doesn't happen hence the need for the extra library. That's what I found anyway, hopefully this info is of some help to others.

mstorchak, 2015-09-30:
@stintel , thank you for finding this,
I'm sorry for late response. I plan to update both attr and acl packages, and attr is going to change include directory. I'll make sure lsof works again and nothing else is broken.
As for this issue - I admit I didn't check non-musl builds. I'll do that next time, at least for uClibc.

mstorchak, 2015-10-25:
Update:
see https://lists.openwrt.org/pipermail/openwrt-devel/2015-October/036652.html and https://lists.openwrt.org/pipermail/openwrt-devel/2015-October/036667.html
Until this is applied I'm afraid I can't fix this issue. If you build firmware yourself, please try to enable xattr in uClibc config. I tried to compile lsof and other packages for octeon target - it worked (compile test passed).
Alternatively, if anyone is able to express dependency of `packages/utils/attr` on `!@USE_UCLIBC` and not introduce recursive dependency, this would help.

stintel, 2017-06-29:
Not seeing this issue in LEDE.

closed 2017-06-29 by stintel