openwrt/packages #1704: prefer GIT over HTTPS

teknoraver, 2015-08-23:
What about preferring GIT protocol over HTTPS?
This should reduce the download time on first build.

I have this path ready to push:
http://pastebin.com/k5hMMNm1

champtar, 2015-08-23:
what we really should do is have an ```@github``` like ```@SF```, and add a config switch
(git can be blocked, it's less likely for https)

btw you should send a pull request instead of a patch

teknoraver, 2015-08-24:
I like the @github idea, I will try to implement it

hnyman, 2015-08-24:
Just a hint: adding a download alias will require modifying both include/download.mk and scripts/download.pl. Last addition of a download alias was my `@SAVANNAH` by r42840:
https://dev.openwrt.org/changeset/42840/
The changeset shows both the creation of a download alias and applying that to a few packages.

if `@GITHUB` is defined, it might define a git address as the first attempt and an https address try as the fallback address.

teknoraver, 2015-08-24:
I was thinking to use git:// and https:// as fallback
I don't know if the right think to do is adding a target in scripts/download.pl, which would add some git logic to the script (actually it does always wget) or just adding it in include/download.mk without invoking the script

hnyman, 2015-08-24:
But I am not quite that sure if preferring "git clone" would be beneficial with github sources. Openwrt download system will in any case create a .tar.bz2 archive out of the sources download by git, so downloading an archive file from github with https might be more efficient than downloadig with git and then creating archive.
See https://dev.openwrt.org/browser/trunk/include/download.mk#L84

teknoraver, 2015-08-24:
so you mean to download directly something like https://github.com/user/repo/archive/master.zip ?

hnyman, 2015-08-24:
Almost. The "releases" like https://github.com/openwrt/packages/archive/v14.07.tar.gz would be most straightforward. But while many packages define a release commit, others specify an arbitrary commit hash between official releases, and then the archive method would not work.

I haven't really verified how the clone from github is exactly happening (what actually happens when you download/clone from github using git: or https: address while specifying git as the source protocol ... Is the traffic amount smaller with git: than with https: address etc.).

Based on your original patches, most of the openwrt github packages define "git" as the source protocol, even when they use an https clone address. So they may use the git to clone the contents anyway. Github itself offers the https address as the clone address, so is there any benefit in specifying git: address?

In any case, a `@GITHUB` alias might be beneficial in future, as more and more packages are hosted there. The download/clone mechanisms offered by github may change and a central alias might help maintaining Openwrt packages.

Ps. the download script has git logic, like I highted in my previous message.

teknoraver, 2015-08-24:
for download script I mean download.pl, it has no git logic inside

hnyman, 2015-08-24:
You are right, that script does not have git logic, as it gets called from download.mk when the download is a generic one instead of a defined special protocol like git/svn/cvs/hg/...
( https://dev.openwrt.org/browser/trunk/include/download.mk#L44 )

Unlike most of the "single address" download locations, Github itself offers redundancy, so the main thing regarding githb is probably defining the most efficient download method/address.

so far all @ aliases have been defined to use that generic wget download method.

If `@github` would use git:, it should probably be defined at the git line at https://dev.openwrt.org/browser/trunk/include/download.mk#L15 , so that download.pl does not get invoked and git download method gets used. 

But if it would use https:, it should be defined at the generic line 14.

Apparently the current logic does not easily support logic like "first try git, then fallback to https".

teknoraver, 2015-08-24:
Well, it seems that a single snapshot can be downloaded with an url like:

https://github.com/teknoraver/benchs/archive/562321f8a71d2bed55f3acd77214114cc04d5e23.tar.gz

so we have fast http single file download and already in archive format

luizluca, 2015-08-24:
Hello,

Http is better when you are behind a firewall that only allows internet
connection through an HTTP proxy. As this is my case, I'll not like this
patch. I have to manually patch packages (like mtd) changing git to https
in order to get a complete built.

Regards,

plntyk, 2015-08-24:
With downloaded file names like 

```
v14.07.tar.gz 
```
downloads might be overwritten and should not be archived as such.

files like
```
562321f8a71d2bed55f3acd77214114cc04d5e23.tar.gz
```
should be renamed to saner defaults (like current git checkouts are)

Managing a local download mirror is easier (if you need space you can easier guess the source files/package relation).

teknoraver, 2015-08-24:
it seems that wgetting the tarball is the fastest option overall:
```
matteo@raver:/tmp$ time git clone https://github.com/openwrt/luci
Cloning into 'luci'...
remote: Counting objects: 83143, done.
remote: Compressing objects: 100% (47/47), done.
remote: Total 83143 (delta 14), reused 0 (delta 0), pack-reused 83095
Ricezione degli oggetti: 100% (83143/83143), 19.76 MiB | 1.59 MiB/s, done.
Risoluzione dei delta: 100% (49095/49095), done.
Checking connectivity... fatto.

real    0m16.427s
user    0m2.880s
sys     0m1.108s
matteo@raver:/tmp$ time git clone git://github.com/openwrt/luci
Cloning into 'luci'...
remote: Counting objects: 83143, done.
remote: Compressing objects: 100% (47/47), done.
remote: Total 83143 (delta 14), reused 0 (delta 0), pack-reused 83095
Ricezione degli oggetti: 100% (83143/83143), 19.76 MiB | 1.44 MiB/s, done.
Risoluzione dei delta: 100% (49095/49095), done.
Checking connectivity... fatto.

real    0m15.270s
user    0m2.752s
sys     0m0.900s
matteo@raver:/tmp$ time wget https://codeload.github.com/openwrt/luci/tar.gz/master
--2015-08-24 22:29:44--  https://codeload.github.com/openwrt/luci/tar.gz/master
Risoluzione di codeload.github.com (codeload.github.com)... 192.30.252.144
Connessione a codeload.github.com (codeload.github.com)|192.30.252.144|:443... connesso.
Richiesta HTTP inviata, in attesa di risposta... 200 OK
Lunghezza: 3448919 (3,3M) [application/x-gzip]
Salvataggio in: "master"

master                                          100%[=================>]   3,29M  1,28MB/s   in 2,6s   

2015-08-24 22:29:47 (1,28 MB/s) - "master" salvato [3448919/3448919]

real    0m3.398s
user    0m0.040s
sys     0m0.008s
```

champtar, 2015-08-24:
@teknoraver, don't forget that we need to be able to verify what we have downloaded.
With other http download we have MD5SUM, with git we only have git hash which is enough IF we use git.

teknoraver, 2015-08-24:
Here there is a proof of concept, I have adder @GITHUB to download.pl which gets the archive in HTTP mode and it's very fast, and I've converted two packages as example: mt76 and odhcp6c

actually we don't check md5 for github downloads too, the hash "unknown" is passed to download.pl

lynxis, 2015-09-23:
git:// downloads can easily blocked or tampered, because one hash can exist with different history and root commit. I'm heavily against prefering git:// over https:// because of security.
It's clear why wget is the fastest, the tarball doesn't contain any history.

yousong, 2016-08-19:
The `@GITHUB` change needs to be done in the build system, better if we bring this up with a patch in the mailing list.

Previously I also tried to forge a patch to download archive tarball with `@GITHUB_ARCHIVE` since `@GITHUB` was in use for fetching raw content.  Github actually provided a API for this `https://api.github.com/repos/<user>/<repo>/tarball/<ref>`, but I keep it on hold mainly because of two issues

 1. There is no guarantee from github that md5sums of the same file downloaded across difference HTTP request are to be identical.  This was confirmed after querying github for clarification of the issue.
 2. The directory name inside the tarball is kind of "undefined".  We need to repack it just like what we are doing after fetching with git

thess, 2016-09-08:
Not all git repos support git: access - Ex: git.xiph.org

diizzyy, 2017-06-07:
@thess @hnyman 
Close this, we want to be as "friendly" as possible to corporate users (ie firewalls and what not) so HTTPS is to be preferred over GIT.

closed 2017-06-07 by yousong