openwrt/packages #1707: tmux compile error

taeasy, 2015-08-24:
mv -f $depbase.Tpo $depbase.Po
depbase=`echo cmd-paste-buffer.o | sed 's|[^/]*$|.deps/&|;s|\.o$||'`;\
	mipsel-openwrt-linux-uclibc-gcc -DPACKAGE_NAME=\"tmux\" -DPACKAGE_TARNAME=\"tmux\" -DPACKAGE_VERSION=\"1.9a\" -DPACKAGE_STRING=\"tmux\ 1.9a\" -DPACKAGE_BUGREPORT=\"\" -DPACKAGE_URL=\"\" -DPACKAGE=\"tmux\" -DVERSION=\"1.9a\" -DSTDC_HEADERS=1 -DHAVE_SYS_TYPES_H=1 -DHAVE_SYS_STAT_H=1 -DHAVE_STDLIB_H=1 -DHAVE_STRING_H=1 -DHAVE_MEMORY_H=1 -DHAVE_STRINGS_H=1 -DHAVE_INTTYPES_H=1 -DHAVE_STDINT_H=1 -DHAVE_UNISTD_H=1 -DHAVE_CURSES_H=1 -DHAVE_DIRENT_H=1 -DHAVE_FCNTL_H=1 -DHAVE_INTTYPES_H=1 -DHAVE_NCURSES_H=1 -DHAVE_PATHS_H=1 -DHAVE_PTY_H=1 -DHAVE_STDINT_H=1 -DHAVE_SYS_DIR_H=1 -DHAVE_TERM_H=1 -DHAVE_FORKPTY=1 -DHAVE_DAEMON=1 -DHAVE_SETENV=1 -DHAVE_STRLCPY=1 -DHAVE_STRLCAT=1 -DHAVE_ASPRINTF=1 -DHAVE_STRCASESTR=1 -DHAVE_STRSEP=1 -DHAVE_CFMAKERAW=1 -DHAVE_OPENAT=1 -DHAVE_DECL_OPTARG=1 -DHAVE_DECL_OPTIND=1 -DHAVE_DECL_OPTRESET=0 -DHAVE_BZERO=1 -DHAVE_DIRFD=1 -DHAVE_SYSCONF=1 -DHAVE_CFMAKERAW=1 -DHAVE_BSD_TYPES=1 -DHAVE___PROGNAME=1 -DHAVE_PROC_PID=1 -I.   -I/home/main/brcm/staging_dir/target-mipsel_74kc+dsp2_uClibc-0.9.33.2/usr/include  -I/home/main/brcm/staging_dir/target-mipsel_74kc+dsp2_uClibc-0.9.33.2/usr/include -I/home/main/brcm/staging_dir/target-mipsel_74kc+dsp2_uClibc-0.9.33.2/include -I/home/main/brcm/staging_dir/toolchain-mipsel_74kc+dsp2_gcc-4.8-linaro_uClibc-0.9.33.2/usr/include -I/home/main/brcm/staging_dir/toolchain-mipsel_74kc+dsp2_gcc-4.8-linaro_uClibc-0.9.33.2/include -DTMUX_CONF="\"/etc/tmux.conf\""  -iquote.     -Os -pipe -mno-branch-likely -mips32r2 -mtune=74kc -mdspr2 -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -msoft-float -mips16 -minterlink-mips16  -D_GNU_SOURCE -std=gnu99  -O2   -MT cmd-paste-buffer.o -MD -MP -MF $depbase.Tpo -c -o cmd-paste-buffer.o cmd-paste-buffer.c &&\
	mv -f $depbase.Tpo $depbase.Po
depbase=`echo cmd-pipe-pane.o | sed 's|[^/]*$|.deps/&|;s|\.o$||'`;\
	mipsel-openwrt-linux-uclibc-gcc -DPACKAGE_NAME=\"tmux\" -DPACKAGE_TARNAME=\"tmux\" -DPACKAGE_VERSION=\"1.9a\" -DPACKAGE_STRING=\"tmux\ 1.9a\" -DPACKAGE_BUGREPORT=\"\" -DPACKAGE_URL=\"\" -DPACKAGE=\"tmux\" -DVERSION=\"1.9a\" -DSTDC_HEADERS=1 -DHAVE_SYS_TYPES_H=1 -DHAVE_SYS_STAT_H=1 -DHAVE_STDLIB_H=1 -DHAVE_STRING_H=1 -DHAVE_MEMORY_H=1 -DHAVE_STRINGS_H=1 -DHAVE_INTTYPES_H=1 -DHAVE_STDINT_H=1 -DHAVE_UNISTD_H=1 -DHAVE_CURSES_H=1 -DHAVE_DIRENT_H=1 -DHAVE_FCNTL_H=1 -DHAVE_INTTYPES_H=1 -DHAVE_NCURSES_H=1 -DHAVE_PATHS_H=1 -DHAVE_PTY_H=1 -DHAVE_STDINT_H=1 -DHAVE_SYS_DIR_H=1 -DHAVE_TERM_H=1 -DHAVE_FORKPTY=1 -DHAVE_DAEMON=1 -DHAVE_SETENV=1 -DHAVE_STRLCPY=1 -DHAVE_STRLCAT=1 -DHAVE_ASPRINTF=1 -DHAVE_STRCASESTR=1 -DHAVE_STRSEP=1 -DHAVE_CFMAKERAW=1 -DHAVE_OPENAT=1 -DHAVE_DECL_OPTARG=1 -DHAVE_DECL_OPTIND=1 -DHAVE_DECL_OPTRESET=0 -DHAVE_BZERO=1 -DHAVE_DIRFD=1 -DHAVE_SYSCONF=1 -DHAVE_CFMAKERAW=1 -DHAVE_BSD_TYPES=1 -DHAVE___PROGNAME=1 -DHAVE_PROC_PID=1 -I.   -I/home/main/brcm/staging_dir/target-mipsel_74kc+dsp2_uClibc-0.9.33.2/usr/include  -I/home/main/brcm/staging_dir/target-mipsel_74kc+dsp2_uClibc-0.9.33.2/usr/include -I/home/main/brcm/staging_dir/target-mipsel_74kc+dsp2_uClibc-0.9.33.2/include -I/home/main/brcm/staging_dir/toolchain-mipsel_74kc+dsp2_gcc-4.8-linaro_uClibc-0.9.33.2/usr/include -I/home/main/brcm/staging_dir/toolchain-mipsel_74kc+dsp2_gcc-4.8-linaro_uClibc-0.9.33.2/include -DTMUX_CONF="\"/etc/tmux.conf\""  -iquote.     -Os -pipe -mno-branch-likely -mips32r2 -mtune=74kc -mdspr2 -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -msoft-float -mips16 -minterlink-mips16  -D_GNU_SOURCE -std=gnu99  -O2   -MT cmd-pipe-pane.o -MD -MP -MF $depbase.Tpo -c -o cmd-pipe-pane.o cmd-pipe-pane.c &&\
	mv -f $depbase.Tpo $depbase.Po
cmd-pipe-pane.c: In function 'cmd_pipe_pane_exec':
cmd-pipe-pane.c:63:3: warning: implicit declaration of function 'bufferevent_free' [-Wimplicit-function-declaration]
   bufferevent_free(wp->pipe_event);
   ^
cmd-pipe-pane.c:121:3: warning: implicit declaration of function 'EVBUFFER_LENGTH' [-Wimplicit-function-declaration]
   wp->pipe_off = EVBUFFER_LENGTH(wp->event->input);
   ^
cmd-pipe-pane.c:121:43: error: dereferencing pointer to incomplete type
   wp->pipe_off = EVBUFFER_LENGTH(wp->event->input);
                                           ^
cmd-pipe-pane.c:123:3: warning: implicit declaration of function 'bufferevent_new' [-Wimplicit-function-declaration]
   wp->pipe_event = bufferevent_new(wp->pipe_fd,
   ^
cmd-pipe-pane.c:123:18: warning: assignment makes pointer from integer without a cast [enabled by default]
   wp->pipe_event = bufferevent_new(wp->pipe_fd,
                  ^
cmd-pipe-pane.c:125:3: warning: implicit declaration of function 'bufferevent_enable' [-Wimplicit-function-declaration]
   bufferevent_enable(wp->pipe_event, EV_WRITE);
   ^
make[4]: *** [cmd-pipe-pane.o] Error 1
make[4]: Leaving directory `/home/main/brcm/build_dir/target-mipsel_74kc+dsp2_uClibc-0.9.33.2/tmux-1.9a'
make[3]: *** [/home/main/brcm/build_dir/target-mipsel_74kc+dsp2_uClibc-0.9.33.2/tmux-1.9a/.built] Error 2
make[3]: Leaving directory `/home/main/brcm/feeds/packages/utils/tmux'
make[2]: *** [package/feeds/packages/tmux/compile] Error 2
make[2]: Leaving directory `/home/main/brcm'
make[1]: *** [/home/main/brcm/staging_dir/target-mipsel_74kc+dsp2_uClibc-0.9.33.2/stamp/.package_compile] Error 2
make[1]: Leaving directory `/home/main/brcm'
make: *** [world] Error 2
main@debian:~/brcm$

mstorchak, 2015-08-25:
As I understand it's related to #1542.

yousong, 2016-09-08:
Looks like the package now compiles fine in trunk (no faillogs present at LEDE buildbot)

closed 2016-09-08 by yousong