openwrt/packages #1709: Python PIP not working in recent trunk builds of openwrt

alirz1, 2015-08-25:
In the latest trunk builds of openwrt Python Pip and other related packages install fine but when running pip, it gives PIP not found error. Everything was fine until CC trunk 46687 from 
root@OpenWrt:/home/ali# pip
ash: pip: not found
root@OpenWrt:/home/ali# which pip
/usr/bin/pip

Seems like it might have something to do with the newer pip version 7.1.2. The previous working CC trunk build (46687) contains pip 7.1.0 and i have no problems with that build.

alirz1, 2015-08-25:
To add further. On my working build if i try to upgrade pip to 7.1.2 it does the following:

root@OpenWrt:/home/ali# pip install --upgrade pip
You are using pip version 7.1.0, however version 7.1.2 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.
Collecting pip
  Downloading pip-7.1.2-py2.py3-none-any.whl (1.1MB)
    100% |â–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆ| 1.1MB 39kB/s 
Installing collected packages: pip
  Found existing installation: pip 7.1.0
    Uninstalling pip-7.1.0:
Exception:
Traceback (most recent call last):
  File "/usr/lib/python2.7/site-packages/pip-7.1.0-py2.7.egg/pip/basecommand.py", line 223, in main
    status = self.run(options, args)
  File "/usr/lib/python2.7/site-packages/pip-7.1.0-py2.7.egg/pip/commands/install.py", line 299, in run
    root=options.root_path,
  File "/usr/lib/python2.7/site-packages/pip-7.1.0-py2.7.egg/pip/req/req_set.py", line 640, in install
    requirement.uninstall(auto_confirm=True)
  File "/usr/lib/python2.7/site-packages/pip-7.1.0-py2.7.egg/pip/req/req_install.py", line 726, in uninstall
    paths_to_remove.remove(auto_confirm)
  File "/usr/lib/python2.7/site-packages/pip-7.1.0-py2.7.egg/pip/req/req_uninstall.py", line 125, in remove
    renames(path, new_path)
  File "/usr/lib/python2.7/site-packages/pip-7.1.0-py2.7.egg/pip/utils/__init__.py", line 314, in renames
    shutil.move(old, new)
  File "/usr/lib/python2.7/shutil.py", line 300, in move
    rmtree(src)
  File "/usr/lib/python2.7/shutil.py", line 247, in rmtree
    rmtree(fullname, ignore_errors, onerror)
  File "/usr/lib/python2.7/shutil.py", line 256, in rmtree
    onerror(os.rmdir, path, sys.exc_info())
  File "/usr/lib/python2.7/shutil.py", line 254, in rmtree
    os.rmdir(path)
OSError: [Errno 22] Invalid argument: '/usr/lib/python2.7/site-packages/pip-7.1.0-py2.7.egg/EGG-INFO'

leitec, 2015-08-26:
It's using the build path for the interpreter, not the installed path. E.g. on my system:

```
root@wrt1900ac:/# head -n 1 /usr/bin/pip
#!/vol/dev/openwrt/trunk/staging_dir/host/bin/python2
root@wrt1900ac:/# 
```

You should be able to still do `python2 /usr/bin/pip` instead.

alirz1, 2015-08-26:
yeh on my router it shows
root@OpenWrt:/home/ali# head -n 1 /usr/bin/pip
#!/media/ali/2a31be03-830e-4b99-ac3f-10a60ac8d2d0/openwrt/staging_dir/host/bin/python2

That path is indeed the build path of my linux machine where i compiled the openwrt firmware.

I'm assuming this will be fixed in a future release or something?
Can i fix it manually some where?

leitec, 2015-08-26:
The manual workaround for the time being is to edit /usr/bin/pip on the router and change the first line to `#!/usr/bin/python2`

I'm not familiar with the Python build system so I don't know what the proper build-time fix is.

commodo, 2015-08-27:
Will look at this in 2-3 days.
Atm I'm on vacation with reduced internet acces.

<div>-------- Original message --------</div><div>From: Claudio Leite <notifications@github.com> </div><div>Date:26/08/2015  16:20  (GMT+02:00) </div><div>To: openwrt/packages <packages@noreply.github.com> </div><div>Subject: Re: [packages] Python PIP not working in recent trunk builds of  openwrt (#1709) </div><div>
</div>The manual workaround for the time being is to edit /usr/bin/pip on the router and change the first line to #!/usr/bin/python2

I'm not familiar with the Python build system so I don't know what the proper build-time fix is.

commodo, 2015-08-31:
Found a fix for this.
I'll re-test once it's merged.

commodo, 2015-09-14:
@alirz1 @leitec 
is this still an issue ?

alirz1, 2015-09-14:
I believe this it is working now. I had to reinstall an app using pip yesterday and it installed fine.

The patch in /usr/bin/pip seems to be correct now

root@OpenWrt:/usr/bin# head -n 1 /usr/bin/pip
#!/usr/bin/python2.7

closed 2015-09-14 by alirz1

leitec, 2015-09-14:
Fixed for me as well. Thanks!