openwrt/packages #1711: Some packages fail to download when installing "flexget" using Python-pip

alirz1, 2015-08-25:
On my Openwrt based router i've noticed an issue in the recent trunk builds of Chaos Calmer openwrt.
im install "flexget" by running command : pip install --install-option="--prefix=/mnt/sda1/software" flexget

This has always worked without any issue uptil about two weeks now. As of few days ago some packages fail to download duing the install. I CAN however download those packages manually from the same URL that is in the error message
root@OpenWrt:/home/ali# pip install --install-option="--prefix=/mnt/sda1/software" flexget
You are using pip version 7.1.0, however version 7.1.2 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.
/usr/lib/python2.7/site-packages/pip-7.1.0-py2.7.egg/pip/commands/install.py:198: UserWarning: Disabling all use of wheels due to the use of --build-options / --global-options / --install-options.
  cmdoptions.check_install_build_global(options)
Collecting flexget
  Downloading FlexGet-1.2.346.tar.gz (894kB)
    100% |â–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆ| 897kB 47kB/s 
Collecting FeedParser>=5.1.3 (from flexget)
  Downloading feedparser-5.2.1.tar.bz2 (192kB)
    100% |â–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆ| 192kB 198kB/s 
Collecting SQLAlchemy!=0.9.0,<1.999,>=0.7.5 (from flexget)
  Downloading SQLAlchemy-1.0.8.tar.gz (4.6MB)
    100% |â–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆ| 4.6MB 9.2kB/s 
Collecting PyYAML (from flexget)
  Downloading PyYAML-3.11.tar.gz (248kB)
    100% |â–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆ| 249kB 103kB/s 
Collecting beautifulsoup4!=4.2.0,<4.4,>=4.1 (from flexget)
  Downloading beautifulsoup4-4.3.2.tar.gz (143kB)
    100% |â–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆ| 143kB 251kB/s 
Collecting html5lib>=0.11 (from flexget)
  Downloading html5lib-0.999999.tar.gz (889kB)
    100% |â–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆ| 892kB 32kB/s 
Collecting PyRSS2Gen (from flexget)
  Downloading PyRSS2Gen-1.1.tar.gz
Collecting pynzb (from flexget)
  Downloading pynzb-0.1.0.tar.gz
Collecting progressbar (from flexget)
  Downloading progressbar-2.3.tar.gz
Collecting rpyc (from flexget)
  Downloading rpyc-3.3.0.tar.gz (53kB)
    100% |â–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆ| 57kB 348kB/s 
Collecting jinja2 (from flexget)
  Downloading Jinja2-2.8.tar.gz (357kB)
    100% |â–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆ| 360kB 99kB/s 
Collecting requests!=2.4.0,<2.99,>=1.0 (from flexget)
  Downloading requests-2.7.0.tar.gz (451kB)
    100% |â–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆ| 454kB 93kB/s 
Collecting python-dateutil!=2.0,!=2.2 (from flexget)
  Downloading python-dateutil-2.4.2.tar.gz (209kB)
    100% |â–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆ| 212kB 183kB/s 
Collecting jsonschema>=2.0 (from flexget)
  Downloading jsonschema-2.5.1.tar.gz (50kB)
    100% |â–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆ| 53kB 514kB/s 
    Complete output from command python setup.py egg_info:
    Download error on https://pypi.python.org/simple/vcversioner/: [SSL: CERTIFICATE_VERIFY_FAILED] unknown error (_ssl.c:590) -- Some packages may not be found!
    Couldn't find index page for 'vcversioner' (maybe misspelled?)
    Download error on https://pypi.python.org/simple/: [SSL: CERTIFICATE_VERIFY_FAILED] unknown error (_ssl.c:590) -- Some packages may not be found!
    No local packages or download links found for vcversioner
    Traceback (most recent call last):
      File "<string>", line 20, in <module>
      File "/tmp/pip-build-ufRznR/jsonschema/setup.py", line 44, in <module>
        vcversioner={"version_module_paths" : ["jsonschema/_version.py"]},
      File "/usr/lib/python2.7/distutils/core.py", line 111, in setup
        _setup_distribution = dist = klass(attrs)
      File "/usr/lib/python2.7/site-packages/setuptools-18.0.1-py2.7.egg/setuptools/dist.py", line 267, in __init__
      File "/usr/lib/python2.7/site-packages/setuptools-18.0.1-py2.7.egg/setuptools/dist.py", line 312, in fetch_build_eggs
      File "/usr/lib/python2.7/site-packages/setuptools-18.0.1-py2.7.egg/pkg_resources/__init__.py", line 836, in resolve
      File "/usr/lib/python2.7/site-packages/setuptools-18.0.1-py2.7.egg/pkg_resources/__init__.py", line 1081, in best_match
      File "/usr/lib/python2.7/site-packages/setuptools-18.0.1-py2.7.egg/pkg_resources/__init__.py", line 1093, in obtain
      File "/usr/lib/python2.7/site-packages/setuptools-18.0.1-py2.7.egg/setuptools/dist.py", line 378, in fetch_build_egg
      File "/usr/lib/python2.7/site-packages/setuptools-18.0.1-py2.7.egg/setuptools/command/easy_install.py", line 623, in easy_install
    distutils.errors.DistutilsError: Could not find suitable distribution for Requirement.parse('vcversioner')
    
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-ufRznR/jsonschema

alirz1, 2015-08-25:
The following 4 packages are currently failing to download automatically

vcversioner
setuptools_scm
stevedore
pbr

alirz1, 2015-08-26:
no one?

commodo, 2015-08-29:
back from vacation
so, it looks like these packages failed to install due to some SSL certificate verification errors
no idea why; will check

maybe you don't have openssl installed on your system ?

alirz1, 2015-08-29:
I do have the following openssl packages installed

root@OpenWrt:/home/# opkg list-installed |grep -i ssl
libopenssl - 1.0.2d-1
openssl-util - 1.0.2d-1
python-openssl - 2.7.10-6

alirz1, 2015-08-30:
Is it possible that WGET in openwrt needs the certs to be installed from https://pypi.python.org/ ?

commodo, 2015-09-02:
maybe you need to manually specify the cert location in your case;

if i do
```
pip install flexget
```
it seems to work fine; tho, i am testing this on VM at the moment

maybe you could try
```
pip --cert=/usr/lib/python2.7/site-packages/pip/_vendor/requests/cacert.pem --install-option="--prefix=/mnt/sda1/software" flexget
```
maybe the install option you're adding, requires that the cert be manually specified
not sure on this yet

the idea came from here : http://stackoverflow.com/questions/19377045/pip-cert-failed-but-curl-works

alirz1, 2015-09-02:
Hi,

The install option as i had mentioned above, had always worked in the past. I only started seeing this problem in the last 2 months or so. I've never had to specify a cert manually.I didnt even know how, till you gave the example above.
What platform is your VM? Could it be something specific to openwrt?

commodo, 2015-09-02:
i usually use an openwrt x86 VM to test stuff
in the last 2 months is when i've upgraded pip and setuptools

maybe it's a python-pip bug;
looks like the guys that make python-setuptools and python-pip have been getting a bit sloppier (lately) with regard to cross-compiling these tools
well, to be fair, so have i :)

i'll dig deeper in the pip code in the next couple of days

commodo, 2015-09-02:
Ah, and to answer the question: "Could it be something specific to openwrt?"
Not sure about that.
Packages for OpenWRT are cross-compiled/cross-built (in general), so that's an open window to other stuff that haven't been tested by the packages creators.

alirz1, 2015-09-02:
hmmm openwrt x86 vm :)  ooo that sounds something i would like to spin up.. where did you get that from?

commodo, 2015-09-02:
well, i usually just build mine from source (to get the latest trunk)
but you can also find a few images here:
https://downloads.openwrt.org/chaos_calmer/15.05-rc3/x86/
specifically:
https://downloads.openwrt.org/chaos_calmer/15.05-rc3/x86/generic/

i run them on Linux via virt-manager with qemu-kvm; but you can also run this in virtualbox/vmware (not sure how to mount the disk image in these yet)
if you really want x86_64, you should make sure that your vm emulator/host mounts your hdd via SATA emulation (SCSI and IDE don't work, unless you specifically compile them)
and for eth interfaces, make sure you're emulating e1000 interfaces, and make sure to add 2 (eth0 == lan, eth1 == wan)

alirz1, 2015-09-12:
this is still happening. I tried today:

 100% |â–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆâ–ˆ| 53kB 485kB/s 
    Complete output from command python setup.py egg_info:
    Download error on https://pypi.python.org/simple/vcversioner/: [SSL: CERTIFICATE_VERIFY_FAILED] unknown error (_ssl.c:590) -- Some packages may not be found!
    Couldn't find index page for 'vcversioner' (maybe misspelled?)
    Download error on https://pypi.python.org/simple/: [SSL: CERTIFICATE_VERIFY_FAILED] unknown error (_ssl.c:590) -- Some packages may not be found!
    No local packages or download links found for vcversioner
    Traceback (most recent call last):
      File "<string>", line 20, in <module>
      File "/tmp/pip-build-6FxBG7/jsonschema/setup.py", line 44, in <module>
        vcversioner={"version_module_paths" : ["jsonschema/_version.py"]},
      File "/usr/lib/python2.7/distutils/core.py", line 111, in setup
        _setup_distribution = dist = klass(attrs)
      File "/usr/lib/python2.7/site-packages/setuptools-18.0.1-py2.7.egg/setuptools/dist.py", line 267, in __init__
      File "/usr/lib/python2.7/site-packages/setuptools-18.0.1-py2.7.egg/setuptools/dist.py", line 312, in fetch_build_eggs
      File "/usr/lib/python2.7/site-packages/setuptools-18.0.1-py2.7.egg/pkg_resources/__init__.py", line 836, in resolve
      File "/usr/lib/python2.7/site-packages/setuptools-18.0.1-py2.7.egg/pkg_resources/__init__.py", line 1081, in best_match
      File "/usr/lib/python2.7/site-packages/setuptools-18.0.1-py2.7.egg/pkg_resources/__init__.py", line 1093, in obtain
      File "/usr/lib/python2.7/site-packages/setuptools-18.0.1-py2.7.egg/setuptools/dist.py", line 378, in fetch_build_egg
      File "/usr/lib/python2.7/site-packages/setuptools-18.0.1-py2.7.egg/setuptools/command/easy_install.py", line 623, in easy_install
    distutils.errors.DistutilsError: Could not find suitable distribution for Requirement.parse('vcversioner')
    
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-6FxBG7/jsonschema

commodo, 2015-09-12:
Ah, I forgot to reply earlier.
I got caught up with other issues.

I've managed to reproduce this.
Will look into it further.

alirz1, 2015-10-06:
Is this now fixed? Should it be closed?

alirz1, 2015-12-01:
still having the same problem.

alirz1, 2015-12-01:
worked around by installed ssl cert on openwrt...

closed 2015-12-01 by alirz1

commodo, 2015-12-01:
could you tell me where you downloaded the certs ?
maybe we could install them via  `opkg install python-pip`
i think this was one thing I wanted to try a while back

alirz1, 2015-12-01:
https://wiki.openwrt.org/doc/howto/wget-ssl-certs

Actually all i did was install the "ca-certificates" package in openwrt. While i also did steps 1-3 at the above link, i think installing the package might have been the only required step. 
I didnt do ny other steps on that wiki page there.

After that, pip didnt complain about not being able to download the packages that were failing yesterday.

commodo, 2015-12-01:
So, now I feel a bit stupid, since this had a simple fix.
Well, at least I learned something (about the ca-certificates package).

Thanks

alirz1, 2015-12-02:
Hey a suggestion.. Dont know if i should open a issue for this.
So while i was looking into the whole certificate issue and all. I came across another thing.

In the usual distros, PIP has a .conf file. Where you can for e.g specify the cache durectory.
In openwrt, i couldnt find the config file for PIP. So, when im installing some python stuf, pip stores the cache by default in "/root.cache". That path has limited storage available and i was always running out of space in the router.
Suggestion 1: move cache folder to /tmp which seems to be have the most space available for new routers.
Suggestions 2: Provide a way to define a pip.conf file where a user can specify the path to use as cache folder.

commodo, 2015-12-02:
fine by me;
and makes sense
will look into it

commodo, 2015-12-03:
Will test this tomorrow: https://github.com/commodo/packages/commit/9b25d79aece2ec7473a03982b3a530f8d402328d

alirz1, 2015-12-03:
Awesome.. looking forward to test this out when i do a rebuild in the near future

commodo, 2015-12-06:
@alirz1 
there's one issue that occurs now with the --download-cache option
https://pip.pypa.io/en/stable/reference/pip_install/#caching

> DEPRECATION pip install --download-cache and pip wheel --download-cache command line flags have been deprecated and the functionality removed. Since pip now automatically configures and uses it's internal HTTP cache which supplants the --download-cache the existing options have been made non functional but will still be accepted until their removal in pip v8.0. For more information please see https://pip.pypa.io/en/stable/reference/pip_install.html#caching

I am getting a deprecation warning.
Something like:
```
   "--download-cache has been deprecated and will be removed in "
   "the future. Pip now automatically uses and configures its "
   "cache.", 
```

So, even if I add it now, it may be unused later.
Will check later if there are options to override, or if we should ?

So, we could the `/etc/pip.conf` as a template, with some options.
But, regarding the download-cache, I'll still have to see.

alirz1, 2015-12-06:
Well I'm not a developer and also don't know Python so my ideas might not very helpful.
Strange that they are removing the option to specify the cache location assuming that pip knows the best location to use. Whereas clearly in openwrts case it's a location with the lowest amount of storage.

What ever it's worth, I had gone around this space issue by sym linking ~/.cache/pip to a folder on my mounted USB drive. I could have symlinked it to /tmp also probably.

commodo, 2015-12-09:
https://github.com/openwrt/packages/pull/2089
Seems that `cache-dir` is the option you wanted.

alirz1, 2015-12-09:
@commodo 
Yup :)  i just realized that you had mentioned "--download-cache" which is indeed a different parameter.
thanks.