openwrt/packages #1712: forked-daapd compile failed

taeasy, 2015-08-26:
make[2]: Entering directory '/home/main/x86/feeds/packages/sound/forked-daapd'
CFLAGS="-Os -pipe -march=i486 -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -fpic  -I/home/main/x86/staging_dir/target-i386_i486_uClibc-0.9.33.2/usr/include -I/home/main/x86/staging_dir/target-i386_i486_uClibc-0.9.33.2/include -I/home/main/x86/staging_dir/toolchain-i386_i486_gcc-4.8-linaro_uClibc-0.9.33.2/usr/include -I/home/main/x86/staging_dir/toolchain-i386_i486_gcc-4.8-linaro_uClibc-0.9.33.2/include " CXXFLAGS="-Os -pipe -march=i486 -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -fpic  -I/home/main/x86/staging_dir/target-i386_i486_uClibc-0.9.33.2/usr/include -I/home/main/x86/staging_dir/target-i386_i486_uClibc-0.9.33.2/include -I/home/main/x86/staging_dir/toolchain-i386_i486_gcc-4.8-linaro_uClibc-0.9.33.2/usr/include -I/home/main/x86/staging_dir/toolchain-i386_i486_gcc-4.8-linaro_uClibc-0.9.33.2/include " LDFLAGS="-L/home/main/x86/staging_dir/target-i386_i486_uClibc-0.9.33.2/usr/lib -L/home/main/x86/staging_dir/target-i386_i486_uClibc-0.9.33.2/lib -L/home/main/x86/staging_dir/toolchain-i386_i486_gcc-4.8-linaro_uClibc-0.9.33.2/usr/lib -L/home/main/x86/staging_dir/toolchain-i386_i486_gcc-4.8-linaro_uClibc-0.9.33.2/lib -Wl,-rpath-link,/home/main/x86/staging_dir/target-i386_i486_uClibc-0.9.33.2/usr/lib " make -j1 -C /home/main/x86/build_dir/target-i386_i486_uClibc-0.9.33.2/forked-daapd-23.2/. AR="i486-openwrt-linux-uclibc-gcc-ar" AS="i486-openwrt-linux-uclibc-gcc -c -Os -pipe -march=i486 -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -fpic" LD=i486-openwrt-linux-uclibc-ld NM="i486-openwrt-linux-uclibc-gcc-nm" CC="i486-openwrt-linux-uclibc-gcc" GCC="i486-openwrt-linux-uclibc-gcc" CXX="i486-openwrt-linux-uclibc-g++" RANLIB="i486-openwrt-linux-uclibc-gcc-ranlib" STRIP=i486-openwrt-linux-uclibc-strip OBJCOPY=i486-openwrt-linux-uclibc-objcopy OBJDUMP=i486-openwrt-linux-uclibc-objdump SIZE=i486-openwrt-linux-uclibc-size CROSS="i486-openwrt-linux-uclibc-" ARCH="i386" ;
make[3]: Entering directory '/home/main/x86/build_dir/target-i386_i486_uClibc-0.9.33.2/forked-daapd-23.2'
make  all-recursive
make[4]: Entering directory '/home/main/x86/build_dir/target-i386_i486_uClibc-0.9.33.2/forked-daapd-23.2'
Making all in sqlext
make[5]: Entering directory '/home/main/x86/build_dir/target-i386_i486_uClibc-0.9.33.2/forked-daapd-23.2/sqlext'
make[5]: Nothing to be done for 'all'.
make[5]: Leaving directory '/home/main/x86/build_dir/target-i386_i486_uClibc-0.9.33.2/forked-daapd-23.2/sqlext'
Making all in src
make[5]: Entering directory '/home/main/x86/build_dir/target-i386_i486_uClibc-0.9.33.2/forked-daapd-23.2/src'
make  all-am
make[6]: Entering directory '/home/main/x86/build_dir/target-i386_i486_uClibc-0.9.33.2/forked-daapd-23.2/src'
i486-openwrt-linux-uclibc-gcc -DHAVE_CONFIG_H -I. -I..  -D_GNU_SOURCE -DDATADIR="\"/usr/share/forked-daapd\"" -DCONFDIR="\"/etc\"" -DSTATEDIR="\"/var\"" -DPKGLIBDIR="\"/usr/lib/forked-daapd\"" -I/home/main/x86/staging_dir/target-i386_i486_uClibc-0.9.33.2/usr/include -I/home/main/x86/staging_dir/target-i386_i486_uClibc-0.9.33.2/include -I/home/main/x86/staging_dir/toolchain-i386_i486_gcc-4.8-linaro_uClibc-0.9.33.2/usr/include -I/home/main/x86/staging_dir/toolchain-i386_i486_gcc-4.8-linaro_uClibc-0.9.33.2/include  -DITUNES -DLASTFM -DMPD -DALSA -I/home/main/x86/staging_dir/target-i386_i486_uClibc-0.9.33.2/usr/include  -D_REENTRANT -I/home/main/x86/staging_dir/target-i386_i486_uClibc-0.9.33.2/usr/include  -I/home/main/x86/staging_dir/target-i386_i486_uClibc-0.9.33.2/usr/include  -I/home/main/x86/staging_dir/target-i386_i486_uClibc-0.9.33.2/usr/include -I/home/main/x86/staging_dir/target-i386_i486_uClibc-0.9.33.2/usr/include   -D_THREAD_SAFE -D_REENTRANT -I/home/main/x86/staging_dir/target-i386_i486_uClibc-0.9.33.2/usr/include  -I/home/main/x86/staging_dir/target-i386_i486_uClibc-0.9.33.2/usr/include -I/home/main/x86/staging_dir/target-i386_i486_uClibc-0.9.33.2/usr/include/libxml2 -I/home/main/x86/staging_dir/target-i386_i486_uClibc-0.9.33.2/usr/include -I/home/main/x86/staging_dir/target-i386_i486_uClibc-0.9.33.2/usr/include -I/home/main/x86/staging_dir/target-i386_i486_uClibc-0.9.33.2/usr/include -I/home/main/x86/staging_dir/target-i386_i486_uClibc-0.9.33.2/usr/include/alsa -I/home/main/x86/staging_dir/target-i386_i486_uClibc-0.9.33.2/usr/include  -Os -pipe -march=i486 -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -fpic  -Wall -D_LARGEFILE_SOURCE -MT forked_daapd-main.o -MD -MP -MF .deps/forked_daapd-main.Tpo -c -o forked_daapd-main.o `test -f 'main.c' || echo './'`main.c
In file included from cache.h:6:0,
                 from main.c:61:
/home/main/x86/staging_dir/target-i386_i486_uClibc-0.9.33.2/usr/include/event2/event.h:677:5: error: conflicting types for 'event_base_loopexit'
 int event_base_loopexit(struct event_base *, const struct timeval *);
     ^
In file included from main.c:50:0:
/home/main/x86/staging_dir/target-i386_i486_uClibc-0.9.33.2/usr/include/event.h:164:5: note: previous declaration of 'event_base_loopexit' was here
 int event_base_loopexit (struct event_base *base, struct timeval *tv);
     ^
In file included from cache.h:6:0,
                 from main.c:61:
/home/main/x86/staging_dir/target-i386_i486_uClibc-0.9.33.2/usr/include/event2/event.h:913:5: error: conflicting types for 'event_base_once'
 int event_base_once(struct event_base *, evutil_socket_t, short, event_callback_fn, void *, const struct timeval *);
     ^
In file included from main.c:50:0:
/home/main/x86/staging_dir/target-i386_i486_uClibc-0.9.33.2/usr/include/event.h:166:5: note: previous declaration of 'event_base_once' was here
 int event_base_once (struct event_base *base, int fd, short events, void (*cb)(int, short, void *), void *arg, struct timeval *tv);
     ^
In file included from cache.h:6:0,
                 from main.c:61:
/home/main/x86/staging_dir/target-i386_i486_uClibc-0.9.33.2/usr/include/event2/event.h:937:5: error: conflicting types for 'event_add'
 int event_add(struct event *ev, const struct timeval *timeout);
     ^
In file included from main.c:50:0:
/home/main/x86/staging_dir/target-i386_i486_uClibc-0.9.33.2/usr/include/event.h:151:5: note: previous declaration of 'event_add' was here
 int event_add (struct event *ev, struct timeval *tv);
     ^
In file included from cache.h:6:0,
                 from main.c:61:
/home/main/x86/staging_dir/target-i386_i486_uClibc-0.9.33.2/usr/include/event2/event.h:982:5: error: conflicting types for 'event_pending'
 int event_pending(const struct event *ev, short events, struct timeval *tv);
     ^
In file included from main.c:50:0:
/home/main/x86/staging_dir/target-i386_i486_uClibc-0.9.33.2/usr/include/event.h:155:5: note: previous declaration of 'event_pending' was here
 int event_pending (struct event *ev, short, struct timeval *tv);
     ^
/home/main/x86/staging_dir/target-i386_i486_uClibc-0.9.33.2/usr/include/event2/event.h:1000:5: error: expected identifier or '(' before 'const'
 int event_initialized(const struct event *ev);
     ^
/home/main/x86/staging_dir/target-i386_i486_uClibc-0.9.33.2/usr/include/event2/event.h:1000:5: error: expected ')' before '->' token
 int event_initialized(const struct event *ev);
     ^
Makefile:693: recipe for target 'forked_daapd-main.o' failed
make[6]: *** [forked_daapd-main.o] Error 1
make[6]: Leaving directory '/home/main/x86/build_dir/target-i386_i486_uClibc-0.9.33.2/forked-daapd-23.2/src'
Makefile:512: recipe for target 'all' failed
make[5]: *** [all] Error 2
make[5]: Leaving directory '/home/main/x86/build_dir/target-i386_i486_uClibc-0.9.33.2/forked-daapd-23.2/src'
Makefile:536: recipe for target 'all-recursive' failed
make[4]: *** [all-recursive] Error 1
make[4]: Leaving directory '/home/main/x86/build_dir/target-i386_i486_uClibc-0.9.33.2/forked-daapd-23.2'
Makefile:404: recipe for target 'all' failed
make[3]: *** [all] Error 2
make[3]: Leaving directory '/home/main/x86/build_dir/target-i386_i486_uClibc-0.9.33.2/forked-daapd-23.2'
Makefile:72: recipe for target '/home/main/x86/build_dir/target-i386_i486_uClibc-0.9.33.2/forked-daapd-23.2/.built' failed
make[2]: *** [/home/main/x86/build_dir/target-i386_i486_uClibc-0.9.33.2/forked-daapd-23.2/.built] Error 2
make[2]: Leaving directory '/home/main/x86/feeds/packages/sound/forked-daapd'
package/Makefile:191: recipe for target 'package/feeds/packages/forked-daapd/compile' failed
make[1]: *** [package/feeds/packages/forked-daapd/compile] Error 2
make[1]: Leaving directory '/home/main/x86'
/home/main/x86/include/toplevel.mk:181: recipe for target 'package/forked-daapd/compile' failed
make: *** [package/forked-daapd/compile] Error 2
main@debian:~/x86$

thess, 2015-08-28:
Same problem as #1707 (and referenced by fix to #1542)

yousong, 2016-09-08:
Should already fixed by #460

closed 2016-09-08 by yousong