openwrt/packages #1716: python-psycopg2 build error

build000, 2015-08-26:
(...)
make[3]: Entering directory '/magazyn/Pliki/openwrt/trunk/feeds/packages/lang/python-psycopg2'
install -d -m0755 /magazyn/Pliki/openwrt/trunk/build_dir/target-mips_34kc_musl-1.1.10/psycopg2-2.6.1/ipkg-install//usr/lib/python2.7/site-packages
(	export PYTHONPATH="/magazyn/Pliki/openwrt/trunk/staging_dir/target-mips_34kc_musl-1.1.10/usr/lib/python2.7:/magazyn/Pliki/openwrt/trunk/staging_dir/target-mips_34kc_musl-1.1.10//usr/lib/python2.7/site-packages:/magazyn/Pliki/openwrt/trunk/build_dir/target-mips_34kc_musl-1.1.10/psycopg2-2.6.1/ipkg-install//usr/lib/python2.7/site-packages"; export PYTHONOPTIMIZE=""; export PYTHONDONTWRITEBYTECODE=1; export _python_sysroot="/magazyn/Pliki/openwrt/trunk/staging_dir/target-mips_34kc_musl-1.1.10"; export _python_prefix="/usr"; export _python_exec_prefix="/usr";  cd /magazyn/Pliki/openwrt/trunk/build_dir/target-mips_34kc_musl-1.1.10/psycopg2-2.6.1/.; CC="mips-openwrt-linux-musl-gcc" CCSHARED="mips-openwrt-linux-musl-gcc -fpic" LD="mips-openwrt-linux-musl-gcc" LDSHARED="mips-openwrt-linux-musl-gcc -shared" CFLAGS="-Os -pipe -mno-branch-likely -mips32r2 -mtune=34kc -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -msoft-float -mips16 -minterlink-mips16 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -mno-mips16 -mno-interlink-mips16" CPPFLAGS="-I/magazyn/Pliki/openwrt/trunk/staging_dir/target-mips_34kc_musl-1.1.10/usr/include -I/magazyn/Pliki/openwrt/trunk/staging_dir/target-mips_34kc_musl-1.1.10/include -I/magazyn/Pliki/openwrt/trunk/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_musl-1.1.10/usr/include -I/magazyn/Pliki/openwrt/trunk/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_musl-1.1.10/include/fortify -I/magazyn/Pliki/openwrt/trunk/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_musl-1.1.10/include -I/magazyn/Pliki/openwrt/trunk/staging_dir/target-mips_34kc_musl-1.1.10/usr/include/python2.7" LDFLAGS="-L/magazyn/Pliki/openwrt/trunk/staging_dir/target-mips_34kc_musl-1.1.10/usr/lib -L/magazyn/Pliki/openwrt/trunk/staging_dir/target-mips_34kc_musl-1.1.10/lib -L/magazyn/Pliki/openwrt/trunk/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_musl-1.1.10/usr/lib -L/magazyn/Pliki/openwrt/trunk/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_musl-1.1.10/lib -znow -zrelro -lpython2.7" _PYTHON_HOST_PLATFORM=linux2 __PYVENV_LAUNCHER__="/usr/bin/python2.7"   /magazyn/Pliki/openwrt/trunk/staging_dir/host/bin/python2  ./setup.py build_py ; )
running build_py
warning: build_py: byte-compiling is disabled, skipping.

find /magazyn/Pliki/openwrt/trunk/build_dir/target-mips_34kc_musl-1.1.10/psycopg2-2.6.1/ipkg-install -name "*\.pyc" -o -name "*\.pyo" | xargs rm -f
install -d -m0755 /magazyn/Pliki/openwrt/trunk/build_dir/target-mips_34kc_musl-1.1.10/psycopg2-2.6.1/ipkg-install//usr/lib/python2.7/site-packages
(	export PYTHONPATH="/magazyn/Pliki/openwrt/trunk/staging_dir/target-mips_34kc_musl-1.1.10/usr/lib/python2.7:/magazyn/Pliki/openwrt/trunk/staging_dir/target-mips_34kc_musl-1.1.10//usr/lib/python2.7/site-packages:/magazyn/Pliki/openwrt/trunk/build_dir/target-mips_34kc_musl-1.1.10/psycopg2-2.6.1/ipkg-install//usr/lib/python2.7/site-packages"; export PYTHONOPTIMIZE=""; export PYTHONDONTWRITEBYTECODE=1; export _python_sysroot="/magazyn/Pliki/openwrt/trunk/staging_dir/target-mips_34kc_musl-1.1.10"; export _python_prefix="/usr"; export _python_exec_prefix="/usr";  cd /magazyn/Pliki/openwrt/trunk/build_dir/target-mips_34kc_musl-1.1.10/psycopg2-2.6.1/; CC="mips-openwrt-linux-musl-gcc" CCSHARED="mips-openwrt-linux-musl-gcc -fpic" LD="mips-openwrt-linux-musl-gcc" LDSHARED="mips-openwrt-linux-musl-gcc -shared" CFLAGS="-Os -pipe -mno-branch-likely -mips32r2 -mtune=34kc -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -msoft-float -mips16 -minterlink-mips16 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -mno-mips16 -mno-interlink-mips16" CPPFLAGS="-I/magazyn/Pliki/openwrt/trunk/staging_dir/target-mips_34kc_musl-1.1.10/usr/include -I/magazyn/Pliki/openwrt/trunk/staging_dir/target-mips_34kc_musl-1.1.10/include -I/magazyn/Pliki/openwrt/trunk/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_musl-1.1.10/usr/include -I/magazyn/Pliki/openwrt/trunk/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_musl-1.1.10/include/fortify -I/magazyn/Pliki/openwrt/trunk/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_musl-1.1.10/include -I/magazyn/Pliki/openwrt/trunk/staging_dir/target-mips_34kc_musl-1.1.10/usr/include/python2.7" LDFLAGS="-L/magazyn/Pliki/openwrt/trunk/staging_dir/target-mips_34kc_musl-1.1.10/usr/lib -L/magazyn/Pliki/openwrt/trunk/staging_dir/target-mips_34kc_musl-1.1.10/lib -L/magazyn/Pliki/openwrt/trunk/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_musl-1.1.10/usr/lib -L/magazyn/Pliki/openwrt/trunk/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_musl-1.1.10/lib -znow -zrelro -lpython2.7" _PYTHON_HOST_PLATFORM=linux2 __PYVENV_LAUNCHER__="/usr/bin/python2.7"   /magazyn/Pliki/openwrt/trunk/staging_dir/host/bin/python2  ./setup.py build_ext --pg-config="/magazyn/Pliki/openwrt/trunk/staging_dir/target-mips_34kc_musl-1.1.10/usr/bin/pg_config" ; )
running build_ext
find /magazyn/Pliki/openwrt/trunk/build_dir/target-mips_34kc_musl-1.1.10/psycopg2-2.6.1/ipkg-install -name "*\.pyc" -o -name "*\.pyo" | xargs rm -f
install -d -m0755 /magazyn/Pliki/openwrt/trunk/build_dir/target-mips_34kc_musl-1.1.10/psycopg2-2.6.1/ipkg-install//usr/lib/python2.7/site-packages
(	export PYTHONPATH="/magazyn/Pliki/openwrt/trunk/staging_dir/target-mips_34kc_musl-1.1.10/usr/lib/python2.7:/magazyn/Pliki/openwrt/trunk/staging_dir/target-mips_34kc_musl-1.1.10//usr/lib/python2.7/site-packages:/magazyn/Pliki/openwrt/trunk/build_dir/target-mips_34kc_musl-1.1.10/psycopg2-2.6.1/ipkg-install//usr/lib/python2.7/site-packages"; export PYTHONOPTIMIZE=""; export PYTHONDONTWRITEBYTECODE=1; export _python_sysroot="/magazyn/Pliki/openwrt/trunk/staging_dir/target-mips_34kc_musl-1.1.10"; export _python_prefix="/usr"; export _python_exec_prefix="/usr";  cd /magazyn/Pliki/openwrt/trunk/build_dir/target-mips_34kc_musl-1.1.10/psycopg2-2.6.1/.; CC="mips-openwrt-linux-musl-gcc" CCSHARED="mips-openwrt-linux-musl-gcc -fpic" LD="mips-openwrt-linux-musl-gcc" LDSHARED="mips-openwrt-linux-musl-gcc -shared" CFLAGS="-Os -pipe -mno-branch-likely -mips32r2 -mtune=34kc -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -msoft-float -mips16 -minterlink-mips16 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -mno-mips16 -mno-interlink-mips16" CPPFLAGS="-I/magazyn/Pliki/openwrt/trunk/staging_dir/target-mips_34kc_musl-1.1.10/usr/include -I/magazyn/Pliki/openwrt/trunk/staging_dir/target-mips_34kc_musl-1.1.10/include -I/magazyn/Pliki/openwrt/trunk/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_musl-1.1.10/usr/include -I/magazyn/Pliki/openwrt/trunk/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_musl-1.1.10/include/fortify -I/magazyn/Pliki/openwrt/trunk/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_musl-1.1.10/include -I/magazyn/Pliki/openwrt/trunk/staging_dir/target-mips_34kc_musl-1.1.10/usr/include/python2.7" LDFLAGS="-L/magazyn/Pliki/openwrt/trunk/staging_dir/target-mips_34kc_musl-1.1.10/usr/lib -L/magazyn/Pliki/openwrt/trunk/staging_dir/target-mips_34kc_musl-1.1.10/lib -L/magazyn/Pliki/openwrt/trunk/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_musl-1.1.10/usr/lib -L/magazyn/Pliki/openwrt/trunk/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_musl-1.1.10/lib -znow -zrelro -lpython2.7" _PYTHON_HOST_PLATFORM=linux2 __PYVENV_LAUNCHER__="/usr/bin/python2.7"   /magazyn/Pliki/openwrt/trunk/staging_dir/host/bin/python2  ./setup.py install --prefix=/usr --root=/magazyn/Pliki/openwrt/trunk/build_dir/target-mips_34kc_musl-1.1.10/psycopg2-2.6.1/ipkg-install ; )
running install
running build
running build_py
warning: build_py: byte-compiling is disabled, skipping.

running build_ext
Error: pg_config executable not found.

Please add the directory containing pg_config to the PATH
or specify the full executable path with the option:

    python setup.py build_ext --pg-config /path/to/pg_config build ...

or with the pg_config option in 'setup.cfg'.
Makefile:55: recipe for target '/magazyn/Pliki/openwrt/trunk/build_dir/target-mips_34kc_musl-1.1.10/psycopg2-2.6.1/.built' failed
make[3]: *** [/magazyn/Pliki/openwrt/trunk/build_dir/target-mips_34kc_musl-1.1.10/psycopg2-2.6.1/.built] Error 1
[... cut ...]
make[2]: Leaving directory '/magazyn/Pliki/openwrt/trunk'
package/Makefile:188: recipe for target '/magazyn/Pliki/openwrt/trunk/staging_dir/target-mips_34kc_musl-1.1.10/stamp/.package_compile' failed
make[1]: *** [/magazyn/Pliki/openwrt/trunk/staging_dir/target-mips_34kc_musl-1.1.10/stamp/.package_compile] Error 2
make[1]: Leaving directory '/magazyn/Pliki/openwrt/trunk'
/magazyn/Pliki/openwrt/trunk/include/toplevel.mk:181: polecenia dla obiektu 'world' nie powiodły się
make: *** [world] Błąd 2

the-alien, 2015-09-27:
Hi!
Don't you have libpq package built and installed into staging directory before this package?

build000, 2015-10-01:
I have this packages (libpq) marked as <m>

commodo, 2015-12-13:
@build000 @the-alien can this be closed ?

build000, 2015-12-13:
I do not have time to check - I read threads on how to resolve the problem - temporal think that the bug is fixed close.

closed 2015-12-13 by build000