openwrt/packages #1731: cyassl compile  error

taeasy, 2015-08-29:
^
wolfcrypt/src/sha256.c:611:5: note: in expansion of macro '_RegToDigest'
     _RegToDigest(S_0, S_1, S_2, S_3, S_4, S_5, S_6, S_7 )
     ^
wolfcrypt/src/sha256.c:1182:5: note: in expansion of macro 'RegToDigest'
     RegToDigest(S_0,S_1,S_2,S_3,S_4,S_5,S_6,S_7) ;  
     ^
wolfcrypt/src/sha256.c:599:5: error: unknown register name '%r8' in 'asm'
     __asm__ volatile("movl %"#S_3", %0":"=r"(d)::SSE_REGs) ; sha256->digest[3] += d;\
     ^
wolfcrypt/src/sha256.c:611:5: note: in expansion of macro '_RegToDigest'
     _RegToDigest(S_0, S_1, S_2, S_3, S_4, S_5, S_6, S_7 )
     ^
wolfcrypt/src/sha256.c:1182:5: note: in expansion of macro 'RegToDigest'
     RegToDigest(S_0,S_1,S_2,S_3,S_4,S_5,S_6,S_7) ;  
     ^
wolfcrypt/src/sha256.c:599:5: error: PIC register clobbered by '%ebx' in 'asm'
     __asm__ volatile("movl %"#S_3", %0":"=r"(d)::SSE_REGs) ; sha256->digest[3] += d;\
     ^
wolfcrypt/src/sha256.c:611:5: note: in expansion of macro '_RegToDigest'
     _RegToDigest(S_0, S_1, S_2, S_3, S_4, S_5, S_6, S_7 )
     ^
wolfcrypt/src/sha256.c:1182:5: note: in expansion of macro 'RegToDigest'
     RegToDigest(S_0,S_1,S_2,S_3,S_4,S_5,S_6,S_7) ;  
     ^
wolfcrypt/src/sha256.c:600:5: error: unknown register name '%r15' in 'asm'
     __asm__ volatile("movl %"#S_4", %0":"=r"(d)::SSE_REGs) ; sha256->digest[4] += d;\
     ^
wolfcrypt/src/sha256.c:611:5: note: in expansion of macro '_RegToDigest'
     _RegToDigest(S_0, S_1, S_2, S_3, S_4, S_5, S_6, S_7 )
     ^
wolfcrypt/src/sha256.c:1182:5: note: in expansion of macro 'RegToDigest'
     RegToDigest(S_0,S_1,S_2,S_3,S_4,S_5,S_6,S_7) ;  
     ^
wolfcrypt/src/sha256.c:600:5: error: unknown register name '%r14' in 'asm'
     __asm__ volatile("movl %"#S_4", %0":"=r"(d)::SSE_REGs) ; sha256->digest[4] += d;\
     ^
wolfcrypt/src/sha256.c:611:5: note: in expansion of macro '_RegToDigest'
     _RegToDigest(S_0, S_1, S_2, S_3, S_4, S_5, S_6, S_7 )
     ^
wolfcrypt/src/sha256.c:1182:5: note: in expansion of macro 'RegToDigest'
     RegToDigest(S_0,S_1,S_2,S_3,S_4,S_5,S_6,S_7) ;  
     ^
wolfcrypt/src/sha256.c:600:5: error: unknown register name '%r13' in 'asm'
     __asm__ volatile("movl %"#S_4", %0":"=r"(d)::SSE_REGs) ; sha256->digest[4] += d;\
     ^
wolfcrypt/src/sha256.c:611:5: note: in expansion of macro '_RegToDigest'
     _RegToDigest(S_0, S_1, S_2, S_3, S_4, S_5, S_6, S_7 )
     ^
wolfcrypt/src/sha256.c:1182:5: note: in expansion of macro 'RegToDigest'
     RegToDigest(S_0,S_1,S_2,S_3,S_4,S_5,S_6,S_7) ;  
     ^
wolfcrypt/src/sha256.c:600:5: error: unknown register name '%r12' in 'asm'
     __asm__ volatile("movl %"#S_4", %0":"=r"(d)::SSE_REGs) ; sha256->digest[4] += d;\
     ^
wolfcrypt/src/sha256.c:611:5: note: in expansion of macro '_RegToDigest'
     _RegToDigest(S_0, S_1, S_2, S_3, S_4, S_5, S_6, S_7 )
     ^
wolfcrypt/src/sha256.c:1182:5: note: in expansion of macro 'RegToDigest'
     RegToDigest(S_0,S_1,S_2,S_3,S_4,S_5,S_6,S_7) ;  
     ^
wolfcrypt/src/sha256.c:600:5: error: unknown register name '%r11' in 'asm'
     __asm__ volatile("movl %"#S_4", %0":"=r"(d)::SSE_REGs) ; sha256->digest[4] += d;\
     ^
wolfcrypt/src/sha256.c:611:5: note: in expansion of macro '_RegToDigest'
     _RegToDigest(S_0, S_1, S_2, S_3, S_4, S_5, S_6, S_7 )
     ^
wolfcrypt/src/sha256.c:1182:5: note: in expansion of macro 'RegToDigest'
     RegToDigest(S_0,S_1,S_2,S_3,S_4,S_5,S_6,S_7) ;  
     ^
[... 362 lines cut ...]
wolfcrypt/src/sha256.c:481:5: note: in expansion of macro 'SAVE_XMM_YMM'
     SAVE_XMM_YMM ; /* for Intel AVX */
     ^
wolfcrypt/src/sha256.c:270:25: error: unknown register name 'xmm9' in 'asm'
 #define  SAVE_XMM_YMM   __asm__ volatile("or %%r8d, %%r8d":::\
                         ^
wolfcrypt/src/sha256.c:481:5: note: in expansion of macro 'SAVE_XMM_YMM'
     SAVE_XMM_YMM ; /* for Intel AVX */
     ^
wolfcrypt/src/sha256.c:270:25: error: unknown register name 'xmm8' in 'asm'
 #define  SAVE_XMM_YMM   __asm__ volatile("or %%r8d, %%r8d":::\
                         ^
wolfcrypt/src/sha256.c:481:5: note: in expansion of macro 'SAVE_XMM_YMM'
     SAVE_XMM_YMM ; /* for Intel AVX */
     ^
make[5]: *** [wolfcrypt/src/src_libwolfssl_la-sha256.lo] Error 1
make[5]: Leaving directory `/home/main/x86/build_dir/target-i386_i486_musl-1.1.10/wolfssl-3.6.0'
make[4]: *** [all] Error 2
make[4]: Leaving directory `/home/main/x86/build_dir/target-i386_i486_musl-1.1.10/wolfssl-3.6.0'
make[3]: *** [/home/main/x86/build_dir/target-i386_i486_musl-1.1.10/wolfssl-3.6.0/.built] Error 2
make[3]: Leaving directory `/home/main/x86/package/libs/cyassl'
make[2]: *** [package/libs/cyassl/compile] Error 2
make[2]: Leaving directory `/home/main/x86'
make[1]: *** [/home/main/x86/staging_dir/target-i386_i486_musl-1.1.10/stamp/.package_compile] Error 2
make[1]: Leaving directory `/home/main/x86'

plntyk, 2015-08-31:
buildbot x86 is broken too - probably same error:
http://buildbot.openwrt.org:8010/broken_packages/x86/cyassl/compile.txt
warning: 16M html !

hnyman, 2016-08-26:
compiles currently ok in the trunk buildbot

closed 2016-08-26 by hnyman