openwrt/packages #1745: wifidog init script executing wifidog too early

gareth41, 2015-09-01:
I've been testing wifidog with openwrt on several of the ubiquiti air-routers placed at different locations, cafe's hotels etc...  I've found that the default init script with procd starts wifidog too early and this causes problems.  Although wifidog still functions and behaves normally, memory is gradually leaked until wifidog crashes because it cant malloc anymore, either that or the router needs hard rebooting.

Removing procd from the init script and adding "wifidog -c /etc/myconfig.conf" on its own without procd doesn't work at all and wifidog stops shortly after executing because the vlan interface its set to listen on is not up yet.  With procd, wifidog is somehow able to start even though the vlan listen interface is not up.

Starting wifidog manually via ssh after the router boots doesn't exhibit any memory problems when left running for an extended period of time with many clients connecting and logging in.

Changing START=65 to 95 and adding "sleep 20" into the wifidog init script with or without procd fixes the problem with starting too early and also the memory leak.

I don't know if this problem occurs on other hardware running openwrt as I haven't tested any in heavy use environments like Hotels, just the air-router.

yousong, 2016-09-08:
ping wifidog's maintainer @mhaas 

Here are my brief summary here

1. the memory leak issue may need to be reported to the upstream wifidog project and resolved there first
2. Adding `sleep 20` to a init script is generally not acceptable as it may block the boot process for all other pending jobs for that long

dibdot, 2018-08-18:
please retest with latest release.

closed 2018-08-18 by dibdot