openwrt/packages #1758: libcap doesn't compile on darwin (osx yosemite 10.10.2)

simon2b, 2015-09-08:
The out of the box experience for building libcap is:

```
$ make package/libcap/compile V=s
make[1]: Entering directory `/Volumes/OpenWrtDev/project-openwrt/openwrt'
make[2]: Entering directory `/Volumes/OpenWrtDev/project-openwrt/openwrt/package/libs/toolchain'
if [ -f /Volumes/OpenWrtDev/project-openwrt/openwrt/staging_dir/target-arm_cortex-a9+vfpv3_uClibc-0.9.33.2_eabi/pkginfo/toolchain.default.install.clean ]; then rm -f /Volumes/OpenWrtDev/project-openwrt/openwrt/staging_dir/target-arm_cortex-a9+vfpv3_uClibc-0.9.33.2_eabi/pkginfo/toolchain.default.install /Volumes/OpenWrtDev/project-openwrt/openwrt/staging_dir/target-arm_cortex-a9+vfpv3_uClibc-0.9.33.2_eabi/pkginfo/toolchain.default.install.clean; fi; echo "libc" >> /Volumes/OpenWrtDev/project-openwrt/openwrt/staging_dir/target-arm_cortex-a9+vfpv3_uClibc-0.9.33.2_eabi/pkginfo/toolchain.default.install
if [ -f /Volumes/OpenWrtDev/project-openwrt/openwrt/staging_dir/target-arm_cortex-a9+vfpv3_uClibc-0.9.33.2_eabi/pkginfo/toolchain.default.install.clean ]; then rm -f /Volumes/OpenWrtDev/project-openwrt/openwrt/staging_dir/target-arm_cortex-a9+vfpv3_uClibc-0.9.33.2_eabi/pkginfo/toolchain.default.install /Volumes/OpenWrtDev/project-openwrt/openwrt/staging_dir/target-arm_cortex-a9+vfpv3_uClibc-0.9.33.2_eabi/pkginfo/toolchain.default.install.clean; fi; echo "libgcc" >> /Volumes/OpenWrtDev/project-openwrt/openwrt/staging_dir/target-arm_cortex-a9+vfpv3_uClibc-0.9.33.2_eabi/pkginfo/toolchain.default.install
if [ -f /Volumes/OpenWrtDev/project-openwrt/openwrt/staging_dir/target-arm_cortex-a9+vfpv3_uClibc-0.9.33.2_eabi/pkginfo/toolchain.default.install.clean ]; then rm -f /Volumes/OpenWrtDev/project-openwrt/openwrt/staging_dir/target-arm_cortex-a9+vfpv3_uClibc-0.9.33.2_eabi/pkginfo/toolchain.default.install /Volumes/OpenWrtDev/project-openwrt/openwrt/staging_dir/target-arm_cortex-a9+vfpv3_uClibc-0.9.33.2_eabi/pkginfo/toolchain.default.install.clean; fi; echo "libstdcpp" >> /Volumes/OpenWrtDev/project-openwrt/openwrt/staging_dir/target-arm_cortex-a9+vfpv3_uClibc-0.9.33.2_eabi/pkginfo/toolchain.default.install
if [ -f /Volumes/OpenWrtDev/project-openwrt/openwrt/staging_dir/target-arm_cortex-a9+vfpv3_uClibc-0.9.33.2_eabi/pkginfo/toolchain.default.install.clean ]; then rm -f /Volumes/OpenWrtDev/project-openwrt/openwrt/staging_dir/target-arm_cortex-a9+vfpv3_uClibc-0.9.33.2_eabi/pkginfo/toolchain.default.install /Volumes/OpenWrtDev/project-openwrt/openwrt/staging_dir/target-arm_cortex-a9+vfpv3_uClibc-0.9.33.2_eabi/pkginfo/toolchain.default.install.clean; fi; echo "libpthread" >> /Volumes/OpenWrtDev/project-openwrt/openwrt/staging_dir/target-arm_cortex-a9+vfpv3_uClibc-0.9.33.2_eabi/pkginfo/toolchain.default.install
if [ -f /Volumes/OpenWrtDev/project-openwrt/openwrt/staging_dir/target-arm_cortex-a9+vfpv3_uClibc-0.9.33.2_eabi/pkginfo/toolchain.default.install.clean ]; then rm -f /Volumes/OpenWrtDev/project-openwrt/openwrt/staging_dir/target-arm_cortex-a9+vfpv3_uClibc-0.9.33.2_eabi/pkginfo/toolchain.default.install /Volumes/OpenWrtDev/project-openwrt/openwrt/staging_dir/target-arm_cortex-a9+vfpv3_uClibc-0.9.33.2_eabi/pkginfo/toolchain.default.install.clean; fi; echo "librt" >> /Volumes/OpenWrtDev/project-openwrt/openwrt/staging_dir/target-arm_cortex-a9+vfpv3_uClibc-0.9.33.2_eabi/pkginfo/toolchain.default.install
make[2]: Leaving directory `/Volumes/OpenWrtDev/project-openwrt/openwrt/package/libs/toolchain'
make[2]: Entering directory `/Volumes/OpenWrtDev/project-openwrt/openwrt/feeds/packages/libs/libcap'
CFLAGS="-Os -pipe -march=armv7-a -mtune=cortex-a9 -mfpu=vfpv3-d16 -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -mfloat-abi=soft  -I/Volumes/OpenWrtDev/project-openwrt/openwrt/staging_dir/target-arm_cortex-a9+vfpv3_uClibc-0.9.33.2_eabi/usr/include -I/Volumes/OpenWrtDev/project-openwrt/openwrt/staging_dir/target-arm_cortex-a9+vfpv3_uClibc-0.9.33.2_eabi/include -I/Volumes/OpenWrtDev/project-openwrt/openwrt/staging_dir/toolchain-arm_cortex-a9+vfpv3_gcc-4.8-linaro_uClibc-0.9.33.2_eabi/usr/include -I/Volumes/OpenWrtDev/project-openwrt/openwrt/staging_dir/toolchain-arm_cortex-a9+vfpv3_gcc-4.8-linaro_uClibc-0.9.33.2_eabi/include " CXXFLAGS="-Os -pipe -march=armv7-a -mtune=cortex-a9 -mfpu=vfpv3-d16 -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -mfloat-abi=soft  -I/Volumes/OpenWrtDev/project-openwrt/openwrt/staging_dir/target-arm_cortex-a9+vfpv3_uClibc-0.9.33.2_eabi/usr/include -I/Volumes/OpenWrtDev/project-openwrt/openwrt/staging_dir/target-arm_cortex-a9+vfpv3_uClibc-0.9.33.2_eabi/include -I/Volumes/OpenWrtDev/project-openwrt/openwrt/staging_dir/toolchain-arm_cortex-a9+vfpv3_gcc-4.8-linaro_uClibc-0.9.33.2_eabi/usr/include -I/Volumes/OpenWrtDev/project-openwrt/openwrt/staging_dir/toolchain-arm_cortex-a9+vfpv3_gcc-4.8-linaro_uClibc-0.9.33.2_eabi/include " LDFLAGS="-L/Volumes/OpenWrtDev/project-openwrt/openwrt/staging_dir/target-arm_cortex-a9+vfpv3_uClibc-0.9.33.2_eabi/usr/lib -L/Volumes/OpenWrtDev/project-openwrt/openwrt/staging_dir/target-arm_cortex-a9+vfpv3_uClibc-0.9.33.2_eabi/lib -L/Volumes/OpenWrtDev/project-openwrt/openwrt/staging_dir/toolchain-arm_cortex-a9+vfpv3_gcc-4.8-linaro_uClibc-0.9.33.2_eabi/usr/lib -L/Volumes/OpenWrtDev/project-openwrt/openwrt/staging_dir/toolchain-arm_cortex-a9+vfpv3_gcc-4.8-linaro_uClibc-0.9.33.2_eabi/lib " /Applications/Xcode.app/Contents/Developer/usr/bin/make -j1 -C /Volumes/OpenWrtDev/project-openwrt/openwrt/build_dir/target-arm_cortex-a9+vfpv3_uClibc-0.9.33.2_eabi/libcap-2.24/. AR="arm-openwrt-linux-uclibcgnueabi-gcc-ar" AS="arm-openwrt-linux-uclibcgnueabi-gcc -c -Os -pipe -march=armv7-a -mtune=cortex-a9 -mfpu=vfpv3-d16 -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -mfloat-abi=soft" LD=arm-openwrt-linux-uclibcgnueabi-ld NM="arm-openwrt-linux-uclibcgnueabi-gcc-nm" CC="arm-openwrt-linux-uclibcgnueabi-gcc" GCC="arm-openwrt-linux-uclibcgnueabi-gcc" CXX="arm-openwrt-linux-uclibcgnueabi-g++" RANLIB="arm-openwrt-linux-uclibcgnueabi-gcc-ranlib" STRIP=arm-openwrt-linux-uclibcgnueabi-strip OBJCOPY=arm-openwrt-linux-uclibcgnueabi-objcopy OBJDUMP=arm-openwrt-linux-uclibcgnueabi-objdump SIZE=arm-openwrt-linux-uclibcgnueabi-size CROSS="arm-openwrt-linux-uclibcgnueabi-" ARCH="arm" CFLAGS="-Os -pipe -march=armv7-a -mtune=cortex-a9 -mfpu=vfpv3-d16 -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -mfloat-abi=soft" BUILD_CC="cc" BUILD_CFLAGS="-fpic -I/Volumes/OpenWrtDev/project-openwrt/openwrt/build_dir/target-arm_cortex-a9+vfpv3_uClibc-0.9.33.2_eabi/libcap-2.24/libcap/include" CFLAGS="-Os -pipe -march=armv7-a -mtune=cortex-a9 -mfpu=vfpv3-d16 -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -mfloat-abi=soft" LD="arm-openwrt-linux-uclibcgnueabi-gcc" LDFLAGS="-L/Volumes/OpenWrtDev/project-openwrt/openwrt/staging_dir/target-arm_cortex-a9+vfpv3_uClibc-0.9.33.2_eabi/usr/lib -L/Volumes/OpenWrtDev/project-openwrt/openwrt/staging_dir/target-arm_cortex-a9+vfpv3_uClibc-0.9.33.2_eabi/lib -L/Volumes/OpenWrtDev/project-openwrt/openwrt/staging_dir/toolchain-arm_cortex-a9+vfpv3_gcc-4.8-linaro_uClibc-0.9.33.2_eabi/usr/lib -L/Volumes/OpenWrtDev/project-openwrt/openwrt/staging_dir/toolchain-arm_cortex-a9+vfpv3_gcc-4.8-linaro_uClibc-0.9.33.2_eabi/lib -shared" INDENT="| true" PAM_CAP="no" LIBATTR="no" DYNAMIC="yes" lib="lib" ;
/Applications/Xcode.app/Contents/Developer/usr/bin/make -C libcap all
cc -fpic -I/Volumes/OpenWrtDev/project-openwrt/openwrt/build_dir/target-arm_cortex-a9+vfpv3_uClibc-0.9.33.2_eabi/libcap-2.24/libcap/include _makenames.c -o _makenames
In file included from _makenames.c:10:
/Volumes/OpenWrtDev/project-openwrt/openwrt/build_dir/target-arm_cortex-a9+vfpv3_uClibc-0.9.33.2_eabi/libcap-2.24/libcap/include/sys/capability.h:24:10: fatal error:
      'linux/types.h' file not found
#include <linux/types.h>
         ^
1 error generated.
make[4]: *** [_makenames] Error 1
make[3]: *** [all] Error 2
make[2]: *** [/Volumes/OpenWrtDev/project-openwrt/openwrt/build_dir/target-arm_cortex-a9+vfpv3_uClibc-0.9.33.2_eabi/libcap-2.24/.built] Error 2
make[2]: Leaving directory `/Volumes/OpenWrtDev/project-openwrt/openwrt/feeds/packages/libs/libcap'
make[1]: *** [package/feeds/packages/libcap/compile] Error 2
make[1]: Leaving directory `/Volumes/OpenWrtDev/project-openwrt/openwrt'
make: *** [package/libcap/compile] Error 2
```

Notable is the line:
[... cut ...]
Specifically it is missing the linux-api-headers dependency because they are normally provided by the cross compile toolchain environment. I suspect that on a linux host, those includes are part of the default compiler environment and hence it's working fine on linux?

Anyway, I was able to get it to build by adding an include for the linux-dev headers to the feed Makefile:34.

```BUILD_CFLAGS="$(FPIC) -I$(BUILD_DIR_TOOLCHAIN)/linux-dev/include -I$(PKG_BUILD_DIR)/libcap/include" \```

I defer to the package maintainer for integrating an appropriate form of this fix.

vincas, 2015-10-16:
I did what you recommended and now I'm getting:
clang: error: unknown argument: '-mno-branch-likely'
clang: error: unknown argument: '-fno-caller-saves'
clang: error: unknown argument: '-fhonour-copts'
clang: error: unknown argument: '-minterlink-mips16'

Any suggestions?

yousong, 2016-09-08:
ping licap's maintainer @sbyx

dokterbob, 2016-10-19:
Replicated the problem, the workaround did not seem to work for me.

diizzyy, 2017-06-07:
@simon2b 
Still an issue?

svetlyo, 2017-08-01:
@diizzyy Yes, it still happens. (macOS 10.11.6)

pandysong, 2017-09-25:
I have the same problem. Resolved?

diizzyy, 2019-03-28:
Closing this due to timeout

closed 2019-03-28 by diizzyy