openwrt/packages #1761: openssh: Unable to authenticate with bumped OpenSSH (dropbear remains OK)

zxdavb, 2015-09-08:
Regarding the recent OpenSSH bump from v6.9 to v7.1...

I am unable to authenticate via OpenSSH (on port 22), although DropBear works fine (on port 8022). 
When I first noticed this with a daily build (my buildroot), I was able to get OpenSSH working again by uninstalling OpenSSH-Server v7.1 (and all associated packages), and re-installing the v6.9 version from downloads.openwrt.org. 

Nothing in the syslog. Changing the password did not help.

This is a duplicate of: https://dev.openwrt.org/ticket/20463...

zxdavb, 2015-09-08:
@tripolar, FYI

tripolar, 2015-11-15:
i'm currently using openssh 7.1 without a problem,  try to start sshd with  

 -E log_file
             Append debug logs to log_file instead of the system log.

Without any error logs I can't help you.

zxdavb, 2015-11-15:
Hi @tripolar,

It doesn't work on trunk, DD (build 47373), although it works fine on CC (build 47466).  In either case, I use the exact same **.config file**, and I **sysupgrade** between the two.

I can give you a copy of my **.config** file (*before* **make defconfig**) and the output of **sysupgrade -l** if it helps.  The output of the failed logon is below, I am using port 22 for dropbear at the moment.

```
root@db-router:~# /usr/sbin/sshd -D -E /tmp/sshd.log
^C
root@db-router:~# cat /tmp/sshd.log
Could not load host key: /etc/ssh/ssh_host_ecdsa_key
Could not load host key: /etc/ssh/ssh_host_ed25519_key
Server listening on 0.0.0.0 port 8022.
Server listening on :: port 8022.
Could not load host key: /etc/ssh/ssh_host_ecdsa_key
Could not load host key: /etc/ssh/ssh_host_ed25519_key
Failed password for root from 172.27.0.100 port 50978 ssh2
Failed password for root from 172.27.0.100 port 50978 ssh2
Failed password for root from 172.27.0.100 port 50978 ssh2
Failed password for root from 172.27.0.100 port 50978 ssh2
Failed password for root from 172.27.0.100 port 50978 ssh2
maximum authentication attempts exceeded for root from 172.27.0.100 port 50978 ssh2 [preauth]
Disconnecting: Too many authentication failures [preauth]
```

zxdavb, 2015-11-15:
Sorry, that was rubbish!  This time I have set LogLevel to DEBUG.

This is a single attempt, followed by a Cntrl_C when it failed to authenticate

```
root@db-router:~# cat /tmp/sshd.log
Could not load host key: /etc/ssh/ssh_host_ecdsa_key
Could not load host key: /etc/ssh/ssh_host_ed25519_key
debug1: Set /proc/self/oom_score_adj from 0 to -1000
debug1: Bind to port 8022 on 0.0.0.0.
Server listening on 0.0.0.0 port 8022.
debug1: Bind to port 8022 on ::.
Server listening on :: port 8022.
debug1: Forked child 4261.
debug1: Set /proc/self/oom_score_adj to 0
debug1: rexec start in 7 out 7 newsock 7 pipe 9 sock 10
Could not load host key: /etc/ssh/ssh_host_ecdsa_key
Could not load host key: /etc/ssh/ssh_host_ed25519_key
debug1: inetd sockets after dupping: 5, 5
Connection from 172.27.0.100 port 51027 on 172.27.0.1 port 8022
debug1: Client protocol version 2.0; client software version PuTTY_Release_0.64
debug1: match: PuTTY_Release_0.64 pat PuTTY_Local:*,PuTTY-Release-0.5*,PuTTY_Release_0.5*,PuTTY_Release_0.60*,PuTTY_Release_0.61*,PuTTY_Release_0.62*,PuTTY_Release_0.63*,PuTTY_Release_0.64* compat 0x00004000
debug1: Enabling compatibility mode for protocol 2.0
debug1: Local version string SSH-2.0-OpenSSH_7.1
debug1: permanently_set_uid: 22/22 [preauth]
debug1: list_hostkey_types: ssh-rsa [preauth]
debug1: SSH2_MSG_KEXINIT sent [preauth]
debug1: SSH2_MSG_KEXINIT received [preauth]
debug1: kex: client->server aes256-ctr hmac-sha2-256 none [preauth]
debug1: kex: server->client aes256-ctr hmac-sha2-256 none [preauth]
debug1: expecting SSH2_MSG_KEXDH_INIT [preauth]
debug1: SSH2_MSG_NEWKEYS sent [preauth]
debug1: expecting SSH2_MSG_NEWKEYS [preauth]
debug1: SSH2_MSG_NEWKEYS received [preauth]
debug1: KEX done [preauth]
debug1: userauth-request for user root service ssh-connection method none [preauth]
debug1: attempt 0 failures 0 [preauth]
debug1: userauth-request for user root service ssh-connection method keyboard-interactive [preauth]
debug1: attempt 1 failures 0 [preauth]
debug1: keyboard-interactive devs  [preauth]
debug1: auth2_challenge: user=root devs= [preauth]
debug1: kbdint_alloc: devices '' [preauth]
debug1: userauth-request for user root service ssh-connection method password [preauth]
debug1: attempt 2 failures 1 [preauth]
Failed password for root from 172.27.0.100 port 51027 ssh2
debug1: Unable to open the btmp file /var/log/btmp: No such file or directory
Received disconnect from 172.27.0.100: 13: Unable to authenticate [preauth]
Disconnected from 172.27.0.100 [preauth]
debug1: do_cleanup [preauth]
debug1: monitor_read_log: child log fd closed
debug1: do_cleanup
debug1: Killing privsep child 4262
root@db-router:~#
```

tripolar, 2015-12-07:
seems you use the wrong password
Failed password for root from 172.27.0.100 port 51027 ssh2

closed 2015-12-07 by tripolar

zxdavb, 2015-12-08:
It says wrong password, but I assure you I entered the correct password.

Maybe something wrong with the hash algorithm?

champtar, 2015-12-08:
i know there is a crypt() size optimisation in openwrt
http://git.openwrt.org/?p=openwrt.git;a=commitdiff;h=58afb8c17d363254276752f98c20135591eebb5e
http://git.openwrt.org/?p=openwrt.git;a=commitdiff;h=126753e158e618758d46bde752775902a1c7c2b4

only des and md5 support right now, blowfish, sha256, sha512 are disabled
please paste the first 3 chars of your password hash

zxdavb, 2015-12-08:
Is this what you mean?:

*root@db-router:/# cat
/etc/shadowroot:$1$xxxxxxxxxxxxxxxxxxxxxxxxx:16712:0:99999:7:::*

I think $1$ is MD5.

BTW, the problem with OpenSSH remains, even when I change the password
using passwd through Dropbear.

If any one is interested, I am willing to permit remote access to the
router (I will wipe it, and set up DropBear on one port, and OpenSSH on
another).  It will just take me a day or so to set up.

Dave

tripolar, 2015-12-08:
do you still use sysupgrade to update? please try to setup a new image and flash it without sysupgrade, through tftp or with mtd only

tripolar, 2015-12-09:
do you still use sysupgrade? try to flash the image through tftp or with mtd only

tripolar, 2015-12-09:
and try to use public key authentication to see if this works. i only use public keys and i just tried to login with a password and i works here without a problem, i'm on r47793

zxdavb, 2015-12-09:
Yes, I always sysupgrade.  I will try mtd and get back to you.

zxdavb, 2015-12-13:
@tripolar:

I still have the same problem!

I have done a upgrade via TFTP, bleeding edge: r47890, and I have used the
following minimalist .config file:
  CONFIG_TARGET_ar71xx=y
  CONFIG_TARGET_ar71xx_generic=y
  CONFIG_TARGET_ar71xx_generic_UBNTRSPRO=y
  CONFIG_PACKAGE_wpad-mini=n
  CONFIG_PACKAGE_wpad=y
  CONFIG_PACKAGE_hostapd-utils=y
  CONFIG_PACKAGE_dnsmasq=n
  CONFIG_PACKAGE_openssh-client=y
  CONFIG_PACKAGE_openssh-server=y

I have made no changes to the router, except set the password with passwd,
and:
    opkg update
    opkg install nano

    uci add firewall  rule
    uci set firewall.@rule[-1].name=Allow-Dropbear-Inbound
    uci set firewall.@rule[-1].src=wan
    uci set firewall.@rule[-1].target=ACCEPT
    uci set firewall.@rule[-1].proto=tcp
    uci set firewall.@rule[-1].dest_port=8022

    uci set firewall.@rule[-1].enabled=1

    uci add firewall  rule
    uci set firewall.@rule[-1].name=Allow-OpenSSH-Inbound
    uci set firewall.@rule[-1].src=wan
    uci set firewall.@rule[-1].target=ACCEPT
    uci set firewall.@rule[-1].proto=tcp
    uci set firewall.@rule[-1].dest_port=8122

    uci set firewall.@rule[-1].enabled=1

    uci set dropbear.@dropbear[0].Port='8022'

    sed -i -e 's:^#*.Port 22:Port 8122:' /etc/ssh/sshd_config

    uci commit

    /etc/init.d/dropbear restart
    /etc/init.d/sshd enable
    /etc/init.d/sshd start

And now I offer my router to anyone who is willing to have a look at it &
find out why I can login over SSH with Dropbear, but not with OpenSSH: just PM
me, I will provide the DNS hostname and password.

-DB

tripolar, 2015-12-13:
send me a message with the data

v90, 2015-12-19:
"Server refused keyboard authentication"

If I start dropbear, stop ssh, and change the /etc/ssh/sshd_config file to PermitRootLogin yes, stop dropbear, start ssh, I can login.

Which source file would I edit just to make PermitRootLogin permanently yes when I build? I'm making my way through but not finding one that makes it through the bin post-compile.

zxdavb, 2015-12-19:
Well, don't I feel stupid!  *PermitRootLogin Yes* solves my problem as
well!  I don't ever seem to recall getting a message saying: "Server
refused keyboard authentication" tho'...

Anyway, the 'correct' way to achieve what you want might be to create a
patch, but the way I like to do such things (on my buildroot) is something
like :

  cd ~/openwrt/trunk
  cat << 'EOF' > ./files/etc/uci-defaults/my-openssh
  #!/bin/sh
      echo 'PermitRootLogin Yes' >> /etc/ssh/sshd_config
  #   /etc/init.d/sshd reload ## I don't think this is needed
  EOF
I will test it sometime & let you know, but you'll probably get around to
it before I do.

==================================

v90, 2015-12-19:
I slept on it, and decided I've changed my mind about patching. It's not really 'the OpenBSD way' to login to root interactively with SSH keyboard auth, so it might just be easier to do it the right way rather than perpetuate bad practice.

I'm thinking the right way would be to add an sshkey to the root account, or a non-root account with sudo. If all else fails I'll see about adding a custom sshd_config to the <buildroot>/files directory.

v90, 2015-12-20:
Confirm fixed if you just put your own config /openwrt/files/etc/ssh/sshd_config before building.