openwrt/packages #1763: Aiccu IPv6 tunnel not working

ggiopp, 2015-09-09:
Looks like aiccu starts correctly, but it is killed after a few seconds.
I am using OpenWrt Chaos Calmer r46796.

Here is the log:
```
Tue Sep  8 17:17:30 2015 daemon.notice netifd: Interface 'wan6' is setting up now
Tue Sep  8 17:17:40 2015 daemon.notice netifd: wan6 (7324): Waiting 10 secs for NTP sync...
Tue Sep  8 17:17:50 2015 daemon.notice netifd: wan6 (7324): Waiting 20 secs for NTP sync...
Tue Sep  8 17:18:00 2015 daemon.notice netifd: wan6 (7324): Waiting 30 secs for NTP sync...
	modprobe module
Tue Sep  8 17:19:01 2015 daemon.info modprobe: Usage:
	modprobe module
Tue Sep  8 17:19:02 2015 local7.debug : sock_getline() : "200 SixXS TIC Service on iedub02.sixxs.net ready (http://www.sixxs.net)"
Tue Sep  8 17:19:02 2015 local7.debug : sock_printf()  : "client TIC/draft-00 AICCU/2007.01.15-console-linux Linux/4.1.6"
Tue Sep  8 17:19:02 2015 local7.debug : sock_getline() : "200 Client Identity accepted"
Tue Sep  8 17:19:02 2015 local7.debug : sock_printf()  : "get unixtime"
Tue Sep  8 17:19:02 2015 local7.debug : sock_getline() : "200 1441725543"
Tue Sep  8 17:19:02 2015 local7.debug : sock_printf()  : "username xxx-SIXXS"
Tue Sep  8 17:19:02 2015 local7.debug : sock_getline() : "200 xxx-SIXXS choose your authentication challenge please"
Tue Sep  8 17:19:02 2015 local7.debug : sock_printf()  : "challenge md5"
Tue Sep  8 17:19:02 2015 local7.debug : sock_getline() : "200 xxx"
Tue Sep  8 17:19:02 2015 local7.debug : sock_printf()  : "authenticate md5 xxx"
Tue Sep  8 17:19:02 2015 local7.debug : sock_getline() : "200 Successfully logged in using md5 as xxx-SIXXS (xxx)"
Tue Sep  8 17:19:02 2015 local7.debug : sock_printf()  : "tunnel show Txxx"
Tue Sep  8 17:19:03 2015 local7.debug : sock_getline() : "201 Showing tunnel information for Txxx"
Tue Sep  8 17:19:03 2015 local7.debug : sock_getline() : "TunnelId: Txxx"
Tue Sep  8 17:19:03 2015 local7.debug : sock_getline() : "Type: 6in4-heartbeat"
Tue Sep  8 17:19:03 2015 local7.debug : sock_getline() : "IPv6 Endpoint: 2001:xxx::2"
Tue Sep  8 17:19:03 2015 local7.debug : sock_getline() : "IPv6 POP: 2001:xxx::1"
Tue Sep  8 17:19:03 2015 local7.debug : sock_getline() : "IPv6 PrefixLength: 64"
Tue Sep  8 17:19:03 2015 local7.debug : sock_getline() : "Tunnel MTU: 1472"
Tue Sep  8 17:19:03 2015 local7.debug : sock_getline() : "Tunnel Name: My First Tunnel"
Tue Sep  8 17:19:03 2015 local7.debug : sock_getline() : "POP Id: ittrn01"
Tue Sep  8 17:19:03 2015 local7.debug : sock_getline() : "IPv4 Endpoint: heartbeat"
Tue Sep  8 17:19:03 2015 local7.debug : sock_getline() : "IPv4 POP: 213.254.12.34"
Tue Sep  8 17:19:03 2015 local7.debug : sock_getline() : "UserState: enabled"
Tue Sep  8 17:19:03 2015 local7.debug : sock_getline() : "AdminState: enabled"
Tue Sep  8 17:19:03 2015 local7.debug : sock_getline() : "Password: xxx"
Tue Sep  8 17:19:03 2015 local7.debug : sock_getline() : "Heartbeat_Interval: 60"
Tue Sep  8 17:19:03 2015 local7.debug : sock_getline() : "202 Done"
Tue Sep  8 17:19:03 2015 local7.info : Succesfully retrieved tunnel information for Txxx
Tue Sep  8 17:19:03 2015 local7.debug : sock_printf()  : "QUIT And All That Could Have Been..."
Tue Sep  8 17:19:03 2015 daemon.notice netifd: wan6 (7324): Tunnel Information for Txxx:
Tue Sep  8 17:19:03 2015 daemon.notice netifd: wan6 (7324): POP Id      : ittrn01
Tue Sep  8 17:19:03 2015 daemon.notice netifd: wan6 (7324): IPv6 Local  : 2001:xxx::2/64
Tue Sep  8 17:19:03 2015 daemon.notice netifd: wan6 (7324): IPv6 Remote : 2001:xxx::1/64
Tue Sep  8 17:19:03 2015 daemon.notice netifd: wan6 (7324): Tunnel Type : 6in4-heartbeat
Tue Sep  8 17:19:03 2015 daemon.notice netifd: wan6 (7324): Adminstate  : enabled
Tue Sep  8 17:19:03 2015 daemon.notice netifd: wan6 (7324): Userstate   : enabled
Tue Sep  8 17:19:03 2015 daemon.notice netifd: wan6 (7324): POP Id      : ittrn01
Tue Sep  8 17:19:03 2015 daemon.notice netifd: wan6 (7324): IPv6 Local  : 2001:xxx::2/64
Tue Sep  8 17:19:03 2015 daemon.notice netifd: wan6 (7324): IPv6 Remote : 2001:xxx::1/64
Tue Sep  8 17:19:03 2015 daemon.notice netifd: wan6 (7324): Tunnel Type : 6in4-heartbeat
Tue Sep  8 17:19:03 2015 daemon.notice netifd: wan6 (7324): Adminstate  : enabled
Tue Sep  8 17:19:03 2015 daemon.notice netifd: wan6 (7324): Userstate   : enabled
Tue Sep  8 17:19:03 2015 local7.info : AICCU running as PID 7528
Tue Sep  8 17:19:03 2015 daemon.notice netifd: wan6 (7324): Command failed: Unknown error
Tue Sep  8 17:19:03 2015 local7.debug : heartbeat_socket() - IPv4 : 79.36.61.13
Tue Sep  8 17:19:03 2015 daemon.info modprobe: Usage:
	modprobe module
Tue Sep  8 17:19:03 2015 local7.debug : [HB] HEARTBEAT TUNNEL 2001:xxx::2 sender zzz uuu
Tue Sep  8 17:19:03 2015 local7.debug : [HB] HEARTBEAT TUNNEL 2001:xxx::2 sender zzz uuu
Tue Sep  8 17:19:03 2015 daemon.info modprobe: Usage:
	modprobe module
Tue Sep  8 17:19:03 2015 local7.err : Already running instance HUP'ed, exiting
Tue Sep  8 17:19:03 2015 daemon.notice netifd: Interface 'wan6' is now down
```

The relevant section from /etc/config/network
```
config interface 'wan6'
	option _orig_ifname 'eth0.2'
	option _orig_bridge 'false'
	option proto 'aiccu'
	option username 'xxx-SIXXS'
	option password 'xxx'
	option tunnelid 'Txxx'
	option ip6prefix '2001:xxx::/64'
	option heartbeat '1'
	option verbose '1'
	option mtu '1472'
```

The generated configuration file /tmp/etc/aiccu-wan6.conf
```
username xxx-SIXXS
password xxx
ipv6_interface aiccu-wan6
tunnel_id Txxx
makebeats true
verbose true
defaultroute false
daemonize true
pidfile /var/run/aiccu-wan6.pid
```

ggiopp, 2015-09-09:
@oskar456

oskar456, 2015-09-22:
Sorry for the delay. I have no clue what could be the reason. I'll try to reproduce it. Any ideas welcome :)

Astlor, 2015-09-23:
I am on OpenWrt Chaos Calmer.

Model	NETGEAR WNDR3700v2
Firmware Version	OpenWrt Chaos Calmer 15.05 / LuCI (git-15.248.30277-3836b45)
Kernel Version	3.18.20

This uses the ar71xx branch.  AICCU previously worked under Chaos Calmer RC 2.  I did a sysupdate to Chaos Calmer RTM, and the following occured:

1) AICCU is no longer listed as an installed package.  
2) When I attempt to install I get the following error:

Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for aiccu:
 * 	kernel (= 3.18.14-1-47cf67747cc95dea86ac56de2508504c) * 	kernel (= 3.18.14-1-47cf67747cc95dea86ac56de2508504c) * 	kernel (= 3.18.14-1-47cf67747cc95dea86ac56de2508504c) * 	kernel (= 3.18.14-1-47cf67747cc95dea86ac56de2508504c) * 
 * opkg_install_cmd: Cannot install package aiccu.

It seems that there needs to be a new package compiled for the RTM version, or at least this is the case for the ar71xx branch.  Perhaps it is something as simple as not having the right kernel specified in the requirements for the other branches as well?

The package in question is here: https://downloads.openwrt.org/chaos_calmer/15.05/ar71xx/generic/packages/packages/ and is package aiccu_20070115-12_ar71xx.ipk

Thank you, and I hope some of this helps solve the problem!

luizluca, 2015-09-26:
@Astlor , on sysupgrade all program are lost. You do need to install them again.

The kernel unsolved dependency might indicate that you are using a kernel that is different from the one present in the repo you are using. Please, check /etc/opkg.conf and verify if it is still mentioning the previous rc version. If so, you can grab a clean opkg.conf from /overlay/etc/opkg.conf.

Astlor, 2015-09-30:
That's fixed it.  Cheers!

ezimin, 2015-10-28:
I'm seeing exactly the same issue as @ggiopp described in the first message of this thread.
Please advise how to investigate and troubleshoot it...

oskar456, 2015-10-29:
Does it happen with a clean installation of CC final?

ezimin, 2015-10-29:
It's definitely seen with a clean installation of this image on Asus RT-N16:
https://downloads.openwrt.org/chaos_calmer/15.05/brcm47xx/generic/openwrt-15.05-brcm47xx-generic-squashfs-gz.trx

I haven't tried daily builds yet. 
I'm ready to try, but please share what needs to be checked/collected to help you investigate the issue

LAGonauta, 2015-10-31:
I have the same problem with a Netgear WNDR3700v2. It worked fine on r46760 but stopped working when I updated to r47280 yesterday. Maybe it is something that changed between r46760 and r46796.

Schabram, 2016-01-12:
m.. still brocken @ trunk and Sixxs ban me after few min ... 

SixxsMail"...Any TIC client (eg AICCU) should only connect to the TIC service
at tunnel-setup time. If you connect more than 5 times in an hour
something in your setup is wrong...."

Add a limit of restart / reload etc would be usefull prevents tic server ban

```
Mon Jan 11 02:25:07 2016 local7.err : Already running instance HUP'ed, exiting
Mon Jan 11 02:25:07 2016 daemon.notice netifd: Interface 'wan6' is now down
Mon Jan 11 02:25:07 2016 daemon.notice netifd: Interface 'wan6' is setting up now
Mon Jan 11 02:25:07 2016 daemon.notice netifd: wan6 (15953): NTP synced, stratum 3
Mon Jan 11 02:25:07 2016 daemon.info modprobe: Usage:
	modprobe module
Mon Jan 11 02:25:07 2016 daemon.info modprobe: Usage:
	modprobe module
Mon Jan 11 02:25:08 2016 local7.debug : sock_getline() : "200 SixXS TIC Service on nlhaa01.sixxs.net ready (https://www.sixxs.net)"
Mon Jan 11 02:25:08 2016 local7.debug : sock_printf()  : "client TIC/draft-00 AICCU/2007.01.15-console-linux Linux/4.1.13"
Mon Jan 11 02:25:08 2016 local7.debug : sock_getline() : "200 Client Identity accepted"
Mon Jan 11 02:25:08 2016 local7.debug : sock_printf()  : "get unixtime"
Mon Jan 11 02:25:08 2016 local7.debug : sock_getline() : "200 1452536708"
Mon Jan 11 02:25:08 2016 local7.debug : sock_printf()  : "username ****-****/T******"
Mon Jan 11 02:25:08 2016 local7.debug : sock_getline() : "200 ****-****/T****** choose your authentication challenge please"
Mon Jan 11 02:25:08 2016 local7.debug : sock_printf()  : "challenge md5"
Mon Jan 11 02:25:08 2016 local7.debug : sock_getline() : "200 22206dc4e1f794b1***************"
Mon Jan 11 02:25:08 2016 local7.debug : sock_printf()  : "authenticate md5 36d8e3d762bcb5*****************"
Mon Jan 11 02:25:08 2016 local7.debug : sock_getline() : "200 Successfully logged in using md5 as ****-**** (My Name)"
Mon Jan 11 02:25:08 2016 local7.debug : sock_printf()  : "tunnel list"
Mon Jan 11 02:25:09 2016 local7.debug : sock_getline() : "201 Listing tunnels"
Mon Jan 11 02:25:09 2016 local7.debug : sock_getline() : "T****** 2001:4dd0:ff00:199f::2 heartbeat decgn01"
Mon Jan 11 02:25:09 2016 local7.debug : sock_getline() : "202 <tunnel_id> <ipv6_endpoint> <ipv4_endpoint> <pop_name>"
Mon Jan 11 02:25:09 2016 local7.debug : sock_printf()  : "tunnel show T******"
Mon Jan 11 02:25:09 2016 local7.debug : sock_getline() : "201 Showing tunnel information for T******"
Mon Jan 11 02:25:09 2016 local7.debug : sock_getline() : "TunnelId: T******"
Mon Jan 11 02:25:09 2016 local7.debug : sock_getline() : "Type: 6in4-heartbeat"
Mon Jan 11 02:25:09 2016 local7.debug : sock_getline() : "IPv6 Endpoint: 2001:4dd0:ff00:199f::2"
Mon Jan 11 02:25:09 2016 local7.debug : sock_getline() : "IPv6 POP: 2001:4dd0:ff00:199f::1"
Mon Jan 11 02:25:09 2016 local7.debug : sock_getline() : "IPv6 PrefixLength: 64"
Mon Jan 11 02:25:09 2016 local7.debug : sock_getline() : "Tunnel MTU: 1280"
Mon Jan 11 02:25:09 2016 local7.debug : sock_getline() : "Tunnel Name: My First Tunnel"
Mon Jan 11 02:25:09 2016 local7.debug : sock_getline() : "POP Id: decgn01"
Mon Jan 11 02:25:09 2016 local7.debug : sock_getline() : "IPv4 Endpoint: heartbeat"
Mon Jan 11 02:25:09 2016 local7.debug : sock_getline() : "IPv4 POP: 78.35.24.124"
Mon Jan 11 02:25:09 2016 local7.debug : sock_getline() : "UserState: enabled"
Mon Jan 11 02:25:09 2016 local7.debug : sock_getline() : "AdminState: enabled"
Mon Jan 11 02:25:09 2016 local7.debug : sock_getline() : "Password: 031f078217*********************"
Mon Jan 11 02:25:09 2016 local7.debug : sock_getline() : "Heartbeat_Interval: 60"
Mon Jan 11 02:25:09 2016 local7.debug : sock_getline() : "202 Done"
Mon Jan 11 02:25:09 2016 local7.info : Succesfully retrieved tunnel information for T******
Mon Jan 11 02:25:09 2016 local7.debug : sock_printf()  : "QUIT We will be the only two people left in the world, Yes--Adam and Evil!"
Mon Jan 11 02:25:09 2016 daemon.notice netifd: wan6 (15953): Tunnel Information for T******:
Mon Jan 11 02:25:09 2016 daemon.notice netifd: wan6 (15953): POP Id      : decgn01
Mon Jan 11 02:25:09 2016 daemon.notice netifd: wan6 (15953): IPv6 Local  : 2001:4dd0:ff00:199f::2/64
Mon Jan 11 02:25:09 2016 daemon.notice netifd: wan6 (15953): IPv6 Remote : 2001:4dd0:ff00:199f::1/64
Mon Jan 11 02:25:09 2016 daemon.notice netifd: wan6 (15953): Tunnel Type : 6in4-heartbeat
Mon Jan 11 02:25:09 2016 daemon.notice netifd: wan6 (15953): Adminstate  : enabled
Mon Jan 11 02:25:09 2016 daemon.notice netifd: wan6 (15953): Userstate   : enabled
Mon Jan 11 02:25:09 2016 daemon.notice netifd: wan6 (15953): POP Id      : decgn01
Mon Jan 11 02:25:09 2016 daemon.notice netifd: wan6 (15953): IPv6 Local  : 2001:4dd0:ff00:199f::2/64
Mon Jan 11 02:25:09 2016 daemon.notice netifd: wan6 (15953): IPv6 Remote : 2001:4dd0:ff00:199f::1/64
Mon Jan 11 02:25:09 2016 daemon.notice netifd: wan6 (15953): Tunnel Type : 6in4-heartbeat
Mon Jan 11 02:25:09 2016 daemon.notice netifd: wan6 (15953): Adminstate  : enabled
Mon Jan 11 02:25:09 2016 daemon.notice netifd: wan6 (15953): Userstate   : enabled
Mon Jan 11 02:25:09 2016 local7.info : AICCU running as PID 15976
Mon Jan 11 02:25:09 2016 daemon.notice netifd: wan6 (15953): Command failed: Unknown error
Mon Jan 11 02:25:09 2016 local7.debug : heartbeat_socket() - IPv4 : 2.245.223.108
Mon Jan 11 02:25:09 2016 local7.debug : [HB] HEARTBEAT TUNNEL 2001:4dd0:ff00:199f::2 sender 1452536709 fa51a19b700ab5e*****************
[... cut ...] :199f::2 sender 1452536709 fa51a19b700ab5e*****************
Mon Jan 11 02:25:09 2016 daemon.info modprobe: Usage:
	modprobe module
Mon Jan 11 02:25:09 2016 daemon.info modprobe: Usage:
	modprobe module
-----------BEGIN LOOP:-----------
Mon Jan 11 02:25:09 2016 local7.err : Already running instance HUP'ed, exiting
```
```
config interface 'wan6'
    option proto 'aiccu'
    option username 'xxx-SIXXS/T*****'
    option password 'xxx'
    option ip6prefix '2001:xxx::/48'
    option verbose '1'
```

jow-, 2016-01-12:
I strongly suggest to switch to tunnelbroker.net or any other means to obtain IPv6. The aiccu client and its associated ToS are inherently flawed and unsuitable for headless operation.

ezimin, 2016-01-12:
@jow- it's not always possible. tunnelborker is only option for those, who have static IPv4 address. It's not always the case.

oskar456, 2016-01-12:
@Schabram Sorry, I cannot reproduce your issue, it works for me (logs: http://pastebin.com/tXdTD7s1).
There is already a safeguard that block restarts when AICCU fails. But in your case it looks like aiccu completes successfully and something else fails. It's probably platform specific.

jow-, 2016-01-12:
@ezimin You can use tunnelbroker on dynamic IPs just fine, OpenWrt also implements automatic endpoint update for it.

oskar456, 2016-01-12:
@jow The problem is when you are behind a Carrier Grade NAT, which is more and more common.
And native IPv6 deployment is still far from being generally available :(

Schabram, 2016-01-12:
@oskar456 you use a Ayiya tunnel i run 6in4-hertbeat

oskar456, 2016-01-12:
In case you have public IPv4 address, I would strongly recommend using he.net IPv6 tunnel, which is fully integrated with netifd and doesn't need this broken-by-design AICCU utility.

Schabram, 2016-01-12:
Sad news for me but should be this package by default marked as brocken or edit description with a note that heartbeat is brocken.

Regards,
   Schabram

oskar456, 2016-01-12:
The package works for me in 3 different instances. I've never thought somebody with public IPv4 address would use it, when there's he.net tunnel which is much simpler to use.

I've put a warning to the OpenWRT Wiki that only AYIYA tunnel type has been tested.

Schabram, 2016-01-14:
I managed to fix it. The problem was busybox ip "common/aiccu_linux.c" just use "ip" command to setup i just change it to "/usr/bin/ip"

```
Thu Jan 14 08:13:04 2016 daemon.notice netifd: Interface 'wan6' is setting up now
Thu Jan 14 08:13:04 2016 daemon.notice netifd: wan6 (5267): NTP synced, stratum 4
Thu Jan 14 08:13:04 2016 daemon.info modprobe: Usage:
	modprobe module
Thu Jan 14 08:13:04 2016 daemon.info modprobe: Usage:
	modprobe module
Thu Jan 14 08:13:05 2016 local7.debug : sock_getline() : "200 SixXS TIC Service on nlhaa01.sixxs.net ready (https://www.sixxs.net)"
Thu Jan 14 08:13:05 2016 local7.debug : sock_printf()  : "client TIC/draft-00 AICCU/2007.01.15-console-linux Linux/4.1.15"
Thu Jan 14 08:13:05 2016 local7.debug : sock_getline() : "200 Client Identity accepted"
Thu Jan 14 08:13:05 2016 local7.debug : sock_printf()  : "get unixtime"
Thu Jan 14 08:13:05 2016 local7.debug : sock_getline() : "200 1452755585"
Thu Jan 14 08:13:05 2016 local7.debug : sock_printf()  : "username ****-SIXXS/T******"
Thu Jan 14 08:13:05 2016 local7.debug : sock_getline() : "200 ****-SIXXS/T****** choose your authentication challenge please"
Thu Jan 14 08:13:05 2016 local7.debug : sock_printf()  : "challenge md5"
Thu Jan 14 08:13:05 2016 local7.debug : sock_getline() : "200 *****************************"
Thu Jan 14 08:13:05 2016 local7.debug : sock_printf()  : "authenticate md5 ****************************"
Thu Jan 14 08:13:05 2016 local7.debug : sock_getline() : "200 Successfully logged in using md5 as ****-SIXXS (My Name)"
Thu Jan 14 08:13:05 2016 local7.debug : sock_printf()  : "tunnel list"
Thu Jan 14 08:13:05 2016 local7.debug : sock_getline() : "201 Listing tunnels"
Thu Jan 14 08:13:05 2016 local7.debug : sock_getline() : "T****** 2001:4dd0:ff00:199f::2 heartbeat decgn01"
Thu Jan 14 08:13:05 2016 local7.debug : sock_getline() : "202 <tunnel_id> <ipv6_endpoint> <ipv4_endpoint> <pop_name>"
Thu Jan 14 08:13:05 2016 local7.debug : sock_printf()  : "tunnel show T******"
Thu Jan 14 08:13:06 2016 local7.debug : sock_getline() : "201 Showing tunnel information for T******"
Thu Jan 14 08:13:06 2016 local7.debug : sock_getline() : "TunnelId: T******"
Thu Jan 14 08:13:06 2016 local7.debug : sock_getline() : "Type: 6in4-heartbeat"
Thu Jan 14 08:13:06 2016 local7.debug : sock_getline() : "IPv6 Endpoint: 2001:4dd0:ff00:199f::2"
Thu Jan 14 08:13:06 2016 local7.debug : sock_getline() : "IPv6 POP: 2001:4dd0:ff00:199f::1"
Thu Jan 14 08:13:06 2016 local7.debug : sock_getline() : "IPv6 PrefixLength: 64"
Thu Jan 14 08:13:06 2016 local7.debug : sock_getline() : "Tunnel MTU: 1280"
Thu Jan 14 08:13:06 2016 local7.debug : sock_getline() : "Tunnel Name: My First Tunnel"
Thu Jan 14 08:13:06 2016 local7.debug : sock_getline() : "POP Id: decgn01"
Thu Jan 14 08:13:06 2016 local7.debug : sock_getline() : "IPv4 Endpoint: heartbeat"
Thu Jan 14 08:13:06 2016 local7.debug : sock_getline() : "IPv4 POP: 78.35.24.124"
Thu Jan 14 08:13:06 2016 local7.debug : sock_getline() : "UserState: enabled"
Thu Jan 14 08:13:06 2016 local7.debug : sock_getline() : "AdminState: enabled"
Thu Jan 14 08:13:06 2016 local7.debug : sock_getline() : "Password: ***********************"
Thu Jan 14 08:13:06 2016 local7.debug : sock_getline() : "Heartbeat_Interval: 60"
Thu Jan 14 08:13:06 2016 local7.debug : sock_getline() : "202 Done"
Thu Jan 14 08:13:06 2016 local7.info : Succesfully retrieved tunnel information for T******
Thu Jan 14 08:13:06 2016 local7.debug : sock_printf()  : "QUIT We just stopped breating"
Thu Jan 14 08:13:06 2016 daemon.notice netifd: wan6 (5267): Tunnel Information for T******:
Thu Jan 14 08:13:06 2016 daemon.notice netifd: wan6 (5267): POP Id      : decgn01
Thu Jan 14 08:13:06 2016 daemon.notice netifd: wan6 (5267): IPv6 Local  : 2001:4dd0:ff00:199f::2/64
Thu Jan 14 08:13:06 2016 daemon.notice netifd: wan6 (5267): IPv6 Remote : 2001:4dd0:ff00:199f::1/64
Thu Jan 14 08:13:06 2016 daemon.notice netifd: wan6 (5267): Tunnel Type : 6in4-heartbeat
Thu Jan 14 08:13:06 2016 daemon.notice netifd: wan6 (5267): Adminstate  : enabled
Thu Jan 14 08:13:06 2016 daemon.notice netifd: wan6 (5267): Userstate   : enabled
Thu Jan 14 08:13:06 2016 daemon.notice netifd: wan6 (5267): POP Id      : decgn01
Thu Jan 14 08:13:06 2016 daemon.notice netifd: wan6 (5267): IPv6 Local  : 2001:4dd0:ff00:199f::2/64
Thu Jan 14 08:13:06 2016 daemon.notice netifd: wan6 (5267): IPv6 Remote : 2001:4dd0:ff00:199f::1/64
Thu Jan 14 08:13:06 2016 daemon.notice netifd: wan6 (5267): Tunnel Type : 6in4-heartbeat
Thu Jan 14 08:13:06 2016 daemon.notice netifd: wan6 (5267): Adminstate  : enabled
Thu Jan 14 08:13:06 2016 daemon.notice netifd: wan6 (5267): Userstate   : enabled
Thu Jan 14 08:13:06 2016 local7.info : AICCU running as PID 5291
Thu Jan 14 08:13:06 2016 local7.debug : heartbeat_socket() - IPv4 : 2.245.129.113
Thu Jan 14 08:13:06 2016 daemon.notice netifd: Interface 'wan6' is now up
Thu Jan 14 08:13:06 2016 daemon.notice netifd: Network device 'aiccu-wan6' link is up
Thu Jan 14 08:13:06 2016 local7.debug : [HB] HEARTBEAT TUNNEL 2001:4dd0:ff00:199f::2 sender 1452755586 2eec8b78dea3784fbbebc05a3101111b
Thu Jan 14 08:13:06 2016 local7.debug : [HB] HEARTBEAT TUNNEL 2001:4dd0:ff00:199f::2 sender 1452755586 2eec8b78dea3784fbbebc05a3101111b
Thu Jan 14 08:13:06 2016 user.notice firewall: Reloading firewall due to ifup of wan6 (aiccu-wan6)
Thu Jan 14 08:14:06 2016 local7.debug : [HB] HEARTBEAT TUNNEL 2001:4dd0:ff00:199f::2 sender 1452755646 33289a7c631e4b6e61c9c52673c8e0a8
```

here my bad patch http://pastebin.com/DctkdHzF

hnyman, 2016-01-14:
> I managed to fix it. The problem was busybox ip "common/aiccu_linux.c" just use "ip" command to setup i just change it to "/usr/bin/ip"

You are using CC15.05, right?

This might already be fixed in trunk, thanks to reorganised path and ip (and ip-full) installation dirs: 
* Path was sanitized by https://dev.openwrt.org/changeset/47080/ . In trunk the default path should currently be "/usr/sbin:/usr/bin:/sbin:/bin" everywhere
* normal "ip" and "ip-full" installation directories were adjusted by https://dev.openwrt.org/changeset/47081 
* That should lead to situation where full ip is found first, then normal ip, and finally busybox ip.

But in Chaos Calmer 15.05 path is still a bit varying and the wrong ip might be found first.

Schabram, 2016-01-14:
no i use trunk first post was  r48233

oskar456, 2016-01-14:
I think root of the problem is that the BusyBox binary now includes the `ip` applet, which has been introduced in r46829. When running in busybox shell, these applets take absolute precedence regardless of path.

So I guess the only way around it is to patch a full path into the AICCU binary. There should be also some fix for exiting with an error so aiccu don't get restarted indefinitely.

hnyman, 2016-01-14:
> I think root of the problem is that the BusyBox binary now includes the ip applet, which has been introduced in r46829. When running in busybox shell, these applets take absolute precedence regardless of path.

I am not quite sure about that. At least when run derectly from shell, the plain "ip" command correctly runs the executable from the normal ip-route "ip".
```
root@OpenWrt:~# grep r4 /etc/banner
 DESIGNATED DRIVER (Bleeding Edge, r48226)
root@OpenWrt:~# busybox --help | grep 2016
BusyBox v1.24.1 (2016-01-13 12:30:18 EET) multi-call binary.
root@OpenWrt:~# which ip
/usr/bin/ip
root@OpenWrt:~# ip -V
ip utility, iproute2-ss4.3.0-1-openwrt
```
Haven't tried from inside an executable like aiccu, though.

oskar456, 2016-01-14:
@hnyman You are right. It really depends on PATH. The BusyBox symlink is is /sbin. The problem is that the aiccu is run by proto.sh hook, which is run by netifd. Inspecting environment of netifd running on my r47548 snapshot, I see `PATH=/bin:/sbin:/usr/sbin`. The question is, how to change the PATH environment variable in netifd.

hnyman, 2016-01-14:
@oskar456 
You might check your files that they contain the current path:
```PATH=/usr/sbin:/usr/bin:/sbin:/bin``` should be the path in all places:
/etc/preinit
/etc/profile
/sbin/hotplug-call

Your .config may also contain an old definition of path, but https://dev.openwrt.org/changeset/47590/ should have taken of that, (if you have re-generated .config since then)

But it is interesting, as I do not see your netifd's "PATH=/bin:/sbin:/usr/sbin" anywhere as the previous path definition (as it is missing /usr/bin which was in all other path definitions).

oskar456, 2016-01-14:
Oh sorry, that was a typo, the path I read from `/proc/<neifd pid>/environ` was `PATH=/bin:/sbin:/usr/bin:/usr/sbin`.
I've checked all the files you mentioned – the path is correct everywhere. So I dug one more level deeper to find out that the same PATH is used by `procd`, see `/proc/1/environ`.

I guess it's the early environment [hardcoded in netifd](http://git.openwrt.org/?p=project/procd.git;a=blob;f=initd/early.c;h=e87774f31cd0d44cc99e1ba123044f0ddf98b371;hb=HEAD#l93).

hnyman, 2016-01-14:
> I guess it's the early environment hardcoded in netifd.

... in "procd", you mean ;-) 
At least the links leads there. In cleartext:
http://git.openwrt.org/?p=project/procd.git;a=blob;f=initd/early.c;hb=HEAD#l93

That actually sounds a bit like a procd bug for me. To my knowledge, the goal of https://dev.openwrt.org/changeset/47080/ was to have an uniform path everywhere. But apparently procd still thinks otherwise for the "early" part.

oskar456, 2016-01-14:
Yeah, looks to me as `procd` bug… Already fixed, right now as it seems. Thanks @jow-  :+1:

Schabram, 2016-01-14:
Confirm is fixed with r48235 thanks without my patch.
Thanks.

hnyman, 2016-01-14:
Jow also backported that fix for CC15.05, but I think that other PATH fixes are not yet there, so everything might not work quite similarly.

To my knowledge, CC15.05 is still missing the backports of 
base-files fixes:
https://dev.openwrt.org/changeset/47080/ 
https://dev.openwrt.org/changeset/47590/
ip (iproute2) fixes:
https://dev.openwrt.org/changeset/46831/
https://dev.openwrt.org/changeset/47081/

oskar456, 2016-01-14:
Regarding this issue, it should make no difference. On stable CC 15.05 there's no BusyBox `ip` applet compiled in yet so the right version of `ip` is chosen.

hnyman, 2016-01-15:
This continues the "wrong path" discussion: I found one more place, where a wrong PATH can pop up. From dropbear itself.
I did ``` grep -r PATH /proc/*/environ ``` and found that all other processes had now the correct path, but the ash shell process itself had wrong path, although the user then saw correct path in his environment.
But there is an Openwrt patch for dropbear for setting the

yousong, 2016-09-08:
Seems that the issue has been fixed by unifying PATH env variable for searching ip command ;)

closed 2016-09-08 by yousong