openwrt/packages #1768: clone libvorbisidec error: Connection reset by peer

Love4Taylor, 2015-09-12:
```
echo "Checking out files from the git repository..."; mkdir -p /home/love4taylor/oxnas/trunk/tmp/dl && cd /home/love4taylor/oxnas/trunk/tmp/dl && rm -rf libvorbisidec-1.0.3-20150104 && [ \! -d libvorbisidec-1.0.3-20150104 ] && git clone git://git.xiph.org/tremor.git libvorbisidec-1.0.3-20150104 --recursive && (cd libvorbisidec-1.0.3-20150104 && git checkout b56ffce0c0773ec5ca04c466bc00b1bbcaf65aef && git submodule update) && echo "Packing checkout..." && rm -rf libvorbisidec-1.0.3-20150104/.git && 	tar czf /home/love4taylor/oxnas/trunk/tmp/dl/libvorbisidec-1.0.3-20150104.tar.gz libvorbisidec-1.0.3-20150104 && mv /home/love4taylor/oxnas/trunk/tmp/dl/libvorbisidec-1.0.3-20150104.tar.gz /home/love4taylor/oxnas/trunk/dl/ && rm -rf libvorbisidec-1.0.3-20150104; 
Checking out files from the git repository...
Cloning into 'libvorbisidec-1.0.3-20150104'...
fatal: read error: Connection reset by peer
make[2]: *** [/home/love4taylor/oxnas/trunk/dl/libvorbisidec-1.0.3-20150104.tar.gz] Error 128
```

i found package/feeds/packages/libvorbisidec/Makefile

```
PKG_NAME:=libvorbisidec
PKG_REV:=20150104
PKG_VERSION:=1.0.3-$(PKG_REV)
PKG_RELEASE:=1

PKG_SOURCE_PROTO:=git
PKG_SOURCE_URL:=git://git.xiph.org/tremor.git
```

must be ​https://git.xiph.org/tremor.git

yousong, 2016-09-08:
Please consider opening a pull request to do the switch if you think https is more preferable in general.

Actually issue #1704 was opened before this one with the intention to prefer git over https ;)

thess, 2016-09-08:
This is a bit ancient - The latest trunk version uses https since May. I probably should fix the for-15.05 branch too.

closed 2016-09-08 by thess