openwrt/packages #1769: freeradius2 - radiusd won't start at boot

czakey, 2015-09-12:
It seems that radiusd is not starting at boot. This bug been there for a long time. Init script seems to be oldish too.

Log indicates:
`Sat Sep 12 20:59:41 2015 user.emerg syslog: radiusd: Invalid IP Address or hostname "-p"`

Looking into /etc/init.d/radiusd there is
`IPADDR=$(ifconfig br-lan | sed -n 's/.*dr:\(.*\)Bc.*/\1/p')`
and then
`radiusd -i $IPADDR -p 1812,1813 $OPTIONS`

It seems that there is no valid IP address output from ifconfig. Changing `IPADDR=$(ifconfig br-lan | sed -n 's/.*dr:\(.*\)Bc.*/\1/p')` to something like `IPADDR=$(uci get network.lan.ipaddr)` solves the problem.

closed 2015-10-21 by dangowrt

haa-zee, 2015-11-24:
I'm using Chaos Calmer and I experienced the same.
opkg update && opkg list-upgradable output is empty...
Status is closed. Hm...

dangowrt, 2015-12-24:
We don't provide snapshot package builds for Chaos Calmer. You can use the SDK to re-compile the updated package and then update it on your device.

commit 1f312e49 (2015-10-21): freeradius2: cleanup init script

The initscript originally imported from oldpackages no longer does
the job, sed'ery around ifconfig is just not the way.
Remove stuff which can also be done via /etc/freeradius2/* or
/etc/default/radiusd instead.

Fixes #1769 and #1193