openwrt/packages #1771: fwknopd - can't set working interface.

grissone, 2015-09-14:
When I try to change interface on where the daemon has to listen to, default value appears on 'save and apply' (every time the daemon restart).
I took a look on /etc/init.d/fwknopd and I saw that it's working well if I change the get_conf func with

if [ "uci get fwknopd.@access[0].PCAP_INTF" = ""]
Rather than
if [ 'uci get fwknopd.@access[0].PCAP_INTF' = ""]
Thx

oneru, 2015-10-12:
@grissone Hey, sorry to take so long to look at this.  So, the issue is actually that the line you quoted looks in the access section, whereas PCAP_INTF is in the config section.  So as it is now, PCAP_INTF doesn't exist where this line is looking for it, so it always evaluates to true.  The single quote is correct. Once changed to a double quote, the uci command is never run, so it always evaluates to false.

I'll get this changed today. The corrected line is 
if [ 'uci get fwknopd.@config[0].PCAP_INTF' = ""]

oneru, 2015-12-03:
I believe this issue was fixed with commit d83625938616d9bb165f9d8f8d856f218164ac8c

dkadioglu, 2016-03-20:
Could this be backported to Chaos Calmer? I've applied the fix manually and it works as it should.

oneru, 2016-03-20:
@sbyx This has been fixed for a while, just never got backported to CC, as it's not a security fix.  What's the policy on backports, is this worthy of getting cherrypicked?

hnyman, 2016-03-21:
That is a small bug in the Openwrt scripts. Sure it can be backported to fix the package also in CC.

(I noticed that you have slightly tweaked the fix in the next version upgrade https://github.com/openwrt/packages/commit/8cc558b8a36eabbd21937092e6542dc36e5b6668 , so maybe that should be backported instead of cherrypicking just this?)

dkadioglu, 2016-05-08:
+1 for cherrypicking 8cc558b. Still I would be happy if this small fix could be backported. Can I somehow help? Should I create a pull request?

oneru, 2016-05-08:
*adds this to calendar for this afternoon*

oneru, 2016-05-08:
This fix is now in for-15.05.  Suggest close the issue.

closed 2016-05-08 by champtar