openwrt/packages #1777: make menuconfig error because of package 'sane-backends'

Love4Taylor, 2015-09-15:
```
love4taylor@ubuntu:~/openwrt/oxnas/trunk$ ./scripts/feeds install -a
Installing all packages from feed packages.
Installing package 'sane-backends' from packages
WARNING: No feed for package '(TARGET_x86||TARGET_x86_64)sane-qcam' found, maybe it's already part of the standard packages?
Installing all packages from feed luci.
Installing all packages from feed routing.
Installing all packages from feed telephony.
Installing all packages from feed management.
Installing all packages from feed targets.
Installing all packages from feed ramod.
```
then

```
love4taylor@ubuntu:~/openwrt/oxnas/trunk$ make menuconfig 
Collecting package info: done
tmp/.config-package.in:79429: syntax error
tmp/.config-package.in:79428: invalid option
make: *** [menuconfig] Error 1
```

./scripts/feeds uninstall sane-backends    can fix it

sbyx, 2015-09-15:
@luizluca  I merged that PR earlier but didn't see this issue and still cannot reproduce it.

@Love4Taylor have you tried deleting your "tmp" directory in your buildroot?

hnyman, 2015-09-15:
All buildbot builds have reported "failed shell 8" error at the "make defconfig" step since r46940, about 14 hours ago, mathcing roughly the merge moment of the sane PR #1724...

http://buildbot.openwrt.org:8010/grid

Example:
http://buildbot.openwrt.org:8010/builders/ar71xx.mikrotik/builds/86
http://buildbot.openwrt.org:8010/builders/ar71xx.mikrotik/builds/86/steps/shell_8/logs/stdio
```
Collecting package info: done
tmp/.config-package.in:79156: syntax error
tmp/.config-package.in:79155: invalid option
make: *** [defconfig] Error 1
```
I started looking for a reason for that, and the line numbers matched sane definitions in my /tmp/.config-package.in, so I found this issue report. But I haven't seen the error myself in my buildroot.

luizluca, 2015-09-15:
I'm in a travel and will not be able to check this right now.
I'm reverting

Sorry for the trouble

luizluca, 2015-09-16:
Is there an easy way to reproduce what buildbot does locally?
This bug doesn't seem to manifest in normal local build.

hnyman, 2015-09-16:
To my knowledge, buildbot just selects an architecture and the options to build all kmods and all packages. Then make defconfig will expand the minimal .config to full .config.

I tried to reproduce the issue locally, but did not succeed.

My hunch is that the last-minute addition of x86 dependency requirement for "qcam"-backend is the culprit. It gets wrongly parsed into a package definition in some conditions. 

So far the best clue on the error has been the bug tracker report at dev.openwrt.org, where the dependency clause has been strangely parsed by "scripts/feeds install" into middle of the "qcam" package name:
https://dev.openwrt.org/ticket/20524
```
$ ./scripts/feeds install -a
Installing all packages from feed packages.
Installing package 'sane-backends' from packages
WARNING: No feed for package '(TARGET_x86||TARGET_x86_64)sane-qcam' found, maybe it's already part of the standard packages?
...
$ make defconfig
Collecting package info: done
tmp/.config-package.in:79228: syntax error
tmp/.config-package.in:79227: invalid option
make: *** [defconfig] Error 1

File tmp/.config-package.in:79227 line 79227
...
select PACKAGE_(TARGET_x86||TARGET_x86_64)sane-qcam
```

When I look at my own tmp/config-package.in, the sane-backends-all section looks a bit weird, as there is a "depends" line in the middle of the backend packages listing, and depends is requiring NOT to be x86 (and it is checked twice???).

```
select PACKAGE_sane-dc25
select PACKAGE_sane-sm3840
select PACKAGE_sane-microtek
depends on !(((TARGET_x86||TARGET_x86_64))) || (TARGET_x86||TARGET_x86_64)
select PACKAGE_sane-epson
select PACKAGE_sane-kvs1025
select PACKAGE_sane-umax1220u
select PACKAGE_sane-hp5590
```

luizluca, 2015-09-17:
I have no luck on reproducing the bug:

```
$ git pull
Already up-to-date.
$ grep " packages" feeds.conf
src-git packages https://github.com/openwrt/packages.git^05840b78ec70a07ebc35ad701390234732ab9b63
$ make distclean
$ ./scripts/feeds update
$ ./scripts/feeds install -a
(no error)
$ make defconfig
(no error)

$ sed -n -r -e "/Package: sane-(backends|qcam)/,/^@@/{N;p}" feeds/packages.tmp/.packageinfo 
Package: sane-qcam
Depends: +libc +SSP_SUPPORT:libssp +USE_GLIBC:librt +USE_GLIBC:libpthread +libsane @(TARGET_x86||TARGET_x86_64)
...
@@

Package: sane-backends
@@
Config:
...  
      config PACKAGE_sane-qcam
      prompt "SANE backend (sane-qcam) for Connectix QuickCam cameras"
      depends on (TARGET_x86||TARGET_x86_64)   
...
@@

Package: sane-backends-all
Version: 1.0.24-1
Depends: +libc +SSP_SUPPORT:libssp +USE_GLIBC:librt +USE_GLIBC:libpthread +sane-abaton +sane-agfafocus +sane-apple +sane-artec +sane-artec_eplus48u +sane-as6e +sane-avision +sane-bh +sane-canon +sane-canon630u +sane-canon_dr +sane-cardscan +sane-coolscan +sane-coolscan2 +sane-coolscan3 +sane-dc210 +sane-dc240 +sane-dc25 +sane-dell1600n_net +sane-dmc +sane-epjitsu +sane-epson +sane-epson2 +sane-fujitsu +sane-genesys +sane-gt68xx +sane-hp +sane-hp3500 +sane-hp3900 +sane-hp4200 +sane-hp5400 +sane-hp5590 +sane-hpljm1005 +sane-hs2p +sane-ibm +sane-kodak +sane-kodakaio +sane-kvs20xx +sane-kvs40xx +sane-kvs1025 +sane-leo +sane-lexmark +sane-ma1509 +sane-magicolor +sane-matsushita +sane-microtek +sane-microtek2 +sane-mustek +sane-mustek_usb +sane-mustek_usb2 +sane-nec +sane-net +sane-niash +sane-p5 +sane-pie +sane-pixma +sane-plustek +sane-plustek_pp +((TARGET_x86||TARGET_x86_64)):sane-qcam +sane-ricoh +sane-rts8891 +sane-s9036 +sane-sceptre +sane-sharp +sane-sm3600 +sane-sm3840 +sane-snapscan +sane-sp15c +sane-st400 +sane-stv680 +sane-tamarack +sane-teco1 +sane-teco2 +sane-teco3 +sane-test +sane-u12 +sane-umax +sane-umax1220u +sane-umax_pp +sane-v4l +sane-xerox_mfp
@@
```

sane-qcam should depend on config @(TARGET_x86||TARGET_x86_64)
sane-backends-all should depend on sane-qcam only if @(TARGET_x86||TARGET_x86_64)

hnyman, 2015-09-18:
Could this be something build-system OS version specific? Some (old) host tool behaves differently?

Some ideas on the possible reasons:

When looking at the original reporters' reports, from Love4Taylor and bug20524 reporter, they both have got a wrong package name '(TARGET_x86||TARGET_x86_64)sane-qcam' parsed. So one pair of extra parentheses have been left unparsed, although the contents should have been parsed out and evaluated.

In my own buildroot, there was no error, but there was the line:

    depends on !(((TARGET_x86||TARGET_x86_64))) || (TARGET_x86||TARGET_x86_64)

That is always true, so it is an unnecessary line. 
But contents are also strange.
Note the triple parentheses around the left item. '!' probably adds one set, one is coming from original x86/x86_64 (and matches the right item), but where is the second set coming?
(and why is there such a null dependency line at all?)

In Luizluca's last example, the first dependency get parsed ok, But the second dependency set for sane-backends-all has again the extra parentheses:

    @(TARGET_x86||TARGET_x86_64)
    -->  depends on (TARGET_x86||TARGET_x86_64)
   
    +((TARGET_x86||TARGET_x86_64)):sane-qcam

My hunch is that the build system's evaluation logic gets confused in some situations when there are double parentheses inside each other, without any additional content, just ...((...))... . 
 
Probably the complex logic in Package/sane-backends/config and SaneBackendDependency behaves strangely on some systems and evaluates one nested @ dependency wrongly generating extra parentheses.
https://github.com/openwrt/packages/pull/1724/files#diff-d11b23474a4a7a7d504edae88c342bcaR53
https://github.com/openwrt/packages/pull/1724/files#diff-d11b23474a4a7a7d504edae88c342bcaR86 

The "Syntax error" & "invalid option" errors are probably from the function that splits (A, B, ...) to A, B and ..., which function now gets confused as there is only the single item A = "(TARGET_x86||TARGET_x86_64)" inside the outer parentheses. The function expects at least two components. Or something like that.

If the generation of the extra parentheses could be eliminated, the error might disappear.

But why the error does it materialise only on some systems, no idea :-(

Quick fixes might be 
* to forget about the sane-backends-all. That would remove the nested dependency
* Separate sane-backend-qcam from the rest. Leave it out from sane-backends-all and define it separately with a decidated X86 dependency.

luizluca, 2015-09-18:
@hnyman ,

The extra parentheses should not result in error. :-). The dependency of sane-backends-all is generated using function SaneBackendDependency , called for all backend subpackages. If one of these backends also includes a config dependency (@ANOTHER_CONFIG), it uses this as condition in order to not depend on a package that will not be available.

The original config dep from sane-qcam is:

```
DEPENDS:= ... @(TARGET_x86||TARGET_x86_64)
```

As I do not know how many config deps the backend package has, I join them with && and add parentheses around them. For the single config dep from sane-qcam, it returns:

```
DEPENDS:=... +((TARGET_x86||TARGET_x86_64)):sane-qcam
```

If sane-qcam had one more config dep like LINUX_4_1, it would be something like:

```
+((TARGET_x86||TARGET_x86_64)&&LINUX_4_1):sane-qcam
```

This is the reason for the second parentheses.

OpenWRT build converts conditional dependencies with a config dependency into:

```
depends on !("condition to consider the dependent package") || "config dependency from the dependent package"
```

As "condition to consider the dependent package" is equal to "config dependency from the dependent package", it generates the tautology (always true). When OpenWRT adds the "!" operator when generating the config file, it adds by itself another parentheses. And so, it gets the third parentheses:

```
depends on !(((TARGET_x86||TARGET_x86_64))) || (TARGET_x86||TARGET_x86_64)
```

@hnyman, thanks for the suggestions but I guess it is too risk to test a solution before being able to reproduce the problem (and verify that it really fix it). I would not like to break buildbots again ;-)

@Love4Taylor, can you describe your system, specially make version?

I'll try to build it with an ubuntu 14.04 to see if it hit the bug.

luizluca, 2015-09-18:
Update: the problem is system specific :-) I reproduced it with an ubuntu 14.04.

@Love4Taylor, thanks but I do not need more info anymore.

closed 2016-08-26 by hnyman