openwrt/packages #1781: Updating packages that are not touched for a long time..

miniminime, 2015-09-16:
Hi there,

A lot of the packages are not updated for a while. For example "iftop" is not updated since 2011! Not sure how to tackle this...maybe an "update package weekend" or something like this?

champtar, 2015-09-17:
hi @miniminime,
just send Pull Requests

Zokormazo, 2015-09-21:
iftop last stable release is from 2006

Anyway, I can't find iftop on this repository. :S

miniminime, 2015-09-21:
https://downloads.openwrt.org/chaos_calmer/15.05/x86/generic/packages/base/iftop_1.0pre2-1_x86.ipk (also available for other platforms)

The requirement is to only include packages that are stated "stable"? So, iftop has to be downgraded to the latest stable version then?

Zokormazo, 2015-09-21:
Oh, that package is on a different repository, not here.

Anyway, @jow- is the maintainer

Zokormazo, 2015-09-21:
@miniminime 

What package are the other outdated ones? Are there in this repository or on the base one like iftop? I could make some PRs if they are here.

BTW, just send a 1.0pre4 patch to @jow- , he'll decide merge or not :)

miniminime, 2015-09-21:
@Zokormazo, many thanks!

Maybe my approach is unfair or even totally wrong, but looking at the last changed date f.i. in these two folders I see a lot of untouched ones:

https://github.com/openwrt/packages/tree/master/utils
https://github.com/openwrt/packages/tree/master/net

Admittingly, I probably don't understand the update procedure for Openwrt overall. F.i. I understand you need to do a:

opkg update, then
opkg list-upgradable, 
and then you need to update every single package manually...not sure about security relevant updates, either. But I digress...  :-)

Noltari, 2015-09-21:
Looking at the last changed date isn't a very reliable method to be sure that those packages aren't updated to the latest release.
I prefer using http://sdwalker.github.io/uscan/
As you can see, not so many packages are outdated.

miniminime, 2015-09-21:
@Noltari, cool many thanks for that! 
63% are current for the packages....maybe I am reading it wrong...

yousong, 2016-09-08:
Close this for now as it received no input for nearly a year.  Thanks for bringing it up though ;)

closed 2016-09-08 by yousong