openwrt/packages #1784: Python2 import hashlib missing _sha256  _sha224 _sha256 _sha384  _sha512

chrisber, 2015-09-18:
Openwrt Source 15.05:latest
I have a compilation script that needs pythons `hashlib` library  which is missing some dependencies like `_ sha224`.

How can I fix this, I tried to add libopenssl and libsasl2 but this doesn't helped.

```sh
openwrt@3cc66242449a:~/openwrt$ ./staging_dir/host/bin/python
Python 2.7.9 (default, Sep 18 2015, 14:20:55) 
[GCC 4.8.4] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import hashlib
ERROR:root:code for hash sha224 was not found.
Traceback (most recent call last):
  File "/home/openwrt/openwrt/staging_dir/host/lib/python2.7/hashlib.py", line 147, in <module>
    globals()[__func_name] = __get_hash(__func_name)
  File "/home/openwrt/openwrt/staging_dir/host/lib/python2.7/hashlib.py", line 97, in __get_builtin_constructor
    raise ValueError('unsupported hash type ' + name)
ValueError: unsupported hash type sha224
ERROR:root:code for hash sha256 was not found.
Traceback (most recent call last):
  File "/home/openwrt/openwrt/staging_dir/host/lib/python2.7/hashlib.py", line 147, in <module>
    globals()[__func_name] = __get_hash(__func_name)
  File "/home/openwrt/openwrt/staging_dir/host/lib/python2.7/hashlib.py", line 97, in __get_builtin_constructor
    raise ValueError('unsupported hash type ' + name)
ValueError: unsupported hash type sha256
ERROR:root:code for hash sha384 was not found.
Traceback (most recent call last):
  File "/home/openwrt/openwrt/staging_dir/host/lib/python2.7/hashlib.py", line 147, in <module>
    globals()[__func_name] = __get_hash(__func_name)
  File "/home/openwrt/openwrt/staging_dir/host/lib/python2.7/hashlib.py", line 97, in __get_builtin_constructor
    raise ValueError('unsupported hash type ' + name)
ValueError: unsupported hash type sha384
ERROR:root:code for hash sha512 was not found.
Traceback (most recent call last):
  File "/home/openwrt/openwrt/staging_dir/host/lib/python2.7/hashlib.py", line 147, in <module>
    globals()[__func_name] = __get_hash(__func_name)
  File "/home/openwrt/openwrt/staging_dir/host/lib/python2.7/hashlib.py", line 97, in __get_builtin_constructor
    raise ValueError('unsupported hash type ' + name)
ValueError: unsupported hash type sha512
>>> 
```

commodo, 2015-09-18:
It looks like you're trying to build something for the (or with) host Python.

This looks like a good reason to remove the Python host build from the OpenWRT build and rely on the host's built-in Python.
I mean, the OpenWRT build does not start if you don't have Python on your system.
And maintaining the host python build, is starting to look like a hassle.

What if you replace  HostPython  from the staging_dir (i.e ./staging_dir/host/bin/python ) with just /usr/bin/python (or /usr/bin/python2 ) ?

In order to build hashlib for the OpenWRT host Python, you'd need an OpenWRT host version of libopenssl (and there doesn't seem to be one).
Looking here https://github.com/commodo/openwrt/blob/master/package/libs/openssl/Makefile
Seems that libopenssl does not have a HostBuild version.

I am not sure if the OpenWRT build system should pick up the libopenssl-dev from your main host system. (i.e. from /usr/include + /usr/lib).

Still, please try to use the host's python (i.e. /usr/bin/python )

commodo, 2015-10-10:
@chrisber 

Can you offer more details about your build host ?
What distro, etc

For me this works.
I'm building it on a Ubuntu VM system, and it works for me to import hashlib,

Thanks
Alex

artynet, 2015-11-07:
I have written this new Makefile for my custom distro :

https://gist.github.com/artynet/af47d13947a929cf60df

Basically I have borrowed the HOSTBUILD code from :

https://dev.openwrt.org/browser/trunk/package/libs/openssl/Makefile?rev=42988

and it seems to be working ;)

commodo, 2015-11-14:
@artynet @chrisber 
Sorry to take so long to look into this.
This was submitted sometime at the beginning of some vacation time so I deferred looking into it deeper.

The changes on the openssl host build look reasonable to me.
I was trying to figure out why there's a need for an openssl host build but then I started thinking about hosts with an openssl dev package that's too old (CentOS) or may have some missing bits (like the hashlb), so OpenWRT would find it, but the Python build may ignore it or not use it.

So, the openssl changes need to be submitted to OpenWRT core.
Which means sending patches via email to the openwrt-devel mailing list.

You can do that if you want, or I can do it if you want.
Whatever you prefer.

I do have a comment on this commit (regarding the Python package): https://github.com/linino/linino_CC/commit/dc82b825cfcb3d95b5b4808dc7ecbb9864c18283

Thanks
Alex

artynet, 2015-11-14:
Hello Alex, you are totally right. Since I was in a bit of a hurry for a deadline I decided to pursue that solution, however it turned out that it did not suit my needs. Altough the hashlib support was provided, the SSL one was still missing (when launching the host python and running "import ssl").

The best solution was, after all, to borrow the python sources from the main package feed so to override the ones from the `for-15.05` branch which seem to have fixed all the encountered problems (hashlib and ssl import).

Now everything works fine ;)

Regards, Arturo

JohannesRudolph, 2015-12-05:
Running into the same issue. We are running open-wrt 15.05 master builds (only custom config) + ``for-15.05`` package feed on our CI servers. Interestingly enough the same build used to work at least once (with exact same source revision) but has started to fail with exactly this error. Maybe related to parallel builds in some way?

Anyway, trying out the patch by @artynet

commodo, 2015-12-05:
Well, you could try the Python source version from trunk ?
Seems in trunk it's Python 2.7.10 and in for-15.05 it's 2.7.9.

Anyway, looking at the 2 branches, this looks like it probably fixes the issue
https://github.com/openwrt/packages/commit/73244f09ecf65e1722cfa10529f6c966f5fa8575

I think we could backport this for-15.05 and it shouldn't be too invasive.
Could you try with this patch ?

If it works, I can create a PR to backport it into for-15.05 (or you can do it if you want).

Thanks
Alex

artynet, 2015-12-05:
Hello @JohannesRudolph, @commodo is right. I overrode the python default version of chaos calmer (residing in the `for-15.05`) with the one from  the main trunk of the packages repo. The version is 2.7.10 as @commodo says and it is quite reliable (maybe is https://github.com/openwrt/packages/commit/73244f09ecf65e1722cfa10529f6c966f5fa8575 that solves the problem) . Now everything works smoothly !

I am now able to build all the python packages requiring ssl or hashlib support without any hassle. As I told in my previous post, I was in a bit in a hurry since I had a work deadline but there is no need at all to build a host version of openssl. Please get in touch with me if you need further help. Best, Arturo.

JohannesRudolph, 2015-12-05:
> 73244f0

Thanks, got it working. Had to add some additional Host Build dependencies
(expat) though, will submit my changes when back at work so we can
cross-check before upstreaming this.

I would need to try without host build of ssl.

commodo, 2015-12-05:
expat/host was also added in trunk
so, we could backport that too in 15.05

commodo, 2016-04-05:
@chrisber any objections on closing this ?

yousong, 2016-09-08:
Let's close this for now and reopen it if the issue persists.

closed 2016-09-08 by yousong