openwrt/packages #1787: error build tvheadend if global setings "full language support"

build000, 2015-09-19:
make[3]: Entering directory '/magazyn/Pliki/openwrt/trunk/feeds/packages/multimedia/tvheadend'
mkdir -p /magazyn/Pliki/openwrt/trunk/dl
echo "Checking out files from the git repository..."; mkdir -p /magazyn/Pliki/openwrt/trunk/tmp/dl && cd /magazyn/Pliki/openwrt/trunk/tmp/dl && rm -rf tvheadend && [ \! -d tvheadend ] && git clone https://github.com/tvheadend/tvheadend.git tvheadend --recursive && (cd tvheadend && git checkout v4.0.5 && git submodule update) && echo "Packing checkout..." && rm -rf tvheadend/.git && 	tar czf /magazyn/Pliki/openwrt/trunk/tmp/dl/tvheadend-v4.0.5.tar.gz tvheadend && mv /magazyn/Pliki/openwrt/trunk/tmp/dl/tvheadend-v4.0.5.tar.gz /magazyn/Pliki/openwrt/trunk/dl/ && rm -rf tvheadend; 
Checking out files from the git repository...
Cloning into 'tvheadend'...
remote: Counting objects: 50991, done.
remote: Compressing objects: 100% (206/206), done.
remote: Total 50991 (delta 114), reused 1 (delta 1), pack-reused 50784
Receiving objects: 100% (50991/50991), 30.57 MiB | 158.00 KiB/s, done.
Resolving deltas: 100% (37056/37056), done.
Checking connectivity... done.
Note: checking out 'v4.0.5'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by performing another checkout.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -b with the checkout command again. Example:

  git checkout -b new_branch_name

HEAD is now at ffc3ff1... channels: tags are saves as strings (uuid)
Packing checkout...
. /magazyn/Pliki/openwrt/trunk/include/shell.sh; gzip -dc /magazyn/Pliki/openwrt/trunk/dl/tvheadend-v4.0.5.tar.gz | tar -C /magazyn/Pliki/openwrt/trunk/build_dir/target-mips_34kc_musl-1.1.11/tvheadend/.. -xf - 
echo 'Tvheadend 4.0.5~openwrt2' > /magazyn/Pliki/openwrt/trunk/build_dir/target-mips_34kc_musl-1.1.11/tvheadend/debian/changelog
touch /magazyn/Pliki/openwrt/trunk/build_dir/target-mips_34kc_musl-1.1.11/tvheadend/.prepared_6b599935f15ec68f6c5c78958f242a92
rm -f /magazyn/Pliki/openwrt/trunk/staging_dir/target-mips_34kc_musl-1.1.11/stamp/.tvheadend_installed
(cd /magazyn/Pliki/openwrt/trunk/build_dir/target-mips_34kc_musl-1.1.11/tvheadend; rm -f aclocal.m4; if [ -f ./configure.ac ] || [ -f ./configure.in ]; then [ -d ./autom4te.cache ] && rm -rf autom4te.cache; [ -e ./config.rpath ] || ln -s /magazyn/Pliki/openwrt/trunk/scripts/config.rpath ./config.rpath; touch NEWS AUTHORS COPYING ABOUT-NLS ChangeLog; AUTOM4TE=/magazyn/Pliki/openwrt/trunk/staging_dir/host/bin/autom4te AUTOCONF=/magazyn/Pliki/openwrt/trunk/staging_dir/host/bin/autoconf AUTOMAKE=/magazyn/Pliki/openwrt/trunk/staging_dir/host/bin/automake ACLOCAL=/magazyn/Pliki/openwrt/trunk/staging_dir/host/bin/aclocal AUTOHEADER=/magazyn/Pliki/openwrt/trunk/staging_dir/host/bin/autoheader LIBTOOLIZE=/magazyn/Pliki/openwrt/trunk/staging_dir/host/bin/libtoolize LIBTOOL=/magazyn/Pliki/openwrt/trunk/staging_dir/host/bin/libtool M4=/magazyn/Pliki/openwrt/trunk/staging_dir/host/bin/m4 AUTOPOINT=true /magazyn/Pliki/openwrt/trunk/staging_dir/host/bin/autoreconf -v -f -i -s -B /magazyn/Pliki/openwrt/trunk/staging_dir/host/share/aclocal -I /magazyn/Pliki/openwrt/trunk/staging_dir/target-mips_34kc_musl-1.1.11/host/share/aclocal -I /magazyn/Pliki/openwrt/trunk/staging_dir/target-mips_34kc_musl-1.1.11/usr/share/aclocal -I m4 -I . . || true; fi; );
(cd /magazyn/Pliki/openwrt/trunk/build_dir/target-mips_34kc_musl-1.1.11/tvheadend/./; if [ -x ./configure ]; then find /magazyn/Pliki/openwrt/trunk/build_dir/target-mips_34kc_musl-1.1.11/tvheadend/ -name config.guess | xargs -r chmod u+w; find /magazyn/Pliki/openwrt/trunk/build_dir/target-mips_34kc_musl-1.1.11/tvheadend/ -name config.guess | xargs -r -n1 cp --remove-destination /magazyn/Pliki/openwrt/trunk/scripts/config.guess; find /magazyn/Pliki/openwrt/trunk/build_dir/target-mips_34kc_musl-1.1.11/tvheadend/ -name config.sub | xargs -r chmod u+w; find /magazyn/Pliki/openwrt/trunk/build_dir/target-mips_34kc_musl-1.1.11/tvheadend/ -name config.sub | xargs -r -n1 cp --remove-destination /magazyn/Pliki/openwrt/trunk/scripts/config.sub; AR="mips-openwrt-linux-musl-gcc-ar" AS="mips-openwrt-linux-musl-gcc -c -Os -pipe -mno-branch-likely -mips32r2 -mtune=34kc -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -msoft-float -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -I/magazyn/Pliki/openwrt/trunk/staging_dir/target-mips_34kc_musl-1.1.11/usr/lib/libiconv-full/include -I/magazyn/Pliki/openwrt/trunk/staging_dir/target-mips_34kc_musl-1.1.11/usr/lib/libintl-full/include" LD=mips-openwrt-linux-musl-ld NM="mips-openwrt-linux-musl-gcc-nm" CC="mips-openwrt-linux-musl-gcc" GCC="mips-openwrt-linux-musl-gcc" CXX="mips-openwrt-linux-musl-g++" RANLIB="mips-openwrt-linux-musl-gcc-ranlib" STRIP=mips-openwrt-linux-musl-strip OBJCOPY=mips-openwrt-linux-musl-objcopy OBJDUMP=mips-openwrt-linux-musl-objdump SIZE=mips-openwrt-linux-musl-size CFLAGS="-Os -pipe -mno-branch-likely -mips32r2 -mtune=34kc -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -msoft-float -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -I/magazyn/Pliki/openwrt/trunk/staging_dir/target-mips_34kc_musl-1.1.11/usr/lib/libiconv-full/include -I/magazyn/Pliki/openwrt/trunk/staging_dir/target-mips_34kc_musl-1.1.11/usr/lib/libintl-full/include " CXXFLAGS="-Os -pipe -mno-branch-likely -mips32r2 -mtune=34kc -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -msoft-float -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -I/magazyn/Pliki/openwrt/trunk/staging_dir/target-mips_34kc_musl-1.1.11/usr/lib/libiconv-full/include -I/magazyn/Pliki/openwrt/trunk/staging_dir/target-mips_34kc_musl-1.1.11/usr/lib/libintl-full/include " CPPFLAGS="-I/magazyn/Pliki/openwrt/trunk/staging_dir/target-mips_34kc_musl-1.1.11/usr/include -I/magazyn/Pliki/openwrt/trunk/staging_dir/target-mips_34kc_musl-1.1.11/include -I/magazyn/Pliki/openwrt/trunk/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_musl-1.1.11/usr/include -I/magazyn/Pliki/openwrt/trunk/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_musl-1.1.11/include/fortify -I/magazyn/Pliki/openwrt/trunk/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_musl-1.1.11/include -I/magazyn/Pliki/openwrt/trunk/staging_dir/target-mips_34kc_musl-1.1.11/usr/lib/libiconv-full/include -I/magazyn/Pliki/openwrt/trunk/staging_dir/target-mips_34kc_musl-1.1.11/usr/lib/libintl-full/include " LDFLAGS="-L/magazyn/Pliki/openwrt/trunk/staging_dir/target-mips_34kc_musl-1.1.11/usr/lib -L/magazyn/Pliki/openwrt/trunk/staging_dir/target-mips_34kc_musl-1.1.11/lib -L/magazyn/Pliki/openwrt/trunk/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_musl-1.1.11/usr/lib -L/magazyn/Pliki/openwrt/trunk/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_musl-1.1.11/lib -znow -zrelro -L/magazyn/Pliki/openwrt/trunk/staging_dir/target-mips_34kc_musl-1.1.11/usr/lib/libiconv-full/lib -Wl,-rpath-link=/magazyn/Pliki/openwrt/trunk/staging_dir/target-mips_34kc_musl-1.1.11/usr/lib/libiconv-full/lib -L/magazyn/Pliki/openwrt/trunk/staging_dir/target-mips_34kc_musl-1.1.11/usr/lib/libintl-full/lib -Wl,-rpath-link=/magazyn/Pliki/openwrt/trunk/staging_dir/target-mips_34kc_musl-1.1.11/usr/lib/libintl-full/lib "   ./configure --target=mips-openwrt-linux --host=mips-openwrt-linux --build=x86_64-linux-gnu --program-prefix="" --program-suffix="" --prefix=/usr --exec-prefix=/usr --bindir=/usr/bin --sbindir=/usr/sbin --libexecdir=/usr/lib --sysconfdir=/etc --datadir=/usr/share --localstatedir=/var --mandir=/usr/man --infodir=/usr/info    --disable-avahi --arch=mips --disable-dbus_1 --disable-libav --enable-bundle ; fi; )
Checking support/features
  checking for cc execinfo.h ...                    fail
  checking for cc -mmmx ...                         fail
  checking for cc -msse2 ...                        fail
  checking for cc getloadavg ...                    ok
  checking for cc atomic64 ...                      fail
  checking for cc lockowner ...                     fail
  checking for cc qsort_r ...                       fail
  checking for cc stime ...                         ok
  checking for cc recvmmsg ...                      ok
  checking for cc sendmmsg ...                      ok
  checking for cc libiconv ...                      ok
[... cut ...]
make[2]: Leaving directory '/magazyn/Pliki/openwrt/trunk'
package/Makefile:188: recipe for target '/magazyn/Pliki/openwrt/trunk/staging_dir/target-mips_34kc_musl-1.1.11/stamp/.package_compile' failed
make[1]: *** [/magazyn/Pliki/openwrt/trunk/staging_dir/target-mips_34kc_musl-1.1.11/stamp/.package_compile] Error 2
make[1]: Leaving directory '/magazyn/Pliki/openwrt/trunk'
/magazyn/Pliki/openwrt/trunk/include/toplevel.mk:179: polecenia dla obiektu 'world' nie powiodły się
make: *** [world] Błąd 2

sairon, 2015-11-05:
Should be fixed by 2849864c1ec62121c96701ca2566d17025297d03.

thess, 2015-11-05:
Yes, it is.

closed 2015-11-05 by thess