openwrt/packages #1814: multimedia: motion: segfault when starting / r47055 on ar71xx

bittorf, 2015-09-27:
maybe a musl thing?

```
root@box:~ motion
Segmentation fault

root@box:~ dmesg 
[  782.046652] 
[  782.046652] do_page_fault(): sending SIGSEGV to motion for invalid read access from 00000000
[  782.055165] epc = 77020610 in libc.so[76faa000+8f000]
[  782.060245] ra  = 004052ed in motion[400000+24000]
[  782.065187] 

root@box:~ ldd /usr/bin/motion
        /lib/ld-musl-mips-sf.so.1 (0x56448000)
        libjpeg.so.9 => /usr/lib/libjpeg.so.9 (0x77af0000)
        libgcc_s.so.1 => /lib/libgcc_s.so.1 (0x77acc000)
        libc.so => /lib/ld-musl-mips-sf.so.1 (0x56448000)
```

build000, 2015-09-27:
confirm:
root@revers:~# source /etc/profile
  _______                     ________        __
 |       |.-----.-----.-----.|  |  |  |.----.|  |_
 |   -   ||  _  |  -__|     ||  |  |  ||   _||   _|
 |_______||   __|_____|__|__||________||__|  |____|
          |__| W I R E L E S S   F R E E D O M
 -----------------------------------------------------
 DESIGNATED DRIVER (Bleeding Edge, r47058)
 -----------------------------------------------------
  * 2 oz. Orange Juice         Combine all juices in a
  * 2 oz. Pineapple Juice      tall glass filled with
  * 2 oz. Grapefruit Juice     ice, stir well.
  * 2 oz. Cranberry Juice
 -----------------------------------------------------
root@revers:~# 
[12523.777023] do_page_fault(): sending SIGSEGV to motion for invalid read access from 00000000
[12523.785713] epc = 77616610 in libc.so[775a0000+8f000]
[12523.790876] ra  = 004052ed in motion[400000+24000]
[12523.795907]

bittorf, 2015-09-27:
this also happens on x86 / UML - so easy testable...

bittorf, 2015-10-01:
compiling with uclibc does solve the issue - is anybody able to look deeper into this? should be report it to  the musl list? does 'motion' work e.g. with sabotage linux? any idea rogerdammit@gmail.com ?

roger-, 2015-10-07:
Hmm... sorry I'm not sure what could be wrong.

Can you raise the issue on https://github.com/Mr-Dave/motion?

jow-, 2015-10-08:
See #1847 for a possible solution.

build000, 2015-10-10:
when this solusion https://github.com/openwrt/packages/pull/1847 add to global ?

build000, 2015-10-21:
This patch https://github.com/openwrt/packages/commit/4714b0c9849de56ace62be2ea82c17bab856723b fixing problem - motion working again - thx  sbyx
:dancer: ... :-D

yousong, 2016-08-13:
Seems that the issue was solved ;)

closed 2016-08-13 by yousong