openwrt/packages #1818: forked-daapd in RC3 of CC, but not in release?

ejurgensen, 2015-09-28:
A user notified me that forked-daapd, which I maintain and for which I have submitted Makefile (https://github.com/openwrt/packages/tree/for-15.05/sound/forked-daapd), is not to be found in the final CC release:
https://downloads.openwrt.org/chaos_calmer/15.05/ar71xx/generic/packages/packages/

However, it was in RC3:
https://downloads.openwrt.org/chaos_calmer/15.05-rc3/ar71xx/generic/packages/packages/

I can't find any information about why it is not in the release? I searched tickets and issues here and on dev.openwrt.org, but there wasn't anything. It also didn't find any build errors on http://buildbot.openwrt.org:8010/broken_packages/ar71xx/

What might be the explanation?

Reference to user notification: https://github.com/ejurgensen/forked-daapd/issues/184

ejurgensen, 2015-10-15:
Anyone able to give me advice on this issue? @thess maybe?

thess, 2015-10-15:
I don't know why it isn't there - It builds OK in the 15.05 branch locally. I don't have access to any of the build systems which produced the 15.05 release so I cannot do much else. Perhaps @sbyx can help here.

ejurgensen, 2016-01-01:
Closing this, hopefully with the update whatever caused this won't happen again.

closed 2016-01-01 by ejurgensen