openwrt/packages #1819: Privoxy cannot start after sysupgrade to Chaos Calmer

elgatov, 2015-09-29:
* **Hardware:** TP-Link TL-WR1043N/ND v2
* **System:** OpenWrt Chaos Calmer 15.05 / LuCI (git-15.248.30277-3836b45) 

The system throws an error when trying to start privoxy through `/etc/init.d/privoxy start`:
```
chgrp: unknown group privoxy
chgrp: unknown group privoxy
chgrp: unknown group privoxy
chown: unknown user/group privoxy:privoxy
2015-09-29 14:03:21.135 77d6f000 Fatal error: Group 'privoxy' not found.
```

Seems like the script was not updated to reflect the changes made in the new privoxy package.

elgatov, 2015-09-29:
After removing the package, deleting left over files and reinstalling the problem is solved:

```sh
opkg remove privoxy
rm -rf /etc/privoxy
opkg update
opkg install privoxy
/etc/init.d/privoxy start
```

yousong, 2016-08-13:
This should already be fixed in OpenWrt/LEDE trunk (see commit log of `package/base-files/files/etc/uci-defaults/13_fix_group_user`).

closed 2016-08-13 by yousong