openwrt/packages #1820: msmtp v1.6.2 with ssl; mail package fails to compile

wwortel, 2015-09-29:
at least two msmtp-ssl package problems; managed to correct the first.
First:
The tarball contains a syntax error in configure.ac:
What should read
AM_GNU_GETTEXT="external"
is in the tarball
AM_GNU_GETTEXT([external)]
Changed it and made a new MD5 for further testing but still no luck.
Second:
the eror messages are:
Making all in po
make[6]: Entering directory '/var/data/OpenWrt/trunk/build_dir/target-mips_34kc_musl-1.1.11/msmtp-ssl/msmtp-1.6.2/po'
make[6]: * No rule to make target 'all'. Stop.

Further details:
compiling for ar71xx, revision r47062, trunk 28/9/2015

srgdts, 2016-01-30:
I confirm the issue.
```
autoreconf: Entering directory `.'
autoreconf: configure.ac: not using Gettext
autoreconf: running: /openwrt_15.05/staging_dir/host/bin/aclocal -I /openwrt_15.05/staging_dir/target-mips_34kc_uClibc-0.9.33.2/host/share/aclocal -I /openwrt_15.05/staging_dir/target-mips_34kc_uClibc-0.9.33.2/usr/share/aclocal -I m4 -I . --force 
configure.ac:45: warning: macro 'AM_GNU_GETTEXT' not found in library
autoreconf: configure.ac: tracing
autoreconf: configure.ac: not using Libtool
autoreconf: running: /openwrt_15.05/staging_dir/host/bin/autoconf --include=/openwrt_15.05/staging_dir/target-mips_34kc_uClibc-0.9.33.2/host/share/aclocal --include=/openwrt_15.05/staging_dir/target-mips_34kc_uClibc-0.9.33.2/usr/share/aclocal --include=m4 --include=. --prepend-include=/openwrt_15.05/staging_dir/host/share/aclocal --force
configure.ac:45: error: possibly undefined macro: AM_GNU_GETTEXT
      If this token and others are legitimate, please use m4_pattern_allow.
      See the Autoconf documentation.
.
.
.
./configure: line 5540: syntax error near unexpected token `external'
./configure: line 5540: `AM_GNU_GETTEXT(external)'
```
msmtp needs gettext as dependency.
**Solution:**
Include the ```libintl``` package.

thess, 2016-08-08:
Resolved by: 39f0b6f067217de83889110428890acad9accab4

closed 2016-08-08 by thess

SuperNami, 2017-05-11:
Which package resolves this issue on Debian Jessie?