openwrt/packages #1824: Fail to compile kmod sched cake with buildroot CC

asriadi, 2015-09-30:
Hi all

When I want add kmod sched cake into my firmware from cerowrt feeds I get this error

cc1: some warnings being treated as errors
make[5]: *** [/home/garfield/tesis/openwrt/OpenWRT-15.05-Chaos-Chalmer/build_dir/target-mips_34kc_uClibc-0.9.33.2/linux-ar71xx_generic/sched-cake-2015-git/sch_cake.o] Error 1
make[4]: *** [_module_/home/garfield/tesis/openwrt/OpenWRT-15.05-Chaos-Chalmer/build_dir/target-mips_34kc_uClibc-0.9.33.2/linux-ar71xx_generic/sched-cake-2015-git] Error 2
make[4]: Leaving directory `/home/garfield/tesis/openwrt/OpenWRT-15.05-Chaos-Chalmer/build_dir/target-mips_34kc_uClibc-0.9.33.2/linux-ar71xx_generic/linux-3.18.21'
make[3]: *** [/home/garfield/tesis/openwrt/OpenWRT-15.05-Chaos-Chalmer/build_dir/target-mips_34kc_uClibc-0.9.33.2/linux-ar71xx_generic/sched-cake-2015-git/.built] Error 2
make[3]: Leaving directory `/home/garfield/tesis/openwrt/OpenWRT-15.05-Chaos-Chalmer/feeds/cero/net/kmod-sched-cake'
make[2]: *** [package/feeds/cero/kmod-sched-cake/compile] Error 2
make[2]: Leaving directory `/home/garfield/tesis/openwrt/OpenWRT-15.05-Chaos-Chalmer'
make[1]: *** [/home/garfield/tesis/openwrt/OpenWRT-15.05-Chaos-Chalmer/staging_dir/target-mips_34kc_uClibc-0.9.33.2/stamp/.package_compile] Error 2
make[1]: Leaving directory `/home/garfield/tesis/openwrt/OpenWRT-15.05-Chaos-Chalmer'
make: *** [world] Error 2

I'm looking around for searching config.log but I didn't found it. Can anyone here help me with this problem? I'm sorry for my bad english

Best Regards
Riyadi

valentt, 2016-01-18:
Does kmod-sched-cake compile with latest code and latest openwrt trunk code?

hnyman, 2016-08-23:
This issue references a feed from "cerowrt", which is unsupported.

Cake (and the relevant patch for tc) has been imported to LEDE master and compiles just fine there. And the same package code compiles ok also for Openwrt trunk (if you manually create the cake Makefile and tc patch). If you want cake, you should use trunk.

closed 2016-08-23 by hnyman