openwrt/packages #1829: libubox buffer overflow when parsing messages

ronny-nilsson, 2015-10-01:
In libubux blobmsg.c:blobmsg_parse() there seems to be a buffer overflow issue. If the policy array (provided as an argument) is large an incorrect stack buffer is allocated. If the lib is compiled with gcc stack protection my system craches during boot... Provided is a patch which seems to fix the issue:
```
diff --git a/blobmsg.c b/blobmsg.c
index 80b984a..67647fc 100644
--- a/blobmsg.c
+++ b/blobmsg.c
@@ -130,11 +130,11 @@ int blobmsg_parse(const struct blobmsg_policy *policy, int policy_len,
 {
        struct blobmsg_hdr *hdr;
        struct blob_attr *attr;
-       uint8_t *pslen;
+       size_t *pslen;
        int i;

        memset(tb, 0, policy_len * sizeof(*tb));
-       pslen = alloca(policy_len);
+       pslen = alloca(policy_len * sizeof(size_t*));
        for (i = 0; i < policy_len; i++) {
                if (!policy[i].name)
                        continue;
```

nbd168, 2015-10-01:
This patch seems completely wrong to me. I also don't see any place where accesses to the pslen array could overflow.
Can you provide a test case to reproduce this issue?

ronny-nilsson, 2015-10-01:
You can easily reproduce by building with
```
CFLAGS += -fstack-protector
```
and provide a policy array argument of about 5-10 elements. The overflow occur at line 142 in HEAD
http://git.openwrt.org/?p=project/libubox.git;a=blob;f=blobmsg.c;h=80b984a133cd52f0d3a5fefe33de4e9739d99e71;hb=HEAD#l142

mkschreder, 2015-10-02:
Notice also how pslen[i] has garbage value if name of policy is null because it has never been initialized. Yes even though it is only used in following code if name is not null, this is still a bug waiting to happen and should not be written like that. 

Ronny, you haven't been by any chance calling blobbuf_free on blobbuf and then initing it again? Or maybe calling blobbuf_init without clearing memory first? Because you see blobbuf_init actually never initializes blobbuf struct properly if you allocate it on stack and its values are garbage. So then when you start using your blob buf all sorts of funny things happen. ;-) kenneth run into same problem as well before. it's stupid.

ronny-nilsson, 2015-10-14:
I back down from my issue report. I'm no longer able to reproduce it so there must have been something wrong with my build environment...
Sorry for bothering and wasting your time.

closed 2015-10-14 by ronny-nilsson