openwrt/packages #1844: MINIDLNA

himuracarter, 2015-10-06:
MiniDLNA doesn't detect ANY media files on 15.05-57-gdce0f18 Release (git-15.276.51144-8832d53) / OpenWrt Chaos Calmer 15.05
I use MiniDLNA installed on "overlay-extroot"
OpenWRT CC (latest git pull) custom build. MiniDLNA on extroot.
Previous builds were fine with DLNA.
By the way. libffmpeg-full failed to install due to "Collected errors:
satisfy_dependencies_for: Cannot satisfy the following dependencies for libffmpeg-full:
kernel (= 3.18.20-1-7bed08fa9c06eb8089e82c200340ec66) * kernel (= 3.18.20-1-7bed08fa9c06eb8089e82c200340ec66) *
opkg_install_cmd: Cannot install package libffmpeg-full."

qiuzi, 2015-10-12:
libffmpeg-mini

diizzyy, 2015-12-03:
Last time I used MiniDLNA I had to use ffmpeg-full otherwise it wouldn't find any media files at all, beware that you'll need swap if you have less than ~256Mb of RAM as it eats a lot of memory during scanning.

thess, 2015-12-04:
Not even MP3? I'm checking this out.

@antonlacon - any insight?

antonlacon, 2015-12-05:
For @himuracarter's issue, minidlna uses ffmpeg for media detection, so ffmpeg failing to install is why minidlna isn't recognizing media files.

For @diizzyy, we need to know what codec / container is being used for the media. For ffmpeg-full to work but ffmpeg-mini not to would point me towards media that isn't in ffmpeg-mini's support profile.

FYI, I'm overseas until Christmas, and have been since August, so won't be able to test / fix until then.

diizzyy, 2015-12-05:
It was quite a while ago since I tried -mini (1y+) but it didn't pick up was a bunch of clips in various formats, mainly mp4,mkv but also flac and mp3 if I recall correctly.

thess, 2015-12-18:
Tested minidlna w/ffmpeg-mini and had no problems with audio files. @diizzyy - without more information, I cannot work on your problem.
For now, I am closing the OP issue since as Ian said, ffmpeg libraries failed to install.

closed 2015-12-18 by thess