openwrt/packages #1848: f2fs tools 1.4.0 fails to format usb flash drive

sandymac, 2015-10-08:
In my efforts to use extroot with a f2fs filesystem I found that I could not format a USB flash drive with the OpenWrt provided mkfs.f2fs tool. I was able to format the same USB flash drive on a newly installed gentoo linux pc that was using f2fs tools 1.4.1.

Sample format failure on OpenWrt CC 15.05:
```
# mkfs.f2fs -l USB14G /dev/sda1

	F2FS-tools: mkfs.f2fs Ver: 1.4.0 (2014-09-18)

Info: Label = USB14G
Info: sector size = 512
Info: total sectors = 128929832984641536 (in 512bytes)
Info: zone aligned segment0 blkaddr: 256
Info: Discarding device
Info: This device doesn't support TRIM
	Error: Failed to Initialise the SIT AREA!!!
	Error: Could not format the device!!!
```

Different format fail with smaller partition size.
```
# mkfs.f2fs -l USB1G /dev/sda1

	F2FS-tools: mkfs.f2fs Ver: 1.4.0 (2014-09-18)

Info: Label = USB1G
Info: sector size = 512
Info: total sectors = 9007199254740992 (in 512bytes)
Info: zone aligned segment0 blkaddr: 256
Info: Discarding device
Info: This device doesn't support TRIM
	Error: Failed to create the root directory!!!
	Error: Could not format the device!!!
```

If I can figure out how to rebuild the f2fs tools package for my device I'll report back if the newer version works on OpenWrt.

FWIW @lperkov is listed as OpenWrt's package maintainer at https://github.com/openwrt/packages/tree/master/utils/f2fs-tools
It seems 1.4.1 is the current release according to http://git.kernel.org/cgit/linux/kernel/git/jaegeuk/f2fs-tools.git

lperkov, 2015-10-11:
If you have time please update a package and send a pull request. I'll add this to my queue to take a look...

sefgit, 2016-02-24:
Following patch fixes the above issue.
```
--- f2fs-tools-1.4.0/lib/libf2fs.c.ori  2016-02-24 17:06:54.836773994 +0700
+++ f2fs-tools-1.4.0/lib/libf2fs.c      2016-02-24 17:24:48.822607921 +0700
@@ -454,10 +454,11 @@
                        }
                }
 
-               if (ioctl(fd, BLKGETSIZE, &c->total_sectors) < 0) {
+               if (ioctl(fd, BLKGETSIZE64, &c->total_sectors) < 0) {
                        MSG(0, "\tError: Cannot get the device size\n");
                        return -1;
                }
+               c->total_sectors /= c->sector_size;
 
                if (ioctl(fd, HDIO_GETGEO, &geom) < 0)
                        c->start_sector = 0;
```

[For more info on how to get the size of a linux block special device in c](http://www.microhowto.info/howto/get_the_size_of_a_linux_block_special_device_in_c.html)

[How to build single package in OpenWRT](https://wiki.openwrt.org/doc/howtobuild/single.package)

lperkov, 2016-02-29:
Great. Please send this to mainline f2fs project for a review first and put a link here please.

vinibali, 2016-05-06:
Hi.
Is there any chance to update the f2fs-tools package at CC 15.05.1?
Fsck.f2fs is also broken and seems like there huge speed degradation if i format the data storage partition at PC with 1.5.0 or higher version.
I must reconnect manually the partition before copying a few 100s of MB to the flash drive.
Thanks

lperkov, 2016-05-18:
Please send pull request and link to where is visible that the patch above is checked by the upstream project.

bol-van, 2016-11-24:
fsck.f2fs from f2fstools 1.7.0 is heavily broken on ar71xx (mips32 big endian) platform
It corrupts valid f2fs partitions.

```
fsck.f2fs /dev/sda2
Info: Segments per section = 1
Info: Sections per zone = 1
Info: sector size = 512
Info: total sectors = 6508544 (3178 MB)
Info: MKFS version
  "Linux version 4.4.0-21-generic (buildd@lgw01-21) (gcc version 5.3.1 20160413 (Ubuntu 5.3.1-14ubuntu2) ) #37-Ubuntu SMP Mon Apr 18 18:33:37 UTC 2016"
Info: FSCK version
  from "Linux version 4.4.0-21-generic (buildd@lgw01-21) (gcc version 5.3.1 20160413 (Ubuntu 5.3.1-14ubuntu2) ) #37-Ubuntu SMP Mon Apr 18 18:33:37 UTC 2016"
    to "Linux version 4.4.30 (vasya@nowhere) (gcc version 5.4.0 (LEDE GCC 5.4.0 r2084) ) #0 PREEMPT Thu Nov 3 07:34:38 2016"
Info: superblock features = 0 : 
Info: superblock encrypt level = 0, salt = 00000000000000000000000000000000
Info: total FS sectors = 6508544 (3178 MB)
Info: CKPT version = 1
Info: checkpoint state = 5 :  compacted_summary unmount
[ASSERT] (sanity_check_nid: 356)  --> nid is not valid. [0x3]

NID[0x3] is unreachable
[FSCK] Unreachable nat entries                        [Fail] [0x1]
[FSCK] SIT valid block bitmap checking                [Fail]
[FSCK] Hard link checking for regular file            [Ok..] [0x0]
[FSCK] valid_block_count matching with CP             [Fail] [0x0]
[FSCK] valid_node_count matcing with CP (de lookup)   [Fail] [0x0]
[FSCK] valid_node_count matcing with CP (nat lookup)  [Ok..] [0x1]
[FSCK] valid_inode_count matched with CP              [Fail] [0x0]
[FSCK] free segment_count matched with CP             [Fail] [0x621]
[FSCK] next block offset is free                      [Ok..]
[FSCK] fixing SIT types
[FSCK] other corrupted bugs                           [Fail]
[FIX] (nullify_nat_entry:1688)  --> Remove nid [0x3] in nat journal

```

After that filesystem is broken and cant mount.

```
dmesg
F2FS-fs (sda2): Failed to read root inode

```

diizzyy, 2017-06-07:
@hnyman @thess 
This is fixed in 1.8.0, please close this.

closed 2017-06-07 by hnyman