openwrt/packages #1851: python: fix bz2 host build dependency

commodo, 2015-10-10:
Coming from these commit : https://github.com/openwrt/packages/commit/e461e7dda9d398005241ca175cc6b662914b988e and https://github.com/openwrt/packages/commit/6d892fb2fa6c6a41bf7dcfb00379f0197c684b28

Seems that the bz2 host build for Python is causing some issues.
Most likely this is something distro specific.

@vascorppor @chengg11 @wigyori   Can you offer more details about your host OSes ?
@blogic [curious/to clarify since I am vague on the details] This was added for some expat hostbuild dep-thingi ?

Looks like some people are experiencing some issues with this.
If you tell me your distro I can spin-up a VM and try + debug with that.
For me with Ubuntu it seems to work.

Thanks
Alex

chengg11, 2015-10-12:
hi in my case it is a ubuntu 14.04. My target system is a tplink archer c7 v2.

commodo, 2015-10-13:
I'll spin up a VM and try with that too.

If it's not too much trouble, can you try this patch to your OpenWRT build dir ?
https://github.com/commodo/openwrt/commit/cabfba9f1e1e4c8efadeed977fdc472fbe0e4663

@sbyx  could you please review this patch (seems you're the maintainer) ; maybe it makes sense to add it anyway to trunk;

In all cases libbz2.so is not installed in the staging_dir/host/lib folder.
And it could be that in some instances the build does not fail because host Python finds the host's libbz2.so in /lib/ (that's the system's lib folder)

I'll keep looking anyway

Thanks
Alex

chengg11, 2015-10-13:
I am in 15.05 release branch and this patch doesn't seem to apply there?

commodo, 2015-10-13:
oh; i'm on trunk :)
will try with 15.05 and spin up a patch based on 15.05

commodo, 2015-10-13:
hmm, so that patch should apply on 15.05 as far as I can tell;
you could try to download this into your openwrt folder:
http://rank24.net/~sandu/fw/0001-bzip2-extend-fix-the-Host-Install-rule-to-install-li.patch

example:
```
cd ~/work/openwrt-15.05/
wget http://rank24.net/~sandu/fw/0001-bzip2-extend-fix-the-Host-Install-rule-to-install-li.patch
patch -p1 < 0001-bzip2-extend-fix-the-Host-Install-rule-to-install-li.patch
```

chengg11, 2015-10-13:
ah, @commodo , my sincere apologies. 
It appears that I have not updated the openwrt base branch to the latest. ( I kind of didn't do that often after the 15.05 release was announced). 
That created a mismatch between the python feed and the base bzip2 util Makefiles if I understand correctly. That also explains why I couldn't do the patch.

Now after updating the base, I no longer have the build problem.

Sorry for wasting your time on this.

FWIW, with your patch, it builds too.

commodo, 2015-10-14:
@chengg11  no worries; glad it worked out :)

let's see for the rest of the people that have the problem
@vascorppor @wigyori ?

vascorppor, 2015-10-21:
Sorry for my late reply. I have now investigated further.

I'm on git repos `git://git.openwrt.org/15.05/openwrt.git` and `git://git.openwrt.org/15.05/packages.git`. To apply your patch, I had to upgrade (pull) said repos. Following the upgrade, python built just fine, even without your patch. I assume the bug has been addressed by another commit on Oct 1, 2015. The commit hash is 6a69e3064bdfa320f85d730c8d017f69c0215f6b, I got the info here: http://git.openwrt.org/?p=15.05/openwrt.git;a=commit;h=6a69e3064bdfa320f85d730c8d017f69c0215f6b. This seems to be in another repository, so github won't highlight it. I have also (for the sake of testing) reverted said commit (`git revert 6a69e3064bdfa320f85d730c8d017f69c0215f6b`) --> build fails.

So I'm quite confident the original bug I reported has been fixed by 6a69e3064bdfa320f85d730c8d017f69c0215f6b.

Anyway, thank you very much for your time, and sorry for my (faulty) original report. I guess I had updated one repo, but not the other. The multitude of openwrt git repos can be confusing to newbies like me.

commodo, 2015-10-21:
no worries;
glad it worked :)

that leaves  @wigyori to try out 
and we can see about closing this;

wigyori, 2015-10-25:
Right, so on my host (which is a debian 7), the bz2 python module still isn't being copied to the staging_dir. This is with today's trunk + the patch applied. Anyway, feel free to apply this one if not already. :)

commodo, 2015-10-25:
So, there was this PR from 2 days ago (approx):
https://github.com/openwrt/packages/pull/1886

Hmmm, so, if it's not too much to ask, could you try to convert your installation to a VM and share that with me so that I can work on ?
I understand that you could be hesitant to do this, since this could be a personal machine, or have personal stuff on it.
Maybe there could be a way to do this without exposing personal stuff.

If you need a place to upload it, I can provide you with an FTP account with sufficient space for a VM.

I'm a bit out of ideas right now on how to reproduce this issue.

Thanks
Alex

commodo, 2015-10-28:
@wigyori 
hmm, here's an interesting commit that might relate to this: https://github.com/NeoRaider/openwrt-packages/commit/659719005eac938cde961869c9cc12d7c4b7685c

commodo, 2015-11-16:
i'll close it for now;
feel free to re-open if this is still open

closed 2015-11-16 by commodo