openwrt/packages #1855: libgpg-error compile error with gcc5

avbohemen, 2015-10-12:
Libgpg-error does not compile with gcc5. No errors if I use gcc4.8-linaro. This is the output from build log:
```
make[3]: Entering directory `/home/anton/openwrt/archerc7/trunk/feeds/packages/libs/libgpg-error'
CFLAGS="-Os -pipe -mno-branch-likely -march=74kc -mtune=74kc -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -msoft-float -mips16 -minterlink-mips16 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -fpic  -I/home/anton/openwrt/archerc7/trunk/staging_dir/target-mips_34kc_musl-1.1.11/usr/include -I/home/anton/openwrt/archerc7/trunk/staging_dir/target-mips_34kc_musl-1.1.11/include -I/home/anton/openwrt/archerc7/trunk/staging_dir/toolchain-mips_34kc_gcc-5.2.0_musl-1.1.11/usr/include -I/home/anton/openwrt/archerc7/trunk/staging_dir/toolchain-mips_34kc_gcc-5.2.0_musl-1.1.11/include/fortify -I/home/anton/openwrt/archerc7/trunk/staging_dir/toolchain-mips_34kc_gcc-5.2.0_musl-1.1.11/include " CXXFLAGS="-Os -pipe -mno-branch-likely -march=74kc -mtune=74kc -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -msoft-float -mips16 -minterlink-mips16 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -fpic  -I/home/anton/openwrt/archerc7/trunk/staging_dir/target-mips_34kc_musl-1.1.11/usr/include -I/home/anton/openwrt/archerc7/trunk/staging_dir/target-mips_34kc_musl-1.1.11/include -I/home/anton/openwrt/archerc7/trunk/staging_dir/toolchain-mips_34kc_gcc-5.2.0_musl-1.1.11/usr/include -I/home/anton/openwrt/archerc7/trunk/staging_dir/toolchain-mips_34kc_gcc-5.2.0_musl-1.1.11/include/fortify -I/home/anton/openwrt/archerc7/trunk/staging_dir/toolchain-mips_34kc_gcc-5.2.0_musl-1.1.11/include " LDFLAGS="-L/home/anton/openwrt/archerc7/trunk/staging_dir/target-mips_34kc_musl-1.1.11/usr/lib -L/home/anton/openwrt/archerc7/trunk/staging_dir/target-mips_34kc_musl-1.1.11/lib -L/home/anton/openwrt/archerc7/trunk/staging_dir/toolchain-mips_34kc_gcc-5.2.0_musl-1.1.11/usr/lib -L/home/anton/openwrt/archerc7/trunk/staging_dir/toolchain-mips_34kc_gcc-5.2.0_musl-1.1.11/lib -znow -zrelro " make   -C /home/anton/openwrt/archerc7/trunk/build_dir/target-mips_34kc_musl-1.1.11/libgpg-error-1.12/. AR="mips-openwrt-linux-musl-gcc-ar" AS="mips-openwrt-linux-musl-gcc -c -Os -pipe -mno-branch-likely -march=74kc -mtune=74kc -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -msoft-float -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -fpic" LD=mips-openwrt-linux-musl-ld NM="mips-openwrt-linux-musl-gcc-nm" CC="mips-openwrt-linux-musl-gcc" GCC="mips-openwrt-linux-musl-gcc" CXX="mips-openwrt-linux-musl-g++" RANLIB="mips-openwrt-linux-musl-gcc-ranlib" STRIP=mips-openwrt-linux-musl-strip OBJCOPY=mips-openwrt-linux-musl-objcopy OBJDUMP=mips-openwrt-linux-musl-objdump SIZE=mips-openwrt-linux-musl-size CROSS="mips-openwrt-linux-musl-" ARCH="mips" ;
make[4]: Entering directory `/home/anton/openwrt/archerc7/trunk/build_dir/target-mips_34kc_musl-1.1.11/libgpg-error-1.12'
make  all-recursive
make[5]: Entering directory `/home/anton/openwrt/archerc7/trunk/build_dir/target-mips_34kc_musl-1.1.11/libgpg-error-1.12'
Making all in m4
make[6]: Entering directory `/home/anton/openwrt/archerc7/trunk/build_dir/target-mips_34kc_musl-1.1.11/libgpg-error-1.12/m4'
make[6]: Nothing to be done for `all'.
make[6]: Leaving directory `/home/anton/openwrt/archerc7/trunk/build_dir/target-mips_34kc_musl-1.1.11/libgpg-error-1.12/m4'
Making all in src
make[6]: Entering directory `/home/anton/openwrt/archerc7/trunk/build_dir/target-mips_34kc_musl-1.1.11/libgpg-error-1.12/src'
cc -I. -I. -o mkerrcodes ./mkerrcodes.c
In file included from ./mkerrcodes.c:26:0:
./mkerrcodes.h:9:5: error: expected expression before ',' token
   { , "GPG_ERR_E2BIG" },
     ^
./mkerrcodes.h:10:5: error: expected expression before ',' token
   { , "GPG_ERR_EACCES" },

./mkerrcodes.h:10:5: error: expected expression before ',' token
   { , "GPG_ERR_EACCES" },

<...error for every line...>

./mkerrcodes.h:133:5: error: expected expression before ',' token
   { , "GPG_ERR_EXDEV" },
     ^
./mkerrcodes.h:134:5: error: expected expression before ',' token
   { , "GPG_ERR_EXFULL" },
     ^
make[6]: *** [mkerrcodes] Error 1
```
The changelog of libgpg-error says from version 1.19 and above gcc5 is fixed.

stas2z, 2015-10-12:
Posted a temporary patch here a few time ago
https://dev.openwrt.org/ticket/20465

avbohemen, 2015-10-14:
The patch works. It should probably be attached here :)

hauke, 2015-11-18:
see #1990

hauke, 2015-11-20:
fixed in  #1990

closed 2015-11-20 by hauke