openwrt/packages #1856: libcoap not working on AR71xx

bailyzheng, 2015-10-12:
I remove --disable-examples option and build this lib on ar71xx. Finally I got coap-client. But I found this program run in wrong way. I captured by tcpdump and found this program send the wrong packet as follow:
Constrained Application Protocol, Confirmable, GET, MID:3795
00.. .... = Version: 0
..00 .... = Type: Confirmable (0)
.... 0001 = Token Length: 1
Code: GET (1)
Message ID: 3795
Token: b3
Expert Info (Warn/Malformed): Invalid Option Range: 4 (0 < x < 2)
Opt Name: #1: Observe: 1702232941
Expert Info (Warn/Malformed): Invalid Option Range: 4 (0 < x < 2)
Opt Name: #2: Content-Format: Unknown Type 1812099959

Any help is appreciated.

karlp, 2015-10-12:
Could you further elaborate on what you were expecting to happen?  Ie, what should it do on a working system?

bailyzheng, 2015-10-14:
I build libcoap on a x86_64 PC and coap-client work well. The packet is as follows when visit the same address as run on openwrt.
Constrained Application Protocol, Confirmable, GET, MID:35185
01.. .... = Version: 1
..00 .... = Type: Confirmable (0)
.... 0000 = Token Length: 0
Code: GET (1)
Message ID: 35185
Opt Name: #1: Uri-Path: dev
Opt Name: #2: Uri-Path: mdl
Opt Name: #3: Uri-Path: sw

There is no malformed option. It is correct that the version is 1 while the version is 0 on openwrt.

bailyzheng, 2015-10-14:
Maybe It is the big/little endian problem. I will have a test.

bailyzheng, 2015-10-14:
Yes, it is the endian problem. As for big endian platform, you should append  -DWORDS_BIGENDIAN to TARGET_CFLAGS.

Hope someone can see this issue if blocked by the same question.

karlp, 2015-10-14:
Can you try that pull request https://github.com/openwrt/packages/pull/1860 and see if it helps for you?

bailyzheng, 2015-10-23:
Hi Karlp, it works.

karlp, 2015-10-24:
the fix mentioned has already been merged.  Though there may be a fix upstream as well hopefully in the future

closed 2015-10-24 by karlp