openwrt/packages #1878: new node.js build error (WNDR4300)

build000, 2015-10-20:
make[3]: Entering directory '/magazyn/Pliki/openwrt/trunk/feeds/packages/lang/node'
. /magazyn/Pliki/openwrt/trunk/include/shell.sh; gzip -dc /magazyn/Pliki/openwrt/trunk/dl/node-v0.12.7.tar.gz | tar -C /magazyn/Pliki/openwrt/trunk/build_dir/target-mips_34kc_musl-1.1.11/node-v0.12.7/.. -xf - 

Applying ./patches/001-mips-no-fpu.patch using plaintext: 
patching file deps/v8/build/toolchain.gypi

Applying ./patches/002-addr_info.patch using plaintext: 
patching file deps/uv/src/unix/getaddrinfo.c

Applying ./patches/003-path.patch using plaintext: 
patching file lib/module.js
touch /magazyn/Pliki/openwrt/trunk/build_dir/target-mips_34kc_musl-1.1.11/node-v0.12.7/.prepared_36cdb1ea085bfabdefc2f604d9a6f258
rm -f /magazyn/Pliki/openwrt/trunk/staging_dir/target-mips_34kc_musl-1.1.11/stamp/.node_installed
(cd /magazyn/Pliki/openwrt/trunk/build_dir/target-mips_34kc_musl-1.1.11/node-v0.12.7/./; if [ -x ./configure ]; then find /magazyn/Pliki/openwrt/trunk/build_dir/target-mips_34kc_musl-1.1.11/node-v0.12.7/ -name config.guess | xargs -r chmod u+w; find /magazyn/Pliki/openwrt/trunk/build_dir/target-mips_34kc_musl-1.1.11/node-v0.12.7/ -name config.guess | xargs -r -n1 cp --remove-destination /magazyn/Pliki/openwrt/trunk/scripts/config.guess; find /magazyn/Pliki/openwrt/trunk/build_dir/target-mips_34kc_musl-1.1.11/node-v0.12.7/ -name config.sub | xargs -r chmod u+w; find /magazyn/Pliki/openwrt/trunk/build_dir/target-mips_34kc_musl-1.1.11/node-v0.12.7/ -name config.sub | xargs -r -n1 cp --remove-destination /magazyn/Pliki/openwrt/trunk/scripts/config.sub; AR="mips-openwrt-linux-musl-gcc-ar" AS="mips-openwrt-linux-musl-gcc -c -Os -pipe -mno-branch-likely -mips32r2 -mtune=34kc -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -msoft-float -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" LD=mips-openwrt-linux-musl-ld NM="mips-openwrt-linux-musl-gcc-nm" CC="mips-openwrt-linux-musl-gcc" GCC="mips-openwrt-linux-musl-gcc" CXX="mips-openwrt-linux-musl-g++" RANLIB="mips-openwrt-linux-musl-gcc-ranlib" STRIP=mips-openwrt-linux-musl-strip OBJCOPY=mips-openwrt-linux-musl-objcopy OBJDUMP=mips-openwrt-linux-musl-objdump SIZE=mips-openwrt-linux-musl-size CFLAGS="-Os -pipe -mno-branch-likely -mips32r2 -mtune=34kc -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -msoft-float -mips16 -minterlink-mips16 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro " CXXFLAGS="-Os -pipe -mno-branch-likely -mips32r2 -mtune=34kc -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -msoft-float -mips16 -minterlink-mips16 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro " CPPFLAGS="-I/magazyn/Pliki/openwrt/trunk/staging_dir/target-mips_34kc_musl-1.1.11/usr/include -I/magazyn/Pliki/openwrt/trunk/staging_dir/target-mips_34kc_musl-1.1.11/include -I/magazyn/Pliki/openwrt/trunk/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_musl-1.1.11/usr/include -I/magazyn/Pliki/openwrt/trunk/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_musl-1.1.11/include/fortify -I/magazyn/Pliki/openwrt/trunk/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_musl-1.1.11/include " LDFLAGS="-L/magazyn/Pliki/openwrt/trunk/staging_dir/target-mips_34kc_musl-1.1.11/usr/lib -L/magazyn/Pliki/openwrt/trunk/staging_dir/target-mips_34kc_musl-1.1.11/lib -L/magazyn/Pliki/openwrt/trunk/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_musl-1.1.11/usr/lib -L/magazyn/Pliki/openwrt/trunk/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_musl-1.1.11/lib -znow -zrelro "   ./configure --dest-cpu="mips" --dest-os=linux --without-snapshot --shared-zlib --shared-openssl --prefix=/usr ; fi; )
creating  ./icu_config.gypi
{ 'target_defaults': { 'cflags': [ '-I/magazyn/Pliki/openwrt/trunk/staging_dir/target-mips_34kc_musl-1.1.11/usr/include'],
                       'default_configuration': 'Release',
                       'defines': [],
                       'include_dirs': [],
                       'libraries': [ '-lz',
                                      '-L/magazyn/Pliki/openwrt/trunk/staging_dir/target-mips_34kc_musl-1.1.11/usr/lib',
                                      '-lssl',
                                      '-lcrypto']},
  'variables': { 'clang': 0,
                 'gcc_version': 48,
                 'host_arch': 'mips',
                 'icu_small': 'false',
                 'node_install_npm': 'true',
                 'node_prefix': '/usr',
                 'node_shared_cares': 'false',
                 'node_shared_http_parser': 'false',
                 'node_shared_libuv': 'false',
                 'node_shared_openssl': 'true',
                 'node_shared_v8': 'false',
                 'node_shared_zlib': 'true',
                 'node_tag': '',
                 'node_use_dtrace': 'false',
                 'node_use_etw': 'false',
                 'node_use_mdb': 'false',
                 'node_use_openssl': 'true',
                 'node_use_perfctr': 'false',
                 'openssl_no_asm': 0,
                 'python': '/magazyn/Pliki/openwrt/trunk/staging_dir/host/bin/python',
                 'target_arch': 'mips',
                 'uv_library': 'static_library',
                 'uv_parent_path': '/deps/uv/',
                 'uv_use_dtrace': 'false',
                 'v8_enable_gdbjit': 0,
                 'v8_enable_i18n_support': 0,
                 'v8_no_strict_aliasing': 1,
                 'v8_optimized_debug': 0,
                 'v8_random_seed': 0,
                 'v8_use_snapshot': 'false',
                 'want_separate_host_toolset': 0}}
creating  ./config.gypi
creating  ./config.mk
rm -f /magazyn/Pliki/openwrt/trunk/build_dir/target-mips_34kc_musl-1.1.11/node-v0.12.7/.configured_*
touch /magazyn/Pliki/openwrt/trunk/build_dir/target-mips_34kc_musl-1.1.11/node-v0.12.7/.configured_yyyyyyyyy
CFLAGS="-Os -pipe -mno-branch-likely -mips32r2 -mtune=34kc -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -msoft-float -mips16 -minterlink-mips16 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/magazyn/Pliki/openwrt/trunk/staging_dir/target-mips_34kc_musl-1.1.11/usr/include -I/magazyn/Pliki/openwrt/trunk/staging_dir/target-mips_34kc_musl-1.1.11/include -I/magazyn/Pliki/openwrt/trunk/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_musl-1.1.11/usr/include -I/magazyn/Pliki/openwrt/trunk/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_musl-1.1.11/include/fortify -I/magazyn/Pliki/openwrt/trunk/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_musl-1.1.11/include " CXXFLAGS="-Os -pipe -mno-branch-likely -mips32r2 -mtune=34kc -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -msoft-float -mips16 -minterlink-mips16 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/magazyn/Pliki/openwrt/trunk/staging_dir/target-mips_34kc_musl-1.1.11/usr/include -I/magazyn/Pliki/openwrt/trunk/staging_dir/target-mips_34kc_musl-1.1.11/include -I/magazyn/Pliki/openwrt/trunk/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_musl-1.1.11/usr/include -I/magazyn/Pliki/openwrt/trunk/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_musl-1.1.11/include/fortify -I/magazyn/Pliki/openwrt/trunk/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_musl-1.1.11/include " LDFLAGS="-L/magazyn/Pliki/openwrt/trunk/staging_dir/target-mips_34kc_musl-1.1.11/usr/lib -L/magazyn/Pliki/openwrt/trunk/staging_dir/target-mips_34kc_musl-1.1.11/lib -L/magazyn/Pliki/openwrt/trunk/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_musl-1.1.11/usr/lib -L/magazyn/Pliki/openwrt/trunk/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_musl-1.1.11/lib -znow -zrelro " make -j1 -C /magazyn/Pliki/openwrt/trunk/build_dir/target-mips_34kc_musl-1.1.11/node-v0.12.7/. AR="mips-openwrt-linux-musl-gcc-ar" AS="mips-openwrt-linux-musl-gcc -c -Os -pipe -mno-branch-likely -mips32r2 -mtune=34kc -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -msoft-float -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" LD=mips-openwrt-linux-musl-ld NM="mips-openwrt-linux-musl-gcc-nm" CC="mips-openwrt-linux-musl-gcc" GCC="mips-openwrt-linux-musl-gcc" CXX="mips-openwrt-linux-musl-g++" RANLIB="mips-openwrt-linux-musl-gcc-ranlib" STRIP=mips-openwrt-linux-musl-strip OBJCOPY=mips-openwrt-linux-musl-objcopy OBJDUMP=mips-openwrt-linux-musl-objdump SIZE=mips-ope [... cut ...]
package/Makefile:188: recipe for target '/magazyn/Pliki/openwrt/trunk/staging_dir/target-mips_34kc_musl-1.1.11/stamp/.package_compile' failed
make[1]: *** [/magazyn/Pliki/openwrt/trunk/staging_dir/target-mips_34kc_musl-1.1.11/stamp/.package_compile] Error 2
make[1]: Leaving directory '/magazyn/Pliki/openwrt/trunk'
/magazyn/Pliki/openwrt/trunk/include/toplevel.mk:177: polecenia dla obiektu 'world' nie powiodły się
make: *** [world] Błąd 2
[16:29:33] slawek@slawek-dom:/magazyn/Pliki/openwrt/trunk$

jow-, 2015-10-20:
Can you retry with `PKG_USE_MIPS16:=0` ?

ralphtheninja, 2015-10-20:
May I ask why you are using node 0.12 line rather than the latest node 4?

jow-, 2015-10-20:
Most likely because the author didn't succeed in cross building node 4.

build000, 2015-10-21:
Use this scope to Makefile, like this:
[...]
include $(TOPDIR)/rules.mk

PKG_NAME:=node
PKG_VERSION:=v0.12.7
PKG_RELEASE:=1
PKG_USE_MIPS16:=0

PKG_SOURCE:=node-$(PKG_VERSION).tar.gz
[...]
...and builds properly - full log builds this package (only package "node"), like this:
http://wklej.org/id/1821739/
Problem fixing.
Will there be added to appropriate fix for this architecture, or this router, or global in branches of "master" openwrt ?
What about the other packages node-hid, node-serialport, node-cylon ?

build000, 2015-10-26:
https://github.com/openwrt/packages/commit/8ca186cd593479d0b6b51455a355287cb2317d88 - fixing build node, or not fixing build node-hid, node-serialport, node-cylon

karlp, 2016-05-24:
if _node_ is now building ok, can we close this ticket?  in general mips16 bugs will go away eventually with newer compilers, and it's not automatic that other node apps will fail just because node core did.  Can we just file new tickets for the others if we need them?

yousong, 2016-09-12:
Close this as the node.js build error is now fixed by disabling mips16

closed 2016-09-12 by yousong