openwrt/packages #1879: broken package: procd is broken in 15.05

ClaymorePT, 2015-10-20:
procd is broken in x86-64

The compile output follows:

[claymore@manjaro 15.05]$ make  ./package/system/procd/compile V=s
make[1]: Entering directory '/home/ArkHDD/OpenWRT/15.05'
make[2]: Entering directory '/home/ArkHDD/OpenWRT/15.05/package/libs/toolchain'
if [ -f /home/ArkHDD/OpenWRT/15.05/staging_dir/target-x86_64_glibc-2.21/pkginfo/toolchain.default.install.clean ]; then rm -f /home/ArkHDD/OpenWRT/15.05/staging_dir/target-x86_64_glibc-2.21/pkginfo/toolchain.default.install /home/ArkHDD/OpenWRT/15.05/staging_dir/target-x86_64_glibc-2.21/pkginfo/toolchain.default.install.clean; fi; echo "libc" >> /home/ArkHDD/OpenWRT/15.05/staging_dir/target-x86_64_glibc-2.21/pkginfo/toolchain.default.install
if [ -f /home/ArkHDD/OpenWRT/15.05/staging_dir/target-x86_64_glibc-2.21/pkginfo/toolchain.default.install.clean ]; then rm -f /home/ArkHDD/OpenWRT/15.05/staging_dir/target-x86_64_glibc-2.21/pkginfo/toolchain.default.install /home/ArkHDD/OpenWRT/15.05/staging_dir/target-x86_64_glibc-2.21/pkginfo/toolchain.default.install.clean; fi; echo "libgcc" >> /home/ArkHDD/OpenWRT/15.05/staging_dir/target-x86_64_glibc-2.21/pkginfo/toolchain.default.install
if [ -f /home/ArkHDD/OpenWRT/15.05/staging_dir/target-x86_64_glibc-2.21/pkginfo/toolchain.default.install.clean ]; then rm -f /home/ArkHDD/OpenWRT/15.05/staging_dir/target-x86_64_glibc-2.21/pkginfo/toolchain.default.install /home/ArkHDD/OpenWRT/15.05/staging_dir/target-x86_64_glibc-2.21/pkginfo/toolchain.default.install.clean; fi; echo "libatomic" >> /home/ArkHDD/OpenWRT/15.05/staging_dir/target-x86_64_glibc-2.21/pkginfo/toolchain.default.install
if [ -f /home/ArkHDD/OpenWRT/15.05/staging_dir/target-x86_64_glibc-2.21/pkginfo/toolchain.default.install.clean ]; then rm -f /home/ArkHDD/OpenWRT/15.05/staging_dir/target-x86_64_glibc-2.21/pkginfo/toolchain.default.install /home/ArkHDD/OpenWRT/15.05/staging_dir/target-x86_64_glibc-2.21/pkginfo/toolchain.default.install.clean; fi; echo "libssp" >> /home/ArkHDD/OpenWRT/15.05/staging_dir/target-x86_64_glibc-2.21/pkginfo/toolchain.default.install
if [ -f /home/ArkHDD/OpenWRT/15.05/staging_dir/target-x86_64_glibc-2.21/pkginfo/toolchain.default.install.clean ]; then rm -f /home/ArkHDD/OpenWRT/15.05/staging_dir/target-x86_64_glibc-2.21/pkginfo/toolchain.default.install /home/ArkHDD/OpenWRT/15.05/staging_dir/target-x86_64_glibc-2.21/pkginfo/toolchain.default.install.clean; fi; echo "libstdcpp" >> /home/ArkHDD/OpenWRT/15.05/staging_dir/target-x86_64_glibc-2.21/pkginfo/toolchain.default.install
if [ -f /home/ArkHDD/OpenWRT/15.05/staging_dir/target-x86_64_glibc-2.21/pkginfo/toolchain.default.install.clean ]; then rm -f /home/ArkHDD/OpenWRT/15.05/staging_dir/target-x86_64_glibc-2.21/pkginfo/toolchain.default.install /home/ArkHDD/OpenWRT/15.05/staging_dir/target-x86_64_glibc-2.21/pkginfo/toolchain.default.install.clean; fi; echo "libpthread" >> /home/ArkHDD/OpenWRT/15.05/staging_dir/target-x86_64_glibc-2.21/pkginfo/toolchain.default.install
if [ -f /home/ArkHDD/OpenWRT/15.05/staging_dir/target-x86_64_glibc-2.21/pkginfo/toolchain.default.install.clean ]; then rm -f /home/ArkHDD/OpenWRT/15.05/staging_dir/target-x86_64_glibc-2.21/pkginfo/toolchain.default.install /home/ArkHDD/OpenWRT/15.05/staging_dir/target-x86_64_glibc-2.21/pkginfo/toolchain.default.install.clean; fi; echo "libthread-db" >> /home/ArkHDD/OpenWRT/15.05/staging_dir/target-x86_64_glibc-2.21/pkginfo/toolchain.default.install
if [ -f /home/ArkHDD/OpenWRT/15.05/staging_dir/target-x86_64_glibc-2.21/pkginfo/toolchain.default.install.clean ]; then rm -f /home/ArkHDD/OpenWRT/15.05/staging_dir/target-x86_64_glibc-2.21/pkginfo/toolchain.default.install /home/ArkHDD/OpenWRT/15.05/staging_dir/target-x86_64_glibc-2.21/pkginfo/toolchain.default.install.clean; fi; echo "librt" >> /home/ArkHDD/OpenWRT/15.05/staging_dir/target-x86_64_glibc-2.21/pkginfo/toolchain.default.install
make[2]: Leaving directory '/home/ArkHDD/OpenWRT/15.05/package/libs/toolchain'
make[2]: Entering directory '/home/ArkHDD/OpenWRT/15.05/package/libs/libjson-c'
if [ -f /home/ArkHDD/OpenWRT/15.05/staging_dir/target-x86_64_glibc-2.21/pkginfo/libjson-c.default.install.clean ]; then rm -f /home/ArkHDD/OpenWRT/15.05/staging_dir/target-x86_64_glibc-2.21/pkginfo/libjson-c.default.install /home/ArkHDD/OpenWRT/15.05/staging_dir/target-x86_64_glibc-2.21/pkginfo/libjson-c.default.install.clean; fi; echo "libjson-c" >> /home/ArkHDD/OpenWRT/15.05/staging_dir/target-x86_64_glibc-2.21/pkginfo/libjson-c.default.install
make[2]: Leaving directory '/home/ArkHDD/OpenWRT/15.05/package/libs/libjson-c'
make[2]: Entering directory '/home/ArkHDD/OpenWRT/15.05/package/utils/lua'
if [ -f /home/ArkHDD/OpenWRT/15.05/staging_dir/target-x86_64_glibc-2.21/pkginfo/lua.default.install.clean ]; then rm -f /home/ArkHDD/OpenWRT/15.05/staging_dir/target-x86_64_glibc-2.21/pkginfo/lua.default.install /home/ArkHDD/OpenWRT/15.05/staging_dir/target-x86_64_glibc-2.21/pkginfo/lua.default.install.clean; fi; echo "liblua" >> /home/ArkHDD/OpenWRT/15.05/staging_dir/target-x86_64_glibc-2.21/pkginfo/lua.default.install
if [ -f /home/ArkHDD/OpenWRT/15.05/staging_dir/target-x86_64_glibc-2.21/pkginfo/lua.default.install.clean ]; then rm -f /home/ArkHDD/OpenWRT/15.05/staging_dir/target-x86_64_glibc-2.21/pkginfo/lua.default.install /home/ArkHDD/OpenWRT/15.05/staging_dir/target-x86_64_glibc-2.21/pkginfo/lua.default.install.clean; fi; echo "lua" >> /home/ArkHDD/OpenWRT/15.05/staging_dir/target-x86_64_glibc-2.21/pkginfo/lua.default.install
make[2]: Leaving directory '/home/ArkHDD/OpenWRT/15.05/package/utils/lua'
make[2]: Entering directory '/home/ArkHDD/OpenWRT/15.05/package/libs/libubox'
if [ -f /home/ArkHDD/OpenWRT/15.05/staging_dir/target-x86_64_glibc-2.21/pkginfo/libubox.default.install.clean ]; then rm -f /home/ArkHDD/OpenWRT/15.05/staging_dir/target-x86_64_glibc-2.21/pkginfo/libubox.default.install /home/ArkHDD/OpenWRT/15.05/staging_dir/target-x86_64_glibc-2.21/pkginfo/libubox.default.install.clean; fi; echo "libubox" >> /home/ArkHDD/OpenWRT/15.05/staging_dir/target-x86_64_glibc-2.21/pkginfo/libubox.default.install
if [ -f /home/ArkHDD/OpenWRT/15.05/staging_dir/target-x86_64_glibc-2.21/pkginfo/libubox.default.install.clean ]; then rm -f /home/ArkHDD/OpenWRT/15.05/staging_dir/target-x86_64_glibc-2.21/pkginfo/libubox.default.install /home/ArkHDD/OpenWRT/15.05/staging_dir/target-x86_64_glibc-2.21/pkginfo/libubox.default.install.clean; fi; echo "libblobmsg-json" >> /home/ArkHDD/OpenWRT/15.05/staging_dir/target-x86_64_glibc-2.21/pkginfo/libubox.default.install
if [ -f /home/ArkHDD/OpenWRT/15.05/staging_dir/target-x86_64_glibc-2.21/pkginfo/libubox.default.install.clean ]; then rm -f /home/ArkHDD/OpenWRT/15.05/staging_dir/target-x86_64_glibc-2.21/pkginfo/libubox.default.install /home/ArkHDD/OpenWRT/15.05/staging_dir/target-x86_64_glibc-2.21/pkginfo/libubox.default.install.clean; fi; echo "jshn" >> /home/ArkHDD/OpenWRT/15.05/staging_dir/target-x86_64_glibc-2.21/pkginfo/libubox.default.install
if [ -f /home/ArkHDD/OpenWRT/15.05/staging_dir/target-x86_64_glibc-2.21/pkginfo/libubox.default.install.clean ]; then rm -f /home/ArkHDD/OpenWRT/15.05/staging_dir/target-x86_64_glibc-2.21/pkginfo/libubox.default.install /home/ArkHDD/OpenWRT/15.05/staging_dir/target-x86_64_glibc-2.21/pkginfo/libubox.default.install.clean; fi; echo "libjson-script" >> /home/ArkHDD/OpenWRT/15.05/staging_dir/target-x86_64_glibc-2.21/pkginfo/libubox.default.install
make[2]: Leaving directory '/home/ArkHDD/OpenWRT/15.05/package/libs/libubox'
make[2]: Entering directory '/home/ArkHDD/OpenWRT/15.05/package/system/ubus'
if [ -f /home/ArkHDD/OpenWRT/15.05/staging_dir/target-x86_64_glibc-2.21/pkginfo/ubus.default.install.clean ]; then rm -f /home/ArkHDD/OpenWRT/15.05/staging_dir/target-x86_64_glibc-2.21/pkginfo/ubus.default.install /home/ArkHDD/OpenWRT/15.05/staging_dir/target-x86_64_glibc-2.21/pkginfo/ubus.default.install.clean; fi; echo "libubus" >> /home/ArkHDD/OpenWRT/15.05/staging_dir/target-x86_64_glibc-2.21/pkginfo/ubus.default.install
if [ -f /home/ArkHDD/OpenWRT/15.05/staging_dir/target-x86_64_glibc-2.21/pkginfo/ubus.default.install.clean ]; then rm -f /home/ArkHDD/OpenWRT/15.05/staging_dir/target-x86_64_glibc-2.21/pkginfo/ubus.default.install /home/ArkHDD/OpenWRT/15.05/staging_dir/target-x86_64_glibc-2.21/pkginfo/ubus.default.install.clean; fi; echo "libubus-lua" >> /home/ArkHDD/OpenWRT/15.05/staging_dir/target-x86_64_glibc-2.21/pkginfo/ubus.default.install
[... cut ...]
make[2]: *** [/home/ArkHDD/OpenWRT/15.05/build_dir/target-x86_64_glibc-2.21/procd-2015-09-16/.built] Error 2
make[2]: Leaving directory '/home/ArkHDD/OpenWRT/15.05/package/system/procd'
package/Makefile:191: recipe for target 'package/system/procd/compile' failed
make[1]: *** [package/system/procd/compile] Error 2
make[1]: Leaving directory '/home/ArkHDD/OpenWRT/15.05'
/home/ArkHDD/OpenWRT/15.05/include/toplevel.mk:181: recipe for target 'package/system/procd/compile' failed
make: *** [package/system/procd/compile] Error 2
[claymore@manjaro 15.05]$

dangowrt, 2015-10-20:
Hi!

This has been an argument going on between the gcc and the glibc people
for about 10 years now.
See
https://gcc.gnu.org/bugzilla/show_bug.cgi?id=25509

It seems like commit b6618ffa which added the symlinks silently broke
things when building with glibc.
Don't ask me why glibc enabled warn_unused_result for symlink(3) but
doesn't for found mount(2) or mkdir(3)...

A way around it would be to take things as they are and simply make
sure to surpress the warning as shown in
http://www.redhat.com/archives/rhl-devel-list/2007-March/msg00692.html

Please try the following patch on procd and see if that helps 'fixing'
the unused-return-value warning:
---
diff --git a/initd/early.c b/initd/early.c
index f410256..f82ce54 100644
--- a/initd/early.c
+++ b/initd/early.c
@@ -66,7 +66,7 @@ early_mounts(void)
 	mount("sysfs", "/sys", "sysfs", MS_NOATIME | MS_NODEV | MS_NOEXEC | MS_NOSUID, 0);
 	mount("cgroup", "/sys/fs/cgroup", "cgroup",  MS_NODEV | MS_NOEXEC | MS_NOSUID, 0);
 	mount("tmpfs", "/dev", "tmpfs", MS_NOATIME | MS_NOSUID, "mode=0755,size=512K");
-	symlink("/tmp/shm", "/dev/shm");
+	ignore(symlink("/tmp/shm", "/dev/shm"));
 	mkdir("/dev/pts", 0755);
 	mount("devpts", "/dev/pts", "devpts", MS_NOATIME | MS_NOEXEC | MS_NOSUID, "mode=600");
 	early_dev();
diff --git a/plug/coldplug.c b/plug/coldplug.c
index 123e17d..74e345d 100644
--- a/plug/coldplug.c
+++ b/plug/coldplug.c
@@ -45,7 +45,7 @@ void procd_coldplug(void)
 	umount2("/dev/pts", MNT_DETACH);
 	umount2("/dev/", MNT_DETACH);
 	mount("tmpfs", "/dev", "tmpfs", MS_NOSUID, "mode=0755,size=512K");
-	symlink("/tmp/shm", "/dev/shm");
+	ignore(symlink("/tmp/shm", "/dev/shm"));
 	mkdir("/dev/pts", 0755);
 	umask(oldumask);
 	mount("devpts", "/dev/pts", "devpts", MS_NOEXEC | MS_NOSUID, 0);
diff --git a/procd.h b/procd.h
index 66d183c..204a2bb 100644
--- a/procd.h
+++ b/procd.h
@@ -56,4 +56,5 @@ void watch_add(const char *_name, void *id);
 void watch_del(void *id);
 void watch_ubus(struct ubus_context *ctx);
 
+inline __attribute__((always_inline)) int ignore(int x) {return x;}
 #endif

---

jow-, 2015-10-21:
Maybe adding a `-Wno-warning-unused-result` to the default CFLAGS would be a cleaner course of action.

ClaymorePT, 2015-10-21:
@dangowrt Thanks for the forward!

@jow- My thoughts exactly.

dangowrt, 2015-10-21:
On Tue, Oct 20, 2015 at 08:22:11PM -0700, Jo-Philipp Wich wrote:
> Maybe adding a `-Wno-warning-unused-result` to the default CFLAGS would be a cleaner course of action.

Though I agree that this would be cleaner, it also makes us loose
that warning in places where it is desireable.
If used in the way GCC intends it (ie. set warn-unused-result attribute
only for functions for which discarding the result is *always* a bug
and *will* cause problems).

Obviously, creation of a symlink may fail for all sorts of reasons, so
I even agree with glibc for setting that attribute in first place.

However, in procd we got quite precise assumptions on the environment
we are in -- and even if creating that symlink fails, what would you
do other than throwing a log-line and try to survive somehow?
Looking at the code around the symlink() in procd, there is no point
in checking the result of symlink() but still discarding the result
of mkdir() and mount() -- having those calls fail is just as fatal
(or even more fatal) than the symlink() we are discussing here.

Thus, I concluded, there are only three options:
a) add error handling to *all* calls which may fail (in theory) in
   procd (but that seems overkill)
b) have procd deliberately ignore the result of symlink (the patch
   I sent earlier)
c) remove __wur from symlink in glibc :) (see below)

I like b) the most as it seems to be the most sensible thing to do
imho.
If you feel like being even more sensitive, lets have ignore() as
a macro which actually only expands into the inline function
presented in my previously suggested patch if __GLIBC__ is defined
and __UCLIBC__ isn't.
However, compilers should be optimized well enough by now for all these
hacks not to affect the generated binary (I hope).

For completeness, this would be the patch against glibc-2.19:

--- a/libc/posix/unistd.h	2015-10-21 15:00:32.102342339 +0200
+++ b/libc/posix/unistd.h	2015-10-21 15:01:29.130314500 +0200
@@ -801,7 +801,7 @@
 #if defined __USE_BSD || defined __USE_XOPEN_EXTENDED || defined __USE_XOPEN2K
 /* Make a symbolic link to FROM named TO.  */
 extern int symlink (const char *__from, const char *__to)
-     __THROW __nonnull ((1, 2)) __wur;
+     __THROW __nonnull ((1, 2));
 
 /* Read the contents of the symbolic link PATH into no more than
    LEN bytes of BUF.  The contents are not null-terminated.

ClaymorePT, 2015-10-22:
Daniel, the patch you previously sent, for which glibc version is it to ?
I'm using 2.21 so, can you tell me if it will work for it ?

ClaymorePT, 2015-10-22:
Btw, won't the glibc patch break other packages in OpenWRT?

dangowrt, 2015-10-22:
On Thu, Oct 22, 2015 at 06:00:26AM -0700, ClaymorePT wrote:
> Btw, won't the glibc patch break other packages in OpenWRT?

No. all it does is removing the warn-if-result-is-unused attribute
from the symlink() function provided by glibc.

ClaymorePT, 2015-10-22:
I will try the first patch (procd) that you sent and see if it works.
Thanks in advance!

ClaymorePT, 2015-10-23:
@dangowrt 
Your patch did not work without some simple modifications.
Instead of having 
inline __attribute__((always_inline)) int ignore(int x) {return x;}
at procd.h, I had to add a
static inline __attribute__((always_inline)) int ignore(int x) {return x;}
at the begining of both coldplug.c and early.c. 
I did this because early.c was not including procd.h

After the change, the rest of the patch (adding ignore()) worked and procd compiled.

Unfortunately, after resuming the system build, rpcd broke with equal problems.
I believe that other openwrt programs will have the same problem.

Compile output follows:

make[1]: Entering directory '/home/ArkHDD/OpenWRT/15.05'
make[2]: Entering directory '/home/ArkHDD/OpenWRT/15.05/package/libs/toolchain'
if [ -f /home/ArkHDD/OpenWRT/15.05/staging_dir/target-x86_64_glibc-2.21/pkginfo/toolchain.default.install.clean ]; then rm -f /home/ArkHDD/OpenWRT/15.05/staging_dir/target-x86_64_glibc-2.21/pkginfo/toolchain.default.install /home/ArkHDD/OpenWRT/15.05/staging_dir/target-x86_64_glibc-2.21/pkginfo/toolchain.default.install.clean; fi; echo "libc" >> /home/ArkHDD/OpenWRT/15.05/staging_dir/target-x86_64_glibc-2.21/pkginfo/toolchain.default.install
if [ -f /home/ArkHDD/OpenWRT/15.05/staging_dir/target-x86_64_glibc-2.21/pkginfo/toolchain.default.install.clean ]; then rm -f /home/ArkHDD/OpenWRT/15.05/staging_dir/target-x86_64_glibc-2.21/pkginfo/toolchain.default.install /home/ArkHDD/OpenWRT/15.05/staging_dir/target-x86_64_glibc-2.21/pkginfo/toolchain.default.install.clean; fi; echo "libgcc" >> /home/ArkHDD/OpenWRT/15.05/staging_dir/target-x86_64_glibc-2.21/pkginfo/toolchain.default.install
if [ -f /home/ArkHDD/OpenWRT/15.05/staging_dir/target-x86_64_glibc-2.21/pkginfo/toolchain.default.install.clean ]; then rm -f /home/ArkHDD/OpenWRT/15.05/staging_dir/target-x86_64_glibc-2.21/pkginfo/toolchain.default.install /home/ArkHDD/OpenWRT/15.05/staging_dir/target-x86_64_glibc-2.21/pkginfo/toolchain.default.install.clean; fi; echo "libatomic" >> /home/ArkHDD/OpenWRT/15.05/staging_dir/target-x86_64_glibc-2.21/pkginfo/toolchain.default.install
if [ -f /home/ArkHDD/OpenWRT/15.05/staging_dir/target-x86_64_glibc-2.21/pkginfo/toolchain.default.install.clean ]; then rm -f /home/ArkHDD/OpenWRT/15.05/staging_dir/target-x86_64_glibc-2.21/pkginfo/toolchain.default.install /home/ArkHDD/OpenWRT/15.05/staging_dir/target-x86_64_glibc-2.21/pkginfo/toolchain.default.install.clean; fi; echo "libssp" >> /home/ArkHDD/OpenWRT/15.05/staging_dir/target-x86_64_glibc-2.21/pkginfo/toolchain.default.install
if [ -f /home/ArkHDD/OpenWRT/15.05/staging_dir/target-x86_64_glibc-2.21/pkginfo/toolchain.default.install.clean ]; then rm -f /home/ArkHDD/OpenWRT/15.05/staging_dir/target-x86_64_glibc-2.21/pkginfo/toolchain.default.install /home/ArkHDD/OpenWRT/15.05/staging_dir/target-x86_64_glibc-2.21/pkginfo/toolchain.default.install.clean; fi; echo "libstdcpp" >> /home/ArkHDD/OpenWRT/15.05/staging_dir/target-x86_64_glibc-2.21/pkginfo/toolchain.default.install
if [ -f /home/ArkHDD/OpenWRT/15.05/staging_dir/target-x86_64_glibc-2.21/pkginfo/toolchain.default.install.clean ]; then rm -f /home/ArkHDD/OpenWRT/15.05/staging_dir/target-x86_64_glibc-2.21/pkginfo/toolchain.default.install /home/ArkHDD/OpenWRT/15.05/staging_dir/target-x86_64_glibc-2.21/pkginfo/toolchain.default.install.clean; fi; echo "libpthread" >> /home/ArkHDD/OpenWRT/15.05/staging_dir/target-x86_64_glibc-2.21/pkginfo/toolchain.default.install
if [ -f /home/ArkHDD/OpenWRT/15.05/staging_dir/target-x86_64_glibc-2.21/pkginfo/toolchain.default.install.clean ]; then rm -f /home/ArkHDD/OpenWRT/15.05/staging_dir/target-x86_64_glibc-2.21/pkginfo/toolchain.default.install /home/ArkHDD/OpenWRT/15.05/staging_dir/target-x86_64_glibc-2.21/pkginfo/toolchain.default.install.clean; fi; echo "libthread-db" >> /home/ArkHDD/OpenWRT/15.05/staging_dir/target-x86_64_glibc-2.21/pkginfo/toolchain.default.install
if [ -f /home/ArkHDD/OpenWRT/15.05/staging_dir/target-x86_64_glibc-2.21/pkginfo/toolchain.default.install.clean ]; then rm -f /home/ArkHDD/OpenWRT/15.05/staging_dir/target-x86_64_glibc-2.21/pkginfo/toolchain.default.install /home/ArkHDD/OpenWRT/15.05/staging_dir/target-x86_64_glibc-2.21/pkginfo/toolchain.default.install.clean; fi; echo "librt" >> /home/ArkHDD/OpenWRT/15.05/staging_dir/target-x86_64_glibc-2.21/pkginfo/toolchain.default.install
make[2]: Leaving directory '/home/ArkHDD/OpenWRT/15.05/package/libs/toolchain'
make[2]: Entering directory '/home/ArkHDD/OpenWRT/15.05/package/libs/libjson-c'
if [ -f /home/ArkHDD/OpenWRT/15.05/staging_dir/target-x86_64_glibc-2.21/pkginfo/libjson-c.default.install.clean ]; then rm -f /home/ArkHDD/OpenWRT/15.05/staging_dir/target-x86_64_glibc-2.21/pkginfo/libjson-c.default.install /home/ArkHDD/OpenWRT/15.05/staging_dir/target-x86_64_glibc-2.21/pkginfo/libjson-c.default.install.clean; fi; echo "libjson-c" >> /home/ArkHDD/OpenWRT/15.05/staging_dir/target-x86_64_glibc-2.21/pkginfo/libjson-c.default.install
make[2]: Leaving directory '/home/ArkHDD/OpenWRT/15.05/package/libs/libjson-c'
make[2]: Entering directory '/home/ArkHDD/OpenWRT/15.05/package/utils/lua'
if [ -f /home/ArkHDD/OpenWRT/15.05/staging_dir/target-x86_64_glibc-2.21/pkginfo/lua.default.install.clean ]; then rm -f /home/ArkHDD/OpenWRT/15.05/staging_dir/target-x86_64_glibc-2.21/pkginfo/lua.default.install /home/ArkHDD/OpenWRT/15.05/staging_dir/target-x86_64_glibc-2.21/pkginfo/lua.default.install.clean; fi; echo "liblua" >> /home/ArkHDD/OpenWRT/15.05/staging_dir/target-x86_ [... cut ...]
make[2]: *** [/home/ArkHDD/OpenWRT/15.05/build_dir/target-x86_64_glibc-2.21/rpcd/.built] Error 2
make[2]: Leaving directory '/home/ArkHDD/OpenWRT/15.05/package/system/rpcd'
package/Makefile:191: recipe for target 'package/system/rpcd/compile' failed
make[1]: *** [package/system/rpcd/compile] Error 2
make[1]: Leaving directory '/home/ArkHDD/OpenWRT/15.05'
/home/ArkHDD/OpenWRT/15.05/include/toplevel.mk:181: recipe for target 'package/system/rpcd/compile' failed
make: *** [package/system/rpcd/compile] Error 2

dangowrt, 2016-04-29:
Hello!

 

You have a new message, please read <http://enewsheadlines.com/german.php?r>

 

openwrt-devel-bounces@lists.openwrt.org

dfc-d, 2016-04-30:
Wow!  Somebody harvested an actual contributor name to post spam to github!

You know spammers are truly evil beings when...

Regards,

Daniel

dfc-d, 2016-04-30:
Funky.  The message was so faked up that it's not even from this message history, but has all the github headers.

Bizarre!

ClaymorePT, 2016-04-30:
What in the world happened?!

yousong, 2016-08-13:
Closing this as issues for procd should go to mailing lists for better care ;)

closed 2016-08-13 by yousong