openwrt/packages #1881: /etc/init.d/transmission start not work on  CHAOS CALMER (15.05, r46767)

wxg4net, 2015-10-22:
but "/usr/bin/transmission-daemon -g /tmp/transmission/ -f" worked 
"logread | grep transmission" show me:
Thu Oct 22 19:19:50 2015 user.notice transmission: Starting with 37770000 virt mem

/etc/config/transmission:
transmission.@transmission[0]=transmission
transmission.@transmission[0].config_dir='/tmp/transmission'
transmission.@transmission[0].alt_speed_enabled='false'
transmission.@transmission[0].bind_address_ipv4='0.0.0.0'
transmission.@transmission[0].bind_address_ipv6='::'
transmission.@transmission[0].blocklist_enabled='false'
transmission.@transmission[0].dht_enabled='true'
transmission.@transmission[0].download_queue_enabled='true'
transmission.@transmission[0].download_queue_size='4'
transmission.@transmission[0].encryption='1'
transmission.@transmission[0].idle_seeding_limit_enabled='false'
transmission.@transmission[0].incomplete_dir_enabled='false'
transmission.@transmission[0].lazy_bitfield_enabled='true'
transmission.@transmission[0].lpd_enabled='false'
transmission.@transmission[0].message_level='3'
transmission.@transmission[0].peer_limit_global='240'
transmission.@transmission[0].peer_limit_per_torrent='60'
transmission.@transmission[0].peer_port='51413'
transmission.@transmission[0].peer_port_random_on_start='false'
transmission.@transmission[0].peer_socket_tos='default'
transmission.@transmission[0].pex_enabled='true'
transmission.@transmission[0].port_forwarding_enabled='true'
transmission.@transmission[0].preallocation='1'
transmission.@transmission[0].queue_stalled_enabled='true'
transmission.@transmission[0].queue_stalled_minutes='30'
transmission.@transmission[0].ratio_limit_enabled='false'
transmission.@transmission[0].rpc_bind_address='0.0.0.0'
transmission.@transmission[0].rpc_enabled='true'
transmission.@transmission[0].rpc_port='9091'
transmission.@transmission[0].rpc_url='/transmission/'
transmission.@transmission[0].scrape_paused_torrents_enabled='true'
transmission.@transmission[0].script_torrent_done_enabled='false'
transmission.@transmission[0].seed_queue_enabled='false'
transmission.@transmission[0].speed_limit_down_enabled='false'
transmission.@transmission[0].speed_limit_up_enabled='false'
transmission.@transmission[0].start_added_torrents='true'
transmission.@transmission[0].trash_original_torrent_files='false'
transmission.@transmission[0].upload_slots_per_torrent='14'
transmission.@transmission[0].utp_enabled='true'
transmission.@transmission[0].scrape_paused_torrents='true'
transmission.@transmission[0].watch_dir_enabled='false'
transmission.@transmission[0].enabled='1'
transmission.@transmission[0].download_dir='/mnt/data/DLNA/movie'
transmission.@transmission[0].rename_partial_files='false'
transmission.@transmission[0].rpc_whitelist_enabled='false'
transmission.@transmission[0].rpc_authentication_required='true'
transmission.@transmission[0].rpc_username='wxg'
transmission.@transmission[0].rpc_password='wxg'
transmission.@transmission[0].cache_size_mb='5'
transmission.@transmission[0].umask='0'
transmission.@transmission[0].user='root'
root@welcome:/tmp# uci show transmission 
transmission.@transmission[0]=transmission
transmission.@transmission[0].config_dir='/tmp/transmission'
transmission.@transmission[0].alt_speed_enabled='false'
transmission.@transmission[0].bind_address_ipv4='0.0.0.0'
transmission.@transmission[0].bind_address_ipv6='::'
transmission.@transmission[0].blocklist_enabled='false'
transmission.@transmission[0].dht_enabled='true'
transmission.@transmission[0].download_queue_enabled='true'
transmission.@transmission[0].download_queue_size='4'
transmission.@transmission[0].encryption='1'
transmission.@transmission[0].idle_seeding_limit_enabled='false'
transmission.@transmission[0].incomplete_dir_enabled='false'
transmission.@transmission[0].lazy_bitfield_enabled='true'
transmission.@transmission[0].lpd_enabled='false'
transmission.@transmission[0].message_level='3'
transmission.@transmission[0].peer_limit_global='240'
transmission.@transmission[0].peer_limit_per_torrent='60'
transmission.@transmission[0].peer_port='51413'
transmission.@transmission[0].peer_port_random_on_start='false'
transmission.@transmission[0].peer_socket_tos='default'
transmission.@transmission[0].pex_enabled='true'
transmission.@transmission[0].port_forwarding_enabled='true'
transmission.@transmission[0].preallocation='1'
transmission.@transmission[0].queue_stalled_enabled='true'
transmission.@transmission[0].queue_stalled_minutes='30'
transmission.@transmission[0].ratio_limit_enabled='false'
transmission.@transmission[0].rpc_bind_address='0.0.0.0'
transmission.@transmission[0].rpc_enabled='true'
transmission.@transmission[0].rpc_port='9091'
transmission.@transmission[0].rpc_url='/transmission/'
transmission.@transmission[0].scrape_paused_torrents_enabled='true'
transmission.@transmission[0].script_torrent_done_enabled='false'
transmission.@transmission[0].seed_queue_enabled='false'
transmission.@transmission[0].speed_limit_down_enabled='false'
transmission.@transmission[0].speed_limit_up_enabled='false'
transmission.@transmission[0].start_added_torrents='true'
transmission.@transmission[0].trash_original_torrent_files='false'
transmission.@transmission[0].upload_slots_per_torrent='14'
transmission.@transmission[0].utp_enabled='true'
transmission.@transmission[0].scrape_paused_torrents='true'
transmission.@transmission[0].watch_dir_enabled='false'
transmission.@transmission[0].enabled='1'
transmission.@transmission[0].download_dir='/mnt/data/DLNA/movie'
transmission.@transmission[0].rename_partial_files='false'
transmission.@transmission[0].rpc_whitelist_enabled='false'
transmission.@transmission[0].rpc_authentication_required='true'
transmission.@transmission[0].rpc_username='wxg'
transmission.@transmission[0].rpc_password='wxg'
transmission.@transmission[0].cache_size_mb='5'
transmission.@transmission[0].umask='0'
transmission.@transmission[0].user='root'

obsy, 2015-10-23:
Cannot reproduce that. Working fine with these settings on 15.05.  Kill all instances and try run again by /etc/init.d/transmission start. Show 

ps | grep transmission

KisImre, 2015-12-20:
After sysupgrade I had the same issue. The only difference I found was in the init.d script. The newer version limits the transmission's memory usage to the 3/5 of the system memory by these lines:
```shell
#give transmission 3/5 of the memory of the system
local USE
local MEM=`grep MemTotal /proc/meminfo|sed 's/ \+/ /g'|cut -d ' ' -f 2`
if test "$MEM" -gt 1;then
	USE=`expr $MEM \* 3000 / 5`
fi
```

It's a reasonable thing to have this limit but it ignores if the system has a swap. By running directly the transmission daemon I've measured 27636 kB peak memory usage on a system with 28740 kB memory and 131068 kB swap (TP-Link WR1043ND). It's clearly above the 3/5 limit, and I think transmission silently dies if it can't allocate memory for itself.

roylez, 2015-12-28:
I do not think it is the memory allocation problem. I tried commenting out the following lines which takes `USE` and `MEM` and sets parameters accordingly, but transmission daemon still could not start.

```
        if test -z "$USE";then
                procd_set_param limits core="0 0"
        else
                procd_set_param limits core="0 0" as="$USE $USE"
                logger -t transmission "Starting with $USE virt mem"
        fi
```

roylez, 2015-12-28:
Problem found. In `/etc/config/transmission`, the default configuration has `enabled 0` in the beginning of the section. Setting it to 1 and transmission will start. Personally I think setting the default to be disabled surprises users and should be changed.

wxg4net, 2016-05-27:
reinstall openwrt.  and transmission. failed again

diizzyy, 2017-06-07:
@hnyman @thess 
15.05.1 is unsupported/frozen, please close.

@roylez @wxg4net 
If this still occurs, please open a new issue ticket and please test using LEDE before doing so.

closed 2017-06-07 by thess

xiehuc, 2017-07-09:
i finally found the reason.
1. you install some package on external storage. like opkg -d usb.
2. you use extroot method which run transmission in rootfs. 
so this time, transmission could not locate some library installed in usb (/etc/init.d would not read /etc/profile even you have set PATH or LD_LIBRARY_PATH)
3. so transmission could not up.

i delete all package on dest usb, reinstall to rootfs. the problems gone