openwrt/packages #1884: ddns-scripts: support for variomedia.de

w-oertl, 2015-10-22:
Hi! The following line makes the dyndns service of variomedia.de available for me. You might want to include it. It uses the DynDNS protocol over HTTPS. Manual installation of curl/wget and ca-certificates is needed (I use curl).

"variomedia.de"	"https://[USERNAME]:[PASSWORD]@dyndns.variomedia.de/nic/update?hostname=[DOMAIN]"

chris5560, 2015-10-23:
IPv6 support ?
I'll implement in the next rollup. Thanks

Christian

w-oertl, 2015-10-23:
Good point. Here are the updated lines for IPv4 and IPv6. Note the different hostname, dyndns4.variomedia.de vs. dyndns6.variomedia.de. The host dyndns.variomedia.de has both A and AAAA records, the others only one, with less potential for confusion.

"variomedia.de" "https://[USERNAME]:[PASSWORD]@dyndns4.variomedia.de/nic/update?hostname=[DOMAIN]&myip=[IP]"

"variomedia.de" "https://[USERNAME]:[PASSWORD]@dyndns6.variomedia.de/nic/update?hostname=[DOMAIN]&myip=[IP]"

yousong, 2016-08-13:
Close this as the issue was resolved by ddns-scripts maintainer ;)

closed 2016-08-13 by yousong