openwrt/packages #1893: 15.05 ar71xx: nodogsplash doesn't start at boot

nitr0man, 2015-10-27:
Configured nodogsplash doesn't start at boot, generated config is blank:

```
Tue Oct 27 08:56:15 2015 daemon.notice netifd: Network device 'wlan0' link is up
Tue Oct 27 08:56:15 2015 daemon.notice netifd: Interface 'lan' is enabled
Tue Oct 27 08:56:15 2015 daemon.notice netifd: Interface 'lan' is setting up now
Tue Oct 27 08:56:15 2015 daemon.notice netifd: Interface 'lan' is now up
Tue Oct 27 08:56:15 2015 daemon.notice netifd: Interface 'lan' has link connectivity 
Tue Oct 27 08:56:15 2015 daemon.info dnsmasq[1029]: read /etc/hosts - 1 addresses
Tue Oct 27 08:56:15 2015 daemon.info dnsmasq[1029]: read /tmp/hosts/dhcp - 0 addresses
Tue Oct 27 08:56:15 2015 daemon.info dnsmasq[1029]: read /tmp/hosts/odhcpd - 0 addresses
Tue Oct 27 08:56:15 2015 daemon.info dnsmasq-dhcp[1029]: read /etc/ethers - 0 addresses
Tue Oct 27 08:56:16 2015 user.notice firewall: Reloading firewall due to ifup of lan (wlan0)
Tue Oct 27 08:56:18 2015 daemon.notice nodogsplash[1150]: Reading and validating configuration file /tmp/etc/nodogsplash_cfg0221da.conf
Tue Oct 27 08:56:18 2015 daemon.err nodogsplash[1150]: GatewayInterface is not set
Tue Oct 27 08:56:18 2015 daemon.err nodogsplash[1150]: Configuration is not complete, exiting...
Tue Oct 27 08:56:19 2015 daemon.info dnsmasq[1029]: exiting on receipt of SIGTERM
Tue Oct 27 08:56:19 2015 daemon.info dnsmasq[1172]: started, version 2.73 cachesize 150
Tue Oct 27 08:56:19 2015 daemon.info dnsmasq[1172]: compile time options: IPv6 GNU-getopt no-DBus no-i18n no-IDN DHCP no-DHCPv6 no-Lua TFTP no-conntrack no-ipset no-auth no-DNSSEC loop-detect inotify
Tue Oct 27 08:56:19 2015 daemon.info dnsmasq[1172]: DNS service limited to local subnets
Tue Oct 27 08:56:19 2015 daemon.info dnsmasq-dhcp[1172]: DHCP, IP range 172.16.0.100 -- 172.16.0.249, lease time 10m
Tue Oct 27 08:56:19 2015 daemon.info dnsmasq-dhcp[1172]: DHCP, IP range 172.16.1.100 -- 172.16.1.249, lease time 12h
Tue Oct 27 08:56:19 2015 daemon.info dnsmasq[1172]: using local addresses only for domain lan
Tue Oct 27 08:56:19 2015 daemon.warn dnsmasq[1172]: no servers found in /tmp/resolv.conf.auto, will retry
Tue Oct 27 08:56:19 2015 daemon.info dnsmasq[1172]: read /etc/hosts - 1 addresses
Tue Oct 27 08:56:19 2015 daemon.info dnsmasq[1172]: read /tmp/hosts/dhcp - 1 addresses
Tue Oct 27 08:56:19 2015 daemon.info dnsmasq[1172]: read /tmp/hosts/odhcpd - 0 addresses
Tue Oct 27 08:56:19 2015 daemon.info dnsmasq-dhcp[1172]: read /etc/ethers - 0 addresses
Tue Oct 27 08:56:23 2015 daemon.notice nodogsplash[1174]: Reading and validating configuration file /tmp/etc/nodogsplash_cfg0221da.conf
Tue Oct 27 08:56:23 2015 daemon.err nodogsplash[1174]: GatewayInterface is not set
Tue Oct 27 08:56:23 2015 daemon.err nodogsplash[1174]: Configuration is not complete, exiting...
Tue Oct 27 08:56:28 2015 daemon.notice nodogsplash[1175]: Reading and validating configuration file /tmp/etc/nodogsplash_cfg0221da.conf
Tue Oct 27 08:56:28 2015 daemon.err nodogsplash[1175]: GatewayInterface is not set
Tue Oct 27 08:56:28 2015 daemon.err nodogsplash[1175]: Configuration is not complete, exiting...
Tue Oct 27 08:56:33 2015 daemon.notice nodogsplash[1176]: Reading and validating configuration file /tmp/etc/nodogsplash_cfg0221da.conf
Tue Oct 27 08:56:33 2015 daemon.err nodogsplash[1176]: GatewayInterface is not set
Tue Oct 27 08:56:33 2015 daemon.err nodogsplash[1176]: Configuration is not complete, exiting...
```

When I run it by '/etc/init.d/nodogsplash start' - all is OK.

Image: openwrt-15.05-ar71xx-generic-tl-wr841n-v9-squashfs-factory.bin

Config:
```
root@OpenWrt:~# uci show nodogsplash
nodogsplash.@instance[0]=instance
nodogsplash.@instance[0].network='lan'
nodogsplash.@instance[0].idletimeout='1200'
nodogsplash.@instance[0].preauthenticated_users='allow tcp port 53' 'allow udp port 53' 'FirewallRule allow tcp port 22'
nodogsplash.@instance[0].users_to_router='allow tcp port 22' 'allow tcp port 23' 'allow tcp port 53' 'allow udp port 53' 'allow udp port 67' 'allow tcp port 80' 'allow tcp port 443'
nodogsplash.@instance[0].enabled='1'
nodogsplash.@instance[0].maxclients='30'
nodogsplash.@instance[0].authenticated_users='FirewallRule allow tcp port 53' 'FirewallRule allow udp port 53' 'FirewallRule allow tcp port 22' 'FirewallRule block to 192.168.0.0/16' 'FirewallRule block to 10.0.0.0/8' 'FirewallRule allow tcp port 80' 'FirewallRule allow tcp port 443' 'FirewallRule allow tcp port 143' 'FirewallRule allow tcp port 110' 'FirewallRule allow tcp port 3349' 'FirewallRule allow udp port 3349' 'FirewallRule allow tcp port 1723' 'FirewallRule allow tcp port 5222' 'FirewallRule allow tcp port 5223'
nodogsplash.@instance[0].gatewayname='Free HotSpot'
```

mathiashro, 2016-08-16:
I confirm the issue. Furthermore this issue does not happen when I do a config change via Luci/UCI and apply the config. Looks like a special race condition. See also:
https://dev.opennet-initiative.de/ticket/136

Test via:
(. /lib/functions/network.sh; network_get_device ifname <your-interface-name>; echo $ifname)

When this test does return valid data nodogsplash will be successfully started, if not nodogsplash will fail.

yousong, 2016-08-17:
I cannot find `nodogsplash` in this repo.  Where is it maintained and who is the current maintainer?

BTW. it looks like that the issue was just that `wlan0` as the `lan` interface was just not ready at the time that script was running.  If that was the case, it should possibly be workarounded with 3 choices

- Add procd interface trigger
- Add a hotplug script for wlan0
- Rename the package's init script with a `99-` prefix

diizzyy, 2017-06-02:
@hnyman @thess 
Out of tree/repo package, please close this.

hnyman, 2017-06-02:
The package is in the routing repo, that is the correct place.
https://github.com/openwrt-routing/packages

closed 2017-06-02 by hnyman