openwrt/packages #1902: perl crypt() function is unimplemented due to excessive paranoia

lsoltero, 2015-10-30:
the perl make file breaks crypt().
root@Optimizer:/usr/local/gmnbin# perl -d
Loading DB routines from perl5db.pl version 1.44
Editor support available.
Enter h or 'h h' for help, or 'man perldebug' for more help.
crypt("helllo","hello");
main::(-:1): crypt("helllo","hello")
DB<1> n
The crypt() function is unimplemented due to excessive paranoia. at - line 1.
at - line 1.
Debugged program terminated. Use q to quit or R to restart,
use o inhibit_exit to avoid stopping after program termination,
h q, h R or h o to get additional info.
DB<1>
See ​https://forum.openwrt.org/viewtopic.php?pid=297923#p297923
for details.
--luis

lsoltero, 2015-10-30:
please note that although CONFIG_PERL_THREADS is set crypt still does not work.  I am running on an MT7620a mipsel architecture.  My understanding from 
https://github.com/openwrt/packages/pull/854
is that threading/crypt should work on this architecture.

lsoltero@ubuntu:~/openwrt.cc$ grep -i perl .config | grep -i thread
CONFIG_PERL_THREADS=y
CONFIG_PACKAGE_perlbase-thread=y
CONFIG_PACKAGE_perlbase-threads=y

lsoltero, 2015-10-30:
one final note...  i am running with uClib which does not support crypt_r for which explanation lives here
https://github.com/openwrt/packages/pull/854
we are hesitant to use glibc due to flash size constraints.  our application does not run crypt in a multi-threaded environment.  so for us building with uClib, thread support, and using regular (non thread safe) crypt works fine.

So the make file logic as defined here
https://github.com/openwrt/packages/blob/for-15.05/lang/perl/Makefile#L48
makes sense but does not work for us.

thanks,

--luis

Naoir, 2015-10-30:
Yeah I've also been thinking about that for a while.

The perl documentation actually has a word on thread-safety(see "Thread-Safety of System Libraries" in perlthrtut, http://perldoc.perl.org/perlthrtut.html#Thread-Safety-of-System-Libraries).
To summarize it quickly, it states that perl tries it's best to use thread-safe system library calls _if they are available_, but falls back to unsafe ones otherwise. It would be perfectly fine to re-enable crypt() even if it isn't thread-safe.

So my current view is that crypt() should always be enabled, thead-safe or not. Does anyone see a problem with that?

Please note that I can't introduce new features in the stable tree, so crypt() will remain disabled in CC. I've posted a patch that enables crypt() in CC over at https://gist.github.com/Naoir/5b0023bc67d132f98308.

Hope that helps

Naoir, 2015-11-20:
crypt() is enabled for all libc implementations beginning with perl-5.22.0-4 in trunk. CC will stay with disabled crypt() as previously discussed. Closing.

Thanks!

closed 2015-11-20 by Naoir