openwrt/packages #1904: faild to build erlang on mips (ar71xx)

KazukiShimada, 2015-10-31:
@psycho-nico
Hi, Nico.
I came accross this error.

-D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -D_GNU_SOURCE -I/home/zukky/openwrt/trunk/build_dir/target-mips_34kc_musl-1.1.11/otp_src_17.5/erts/mips-openwrt-linux-gnu -fno-tree-copyrename -I/home/zukky/openwrt/trunk/staging_dir/target-mips_34kc_musl-1.1.11/usr/include -I/home/zukky/openwrt/trunk/staging_dir/target-mips_34kc_musl-1.1.11/include -I/home/zukky/openwrt/trunk/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_musl-1.1.11/usr/include -I/home/zukky/openwrt/trunk/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_musl-1.1.11/include/fortify -I/home/zukky/openwrt/trunk/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_musl-1.1.11/include -D_GNU_SOURCE -DHAVE_CONFIG_H -Wall -Wstrict-prototypes -Wmissing-prototypes -Wdeclaration-after-statement -DUSE_THREADS -D_THREAD_SAFE -D_REENTRANT -DPOSIX_THREADS -D_POSIX_THREAD_SAFE_FUNCTIONS -DERLANG_INTEGRATION -o /home/zukky/openwrt/trunk/build_dir/target-mips_34kc_musl-1.1.11/otp_src_17.5/erts/emulator/pcre/obj/mips-openwrt-linux-gnu/opt/pcre_exec.o /home/zukky/openwrt/trunk/build_dir/target
-mips_34kc_musl-1.1.11/otp_src_17.5/erts/emulator/pcre/pcre_exec.c 
{standard input}: Assembler messages:
{standard input}:1438: Error: operand value out of range for instruction
make[7]: *** [/home/zukky/openwrt/trunk/build_dir/target-mips_34kc_musl-1.1.11/otp_src_17.5/erts/emulator/pcre/obj/mips-openwrt-linux-gnu/opt/pcre_exec.o] Error 1
make[7]: Leaving directory `/home/zukky/openwrt/trunk/build_dir/target-mips_34kc_musl-1.1.11/otp_src_17.5/erts/emulator'
make[6]: *** [opt] Error 2 
make[6]: Leaving directory `/home/zukky/openwrt/trunk/build_dir/target-mips_34kc_musl-1.1.11/otp_src_17.5/erts/emulator'
make[5]: *** [opt] Error 2 
make[5]: Leaving directory `/home/zukky/openwrt/trunk/build_dir/target-mips_34kc_musl-1
.1.11/otp_src_17.5/erts'
make[4]: *** [emulator] Error 2 
make[4]: Leaving directory `/home/zukky/openwrt/trunk/build_dir/target-mips_34kc_musl-1
.1.11/otp_src_17.5'
make[3]: *** [noboot] Error 2 
make[3]: Leaving directory `/home/zukky/openwrt/trunk/build_dir/target-mips_34kc_musl-1
.1.11/otp_src_17.5'
make[2]: *** [/home/zukky/openwrt/trunk/build_dir/target-mips_34kc_musl-1.1.11/otp_src_
17.5/.built] Error 2
make[2]: Leaving directory `/home/zukky/openwrt/trunk/feeds/packages/lang/erlang' 
make[1]: *** [package/feeds/packages/erlang/compile] Error 2
make[1]: Leaving directory `/home/zukky/openwrt/trunk'
make: *** [package/feeds/packages/erlang/compile] エラー 2
zukky@dti-vps-srv123:~/openwrt/trunk$

yousong, 2015-11-16:
duplicate of #1750 

Looks like a MIPS16 related compiler issue.  Try setting `PKG_USE_MIPS16:=0` in the package makefile of `erlang` (the path is likely to be `package/feeds/packages/erlang/Makefile`)

KazukiShimada, 2015-12-16:
Hi, Yousong.
Thank you for your comment.
c

closed 2015-12-16 by KazukiShimada