openwrt/packages #1909: mosquitto package lacks mosquitto_passwd utility

eduperez, 2015-11-03:
Neither mosquitto nor mosquitto_nossl packages contain utility mosquitto_passwd; this utility is necessary to create users for mosquitto.

karlp, 2015-11-03:
Fair enough, I'll add it in.  I don't use it myself, so never bothered packaging it.

mhei, 2016-01-09:
Hi @karlp, could you please have a look on this patch. It's only compile tested.

karlp, 2016-01-11:
Yep, sorry, I was off on holidays all december, will merge this and update mosquitto this week I expect

karlp, 2016-01-12:
What do you think about just https://github.com/remakeelectric/owrt_pub_feeds/commit/472ed6c5a93bda2385e8a0e2ecb3550c49938b12

I'm bumping up to the newest release too, just have to get a CC build tested.

mhei, 2016-01-14:
Actually, I do not use this package at all, I only looked over all tickets to see where I could help to reduce the backlog... so I'm far away from knowing whether your patch would be the better approach.

My patch introduced a new package because I simply don't like packages which vary depending on config symbols, and in this case we already have the ssl/nossl stuff for this source package.
But this is my personal feeling and you are the maintainer :smile: I can live with every decision you will make :+1:

karlp, 2016-01-15:
Fixed in both CC and trunk, along with updates to newest.  Sorry about the delay!

closed 2016-01-15 by karlp

ronyboss, 2016-01-17:
I am getting error while making openwrt with mosquitto_password utility.

make[3]: Entering directory '/home/rony/linuxBuild/openwrt_rpi/openwrt/feeds/packages/net/mosquitto'
Makefile:196: *** missing separator.  Stop.

Any idea what is missing in makefile of mosquitto?

ronyboss, 2016-01-17:
I did tried the recipe of @mhei , and it worked for me. But newly added makefile of you Mr. @karlp . Didn't worked for me somehow. Would you like to check the issue @karlp ? And let me know if I am doing anything wrong.

Thanks in advance.

karlp, 2016-01-17:
make sure you don't have some remnant of both makefiles?

ronyboss, 2016-01-18:
No I don't. I used your makefile, and having error. Then I totally replaced the makefile of your with mhei's one. And that worked.

jow-, 2016-01-18:
Separator error fixed with 6d20db9138156e2acc189cb05a2948fbae8b2d07

karlp, 2016-01-18:
Oh damnit, now I see what had happened.  This got munched moving from my local branch to the commit branch :(  Thanks @jow-  for fixing my mistakes :(

ronyboss, 2016-01-18:
Thanks @jow-  It worked. @karlp It is resolved now.

ortegafernando, 2019-01-13:
Hi, sorry yo reopen it but I have openwrt 17.04 version an after installing mosquitto package I can't run mosquitto_passwd utility. It can't find it.
Thanks.

ortegafernando, 2019-01-14:
ok, thanks and sorry