openwrt/packages #1912: Add --ignore-site-config to bjam call in boost Makefile

g1itch, 2015-11-04:
Boost fails too build on my gentoo with following error message:
```
build_dir/target-mips_34kc_uClibc-0.9.33.2/boost_1_58_0/tools/build/src/build/feature.jam:494: in feature.validate-value-string from module feature
error: "none" is not a known value of feature <optimization>
error: legal values: "off" "speed" "space"
```

I solved this problem by adding the ```--ignore-site-config``` argument to bjam call in Makefile:
```diff
diff --git a/libs/boost/Makefile b/libs/boost/Makefile
index e9c2364..83b9ab1 100644
--- a/libs/boost/Makefile
+++ b/libs/boost/Makefile
@@ -196,6 +196,7 @@ define Build/Compile
                ) \
                bjam \
                        '-sBUILD=release <optimization>space <inlining>on <debug-symbols>off' \
+                       --ignore-site-config \
                        --toolset=gcc-$(ARCH) --build-type=minimal --layout=system abi=$(BOOST_ABI) \
                        --disable-long-double \
                        $(CONFIGURE_ARGS) \
```

Can you add this fix to package?

g1itch, 2015-11-06:
Thanks! Btw why ```for-15.05``` branch contains boost-1.58?

ClaymorePT, 2015-11-09:
@thess Is the boost makefile frozen at the 1.58 version, for the OpenWRT 15.05v?
Is it frozen for "keep the version" reasons ?

thess, 2015-11-09:
Usually we only consider serious bug fixes and security updates not just new features for a frozen branch. @ClaymorePT - Any recommendations?

closed 2015-11-09 by thess

ClaymorePT, 2015-11-10:
In my opinion, every submit that could improve a specific version without breaking compatibility should be accepted, more specifically those which could extend the usage of a lib or software to other targets, not previously supported. This is one of the cases.
Another case is described here https://github.com/openwrt/packages/issues/1938 .

thess, 2015-11-10:
Fine with me (others may disagree) - As long as there are no dependent changes required and compatibility is maintained. As the package maintainer, that is your call.

That said, we still need a separate PR for the 15.05 branch.

ClaymorePT, 2015-11-10:
Ok, I would then prefer this change to take effect after the separation, just to keep it clean and lean.