openwrt/packages #1938: packages/boost boost_python static library without -fPIC with target x86_64 in 15.05

carlosfrade, 2015-11-09:
Adding -fPIC to TARGET_CFLAGS resolved the issue for me.

Tested the same release with target Geode and it worked without changes.

Using packages repo:
src-git packages https://github.com/openwrt/packages.git;for-15.05

Using commit;
commit 48e7befbddb933d6cb046ec074470bbccdb060da
Author: blogic <blogic@3c298f89-4303-0410-b956-a3cf2f4a3e73>
Date:   Mon Aug 3 10:22:38 2015 +0000

ClaymorePT, 2015-11-09:
I'm on it.

Until the issue is properly corrected, for those with the same issue, the solution was already documented in the past.
[Link](http://boost.2283326.n4.nabble.com/How-to-compile-static-library-with-fPIC-from-boost-python-td4635727.html)

ClaymorePT, 2015-11-19:
FYY
https://github.com/openwrt/packages/pull/1991

closed 2015-12-05 by thess